espressif/esp-idf #14095: ESP-WIFI-MESH will connect with WPA3 only if it was connectd with WPA2 before

mmrein, 2024-06-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2 stable

### Espressif SoC revision.

Tested on ESP32 and ESP32-S2

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

Tested on Ethernet-Kit, PICO-KIT (D4), Saola-S2

### Power Supply used.

USB

### What is the expected behavior?

Connect node to root using `WIFI_AUTH_WPA3_PSK`

### What is the actual behavior?

Although the [documentation ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/esp-wifi-mesh.html#further-notes) states *"Data transmission uses Wi-Fi WPA2-PSK encryption"*, the connection with `WIFI_AUTH_WPA3_PSK` seems possible, but ONLY if the devices were previously connected with WPA2_PSK.

This also means root has to use `WIFI_STORAGE_FLASH`, but node seems to be working with `WIFI_STORAGE_RAM`.

### Steps to reproduce.

### Initial test with WPA3

1. Take https://github.com/espressif/esp-idf/tree/release/v5.2/examples/mesh/internal_communication for example
2. Change line: https://github.com/espressif/esp-idf/blob/aabb3699d163de0439c92e3eb1993fdec427e31a/examples/mesh/internal_communication/main/mesh_main.c#L435 to: 
```c
ESP_ERROR_CHECK(esp_mesh_set_ap_authmode(WIFI_AUTH_WPA3_PSK));
```
4. Use `esp_mesh_fix_root(true);` and `esp_mesh_set_type(MESH_ROOT / MESH_NODE);` to set fixed root and node
5. Build and upload with `esp_mesh_set_type(MESH_ROOT)` to root device and with  `esp_mesh_set_type(MESH_NODE)` to node
6. Restart both devices and check serial output. See logs below 

### Try connection with WPA2

7. Now use `WIFI_AUTH_WPA2_PSK` and repeat step 5.
8. Restart both devices and check serial output to see that node connected to root successfully.

### Repeat with WPA3

9. Now use `WIFI_AUTH_WPA3_PSK` again and repeat step 5.
10. Restart both devices and check serial output to see (to much surprise), that node connected to root successfully!

### Just to confirm

11. At this point devices should be able to reconnect even after power off.
12. Now just add `erase-flash` and repeat step 5 (to make sure the NVS partition for WiFi is erased). This only needs to be done for root at this point.
13. Restart both devices and see that node won't connect again.

### Debug Logs.

```plain
ROOT (relevant part that is repeating):
I (11738) wifi:removing station <30:83:98:d2:b6:90> after unsuccessful auth/assoc, AID = 0 
W (11738) wifi:rm mis 
I (11738) wifi:new:<9,0>, old:<9,0>, ap:<9,1>, sta:<0,0>, prof:9 
I (12528) mesh: [wifi]disconnected reason:201(), continuous:4/max:12, root, vote(,)<><>

NODE (relevant part that is repeating):
I (9129) wifi:new:<9,1>, old:<9,0>, ap:<9,1>, sta:<9,1>, prof:9 
I (9139) wifi:state: init -> auth (b0) 
I (9929) wifi:state: auth -> init (1c0) 
I (9929) wifi:new:<9,0>, old:<9,1>, ap:<9,1>, sta:<9,1>, prof:9 
I (9929) mesh: [wifi]disconnected reason:202(), continuous:4/max:12, non-root, vote(,)<><>
```

### More Information.

I am, in fact, using my fork of ESP-MDF (https://github.com/mmrein/esp-mdf) and took quite some time to make it somehow compatible with IDF v5+. **Main motivation was to use WPA3 with ESP-MDF**. 

Last few days I wanted to make sure the WPA3 works and discovered that it actually does not, even tho I'm pretty sure it did before.

So I started digging and dug out here, but I can't dig any deeper as this part of code is out of my control.

zhangyanjiaoesp, 2024-12-27:
@mmrein 

> Although the [documentation ](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/esp-wifi-mesh.html#further-notes)states "Data transmission uses Wi-Fi WPA2-PSK encryption", the connection with WIFI_AUTH_WPA3_PSK seems possible,

The documentation need to be updated. The connection support `WIFI_AUTH_WPA3_PSK`. 

And I just test with the `internal_communication` example, I can reproduce this problem. And the root cause has been found, will merged it ASAP, thanks for your reporting.

mmrein, 2025-01-06:
@zhangyanjiaoesp That's great to hear, thank you!

closed 2025-01-15 by espressif-bot

mmrein, 2025-03-21:
I've just tested release/v5.4 branch with e347bdd commit and indeed it does work even with my MDF fork, very nice.