espressif/esp-idf #14096: ADC Continuous Readings Affected by NIMBLE Initialization (nimble_port_init()) Resulting in Erratic Readings

filzek, 2024-06-27:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2-220-gaabb3699d1

### Espressif SoC revision.

esp32 rev 3

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32 V3 WROVER

### Power Supply used.

External 5V

### What is the expected behavior?

ADC readings should remain stable and accurate even after initializing NIMBLE with nimble_port_init().

### What is the actual behavior?

**After initializing NIMBLE, ADC readings become erratic, showing an offset of more than 24 from the correct central reading.**

When reading ADC without enabling NIMBLE using nimble_port_init(), the ADC readings are very stable on the ESP32 v3.0. Everything works perfectly. However, as soon as nimble_port_init() is called, the ADC readings become erratic, with a significant number of spurious values showing an odd offset of more than 24 from the correct central reading.

**Detailed Description**

- **Stable Readings:** Without calling nimble_port_init(), the ADC readings are stable and accurate.
- **Erratic Readings:** After calling nimble_port_init(), the ADC readings become erratic, with spurious values appearing frequently.
- **Test Scenario:**
1.       Without NIMBLE initialization, the system is in a wait state, and ADC readings are stable and accurate.
2.       After initializing NIMBLE and returning the system to a wait state, ADC readings become erratic.

### Steps to reproduce.

1. Set up the ESP32 v3.0.
2. Perform ADC readings without calling nimble_port_init().
3. Observe the stable and accurate ADC readings.
4. Initialize NIMBLE by calling nimble_port_init().
5. Perform ADC readings again.
6. Observe the erratic ADC readings with significant spurious values.

### Debug Logs.

```plain
BEFORE ENABLE NIMBLE 
Channel [5] Min [1761] Max [1776] offset [15] adjusted [0.02]
Channel [7] Min [1775] Max [1790] offset [15] adjusted [0.02]
Channel [3] Min [1775] Max [1792] offset [17] adjusted [0.03]
Channel [5] Min [1759] Max [1776] offset [17] adjusted [0.03]
Channel [7] Min [1775] Max [1788] offset [13] adjusted [0.01]
Channel [3] Min [1775] Max [1791] offset [16] adjusted [0.02]
Channel [5] Min [1761] Max [1776] offset [15] adjusted [0.02]
Channel [7] Min [1773] Max [1789] offset [16] adjusted [0.02]
Channel [3] Min [1775] Max [1792] offset [17] adjusted [0.03]
Channel [5] Min [1761] Max [1776] offset [15] adjusted [0.02]
Channel [7] Min [1775] Max [1789] offset [14] adjusted [0.02]
Channel [3] Min [1776] Max [1792] offset [16] adjusted [0.02]

AFTER BLE INITIALIZED WITH  nimble_port_init().
Channel [5] Min [1728] Max [1787] offset [59] adjusted [0.20]
Channel [7] Min [1739] Max [1806] offset [67] adjusted [0.23]
Channel [3] Min [1742] Max [1808] offset [66] adjusted [0.23]
Channel [5] Min [1732] Max [1783] offset [51] adjusted [0.17]
Channel [7] Min [1749] Max [1795] offset [46] adjusted [0.15]
Channel [3] Min [1751] Max [1797] offset [46] adjusted [0.15]
Channel [5] Min [1729] Max [1782] offset [53] adjusted [0.17]
Channel [7] Min [1747] Max [1794] offset [47] adjusted [0.15]
Channel [3] Min [1750] Max [1821] offset [71] adjusted [0.25]
```

### More Information.

This issue seems to be directly related to the initialization of NIMBLE. It would be helpful to investigate if there is a conflict between the ADC and NIMBLE subsystems or any shared resources that might be causing this instability.

filzek, 2024-06-27:
**Findings**

We were able to significantly improve the erratic ADC readings by making the following configuration changes:

1. **Pin NIMBLE and Bluetooth to the Second Core:**

- CONFIG_NIMBLE_PINNED_TO_CORE_1=y
- CONFIG_NIMBLE_PINNED_TO_CORE=1
- CONFIG_BT_NIMBLE_PINNED_TO_CORE_1=y
- CONFIG_BT_NIMBLE_PINNED_TO_CORE=1
- CONFIG_BTDM_CONTROLLER_PINNED_TO_CORE=1
- CONFIG_BTDM_CTRL_PINNED_TO_CORE_1=y
- CONFIG_BTDM_CTRL_PINNED_TO_CORE=1

2. **Disable Logging for NIMBLE:**

- 
- CONFIG_BT_NIMBLE_LOG_LEVEL_NONE=y
- # CONFIG_BT_NIMBLE_LOG_LEVEL_INFO is not set

3. **Disable Software Coexistence:**

- # CONFIG_BTDM_CTRL_HLI is not set
- # CONFIG_ESP_COEX_SW_COEXIST_ENABLE is not set
- # CONFIG_SW_COEXIST_ENABLE is not set
- # CONFIG_ESP32_WIFI_SW_COEXIST_ENABLE is not set
- # CONFIG_ESP_WIFI_SW_COEXIST_ENABLE is not set

These changes resulted in more stable ADC readings, reducing the spurious values and improving overall accuracy.

Channel [5] Min [1751] Max [1776] offset [25] adjusted [0.06]
Channel [7] Min [1771] Max [1789] offset [18] adjusted [0.03]
Channel [3] Min [1774] Max [1792] offset [18] adjusted [0.03]
Channel [5] Min [1751] Max [1777] offset [26] adjusted [0.06]
Channel [7] Min [1771] Max [1792] offset [21] adjusted [0.04]
Channel [3] Min [1773] Max [1795] offset [22] adjusted [0.05]

filzek, 2024-06-28:
Another problem found, it also corrupts the NIMBLE to allow devices to connect to it, broke the gatt to status 8 and later fasilled to status 133, this seens to runs only when adc continuous are enable.

xiongweichao, 2025-04-15:
Hi @filzek，may I ask if you are using `ADC_UNIT_2`?

filzek, 2025-04-15:
Hi @xiongweichao,

we are using ADC_UNIT_1, with the ADCS 3, 5 and 7, attn 0.

xiongweichao, 2025-04-16:
It looks very similar to this [issue](https://github.com/espressif/esp-idf/blob/e9bdd395995f60878e01cece10525987bacc8a5f/components/esp_driver_gpio/include/driver/gpio.h#L100).   Please try the [workaround](https://github.com/espressif/esp-idf/blob/e9bdd395995f60878e01cece10525987bacc8a5f/components/driver/deprecated/driver/adc.h#L108) to see if it can resolve the issue.

filzek, 2025-04-16:
we can try and log, but we are using continuous adc reading.

xiongweichao, 2025-04-21:
@filzek Sorry, I can't reproduce the issue. Can you share a reproduction example? Thanks!