espressif/esp-idf #14100: Is oc_rcp running on ESP32-H2 compatible with OBTR on Ubuntu Linux?

Askidea, 2024-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Can `ot_rcp`(examples/openthread/ot_rcp) uploaded to the ESP32-H2 board be connected to OBTR running on Ubuntu Linux for Spinel communication?
There are documents that say Ubuntu OBTR is compatible with nRF52840 RCP, but I am wondering if ESP32-H2 is also possible. Or does H2 RCP only work with the ESP32-S3 board or Border router DevKit?

Reference link: [Commission and control a Matter Thread device via the OTBR Snap](https://github.com/canonical/openthread-border-router-snap/wiki/Commission-and-control-a-Matter-Thread-device-via-the-OTBR-Snap)

zwx1995esp, 2024-06-28:
Hi @Askidea Yes, our ESP32H2 rcp can be connected to OBTR running on Ubuntu. Notice: our RCP example using a default baudrate `460800` for UART, you need to config this on OBTR Host via command:

`spinel+hdlc+uart:///dev/ttyUSB2?uart-baudrate=460800`

For example: "sudo otbr-agent OTBR_AGENT_OPTS="-I wpan0 -B wlan0 spinel+hdlc+uart:///dev/ttyUSB2?uart-baudrate=460800""

Askidea, 2024-06-28:
@zwx1995esp 
Thanks for your comments. I'll try as you mentioned. 
If you have any reference links related to the above, please let me know.

zwx1995esp, 2024-06-28:
@Askidea Yup, for setting up a linux based otbr, you can refer to the official docs https://openthread.io/guides/border-router/raspberry-pi. And for the [config of the RCP](https://openthread.io/guides/border-router/build#attach-and-configure-rcp-device) you can add the baudrate config metioned above(spinel+hdlc+uart:///dev/ttyUSB2?uart-baudrate=460800).

closed 2024-07-17 by chshu

sanzoghenzo, 2025-07-16:
Hi, I'm try to do the same thing these days with a ESP32-H2-DevKitM-1-N4.

Should the default configuration work out of the box, or do I need to change some settings in menuconfig?

zwx1995esp, 2025-07-17:
> [@Askidea](https://github.com/Askidea) Yup, for setting up a linux based otbr, you can refer to the official docs https://openthread.io/guides/border-router/raspberry-pi. And for the [config of the [...]

Hi, @sanzoghenzo you can refer to this comment and if you encounter any issues, feel free to let me know. Wishing you the best of luck!