espressif/esp-idf #14101: i2c new driver exception with sht30

achao1104, 2024-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

idf5.2.2

### Espressif SoC revision.

esp32s2

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

esp32s2

### Power Supply used.

USB

### What is the expected behavior?

I have manually  applied a fix from https://github.com/espressif/esp-idf/commit/a69c6e8f41b17562b01f75535dbccaebdb6c83d2
but it still reports as below

E (2630) i2c.master: s_i2c_synchronous_transaction(832): I2C transaction failed
E (2640) i2c.master: i2c_master_receive(1078): I2C transaction failed
I (2640) i2cutil: x4.
I (2640) sht30: sht3x_get_data 06 error
I (2650) main: tem_val 0, hum_val 0
I (3650) i2cutil: x1.
Guru Meditation Error: Core  0 panic'ed (StoreProhibited). Exception was unhandled.

Core  0 register dump:
PC      : 0x40377a0d  PS      : 0x00060033  A0      : 0x80377aad  A1      : 0x3fc951d0
0x40377a0d: i2c_ll_read_rxfifo at D:/project/datacollection/Espressif/frameworks/esp-idf-v5.2.2/components/hal/esp32s3/include/hal/i2c_ll.h:658
 (inlined by) i2c_isr_receive_handler at D:/project/datacollection/Espressif/frameworks/esp-idf-v5.2.2/components/driver/i2c/i2c_master.c:590

A2      : 0x3c040b4c  A3      : 0x3c040b5c  A4      : 0x3fc95200  A5      : 0x00000000
A6      : 0x3c040e1c  A7      : 0x3c040b80  A8      : 0x00000000  A9      : 0x00000000
A10     : 0x00000018  A11     : 0x00000000  A12     : 0x60013000  A13     : 0x3fc9b3c0
A14     : 0x60013000  A15     : 0x0000cdcd  SAR     : 0x00000000  EXCCAUSE: 0x0000001d
EXCVADDR: 0x00000000  LBEG    : 0x40056f5c  LEND    : 0x40056f72  LCOUNT  : 0xffffffff
0x40056f5c: memcpy in ROM
0x40056f72: memcpy in ROM

Backtrace: 0x40377a0a:0x3fc951d0 0x40377aaa:0x3fc95200 0x40376579:0x3fc95240 0x40377441:0x3fc95260 0x403791bf:0x3fc9bbd0 0x42002ef2:0x3fc9bbf0 0x4037d52d:0x3fc9bc10 0x4037c63d:0x3fc9bc30
0x40377a0a: i2c_ll_read_rxfifo at D:/project/datacollection/Espressif/frameworks/esp-idf-v5.2.2/components/hal/esp32s3/include/hal/i2c_ll.h:658
 (inlined by) i2c_isr_receive_handler at D:/project/datacollection/Espressif/frameworks/esp-idf-v5.2.2/components/driver/i2c/i2c_master.c:590
0x40377aaa: i2c_master_isr_handler_default at D:/project/datacollection/Espressif/frameworks/esp-idf-v5.2.2/components/driver/i2c/i2c_master.c:628
0x40376579: shared_intr_isr at D:/project/datacollection/Espressif/frameworks/esp-idf-v5.2.2/components/esp_hw_support/intr_alloc.c:445
0x40377441: _xt_lowint1 at D:/project/datacollection/Espressif/frameworks/esp-idf-v5.2.2/components/xtensa/xtensa_vectors.S:1240
0x403791bf: xt_utils_wait_for_intr at D:/project/datacollection/Espressif/frameworks/esp-idf-v5.2.2/components/xtensa/include/xt_utils.h:81
 (inlined by) esp_cpu_wait_for_intr at D:/project/datacollection/Espressif/frameworks/esp-idf-v5.2.2/components/esp_hw_support/cpu.c:132
0x42002ef2: esp_vApplicationIdleHook at D:/project/datacollection/Espressif/frameworks/esp-idf-v5.2.2/components/esp_system/freertos_hooks.c:59
0x4037d52d: prvIdleTask at D:/project/datacollection/Espressif/frameworks/esp-idf-v5.2.2/components/freertos/FreeRTOS-Kernel/tasks.c:4307 (discriminator 1)

### What is the actual behavior?

time out issue

### Steps to reproduce.

as above

mythbuster5, 2024-11-06:
Can this still be reproduced on latest master?

mythbuster5, 2025-01-21:
If there is no more update, I will close this issue soon, because I can't reproduce it locally.

mythbuster5, 2025-02-17:
I will close it since there is no more update. Feel free to reopen if there is any more update, thanks.

closed 2025-02-17 by mythbuster5