espressif/esp-idf #14104: mbedtls Linking Error

premkumar02031993, 2024-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

error: #error "MBEDTLS_SSL_TLS_C defined, but no protocols are active"
FAILED: esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aesce.c.obj

diplfranzhoepfinger, 2024-06-28:
#8960

premkumar02031993, 2024-07-01:
> Hi @premkumar02031993 ,
> 
> Could you please provide more details about the issue you're encountering? Specifically, it would be helpful to know:
> 
>     * Which operating system are you using?
> 
>     * What version of the ESP-IDF are you working with?
> 
>     * Under what circumstances or at what point in your workflow are you facing this issue?
> 
> 
> Any additional information you can provide will help us diagnose and address the problem more effectively.
> 
> Thank you!

I am using Linux operating system and  esp-idf v5.2.2 for my project development.while compiling the project , I am getting this error.

nileshkale123, 2024-07-01:
Could you please specify which project you are working on, and kindly send the corresponding sdkconfig file?

closed 2024-07-12 by nileshkale123