espressif/esp-idf #14105: extract Version into Filename

diplfranzhoepfinger, 2024-06-28:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

extract Version into Filename

diplfranzhoepfinger, 2024-07-18:
could this be the base: 

```
get_filename_component(ProjectId ${CMAKE_CURRENT_LIST_DIR} NAME)
string(REPLACE " " "_" ProjectId ${ProjectId})
project(${ProjectId})

```

KonstantinKondrashov, 2024-07-18:
Hi @diplfranzhoepfinger!
You can create your own cmd. Here is an example below. `idf.py my-release-cmd`.
Put this code into your CMakeLists.txt file.

https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#build-project-variables
```cmake
idf_build_get_property(project_ver PROJECT_VER)
idf_build_get_property(project_name PROJECT_NAME)
add_custom_target(my-release-cmd
                    COMMAND ${CMAKE_COMMAND} -E copy "${build_dir}/${project_name}.bin" "${build_dir}/${project_name}_${project_ver}.bin"
                    COMMENT "Release binary"
                )

````
It produces a file like this `app_v5.4-dev-1638-gec3e3724ad-dirty.bin`.

There is topic - the https://esp32.com/viewtopic.php?t=19274

diplfranzhoepfinger, 2024-07-23:
[@KonstantinKondrashov ](https://github.com/KonstantinKondrashov) cool! i will test.

KonstantinKondrashov, 2024-07-26:
I close this issue, feel free to reopen it if needed. Thanks.

closed 2024-07-26 by KonstantinKondrashov