espressif/esp-idf #14107: I want to ask about devkit v4

adrn689, 2024-06-29:
[image]

so in this picture it can be seen that if you have these two problems then the yellow component is removed, but I checked in my esp32 did not find a yellow component only a tin with no components there, so what should i do please help

adokitkat, 2024-07-14:
Does the board function correctly otherwise?

krzychb, 2025-03-07:
> I checked in my esp32 did not find a yellow component only a tin with no components there, so what should i do please help

C15 is not installed on newer boards.

[image]

You don’t need to take any action. The absence of C15 does not affect the board’s operation.

In general, a capacitor across switch contacts is typically used for debouncing or reducing electrical noise.

closed 2025-03-07 by krzychb