espressif/esp-idf #14112: ESP32-S3 wrong microphone data for I2S_SLOT_MODE_MONO

szprytny, 2024-06-30:
Is there a reason why `slot_mask` is always `I2S_STD_SLOT_BOTH`, unlike for ESP32 and ESP32-S2 where it depends on `mono_or_stereo` parameter?

https://github.com/espressif/esp-idf/blob/e7070e777a079695f69720ffb3c631c5fe620cc6/components/esp_driver_i2s/include/driver/i2s_std.h#L144

L-KAYA, 2025-01-10:
The I2S hardware on ESP32/ESP32-S2 are different from ESP32-S3.

On ESP32 and ESP32-S2, even you set mono, the both slots will send same data on TX side. But other chips can decide which slot to send. To align the MONO behavior on different chips, the later chips uses BOTH slot to transmit mono audio.

Alvin1Zhang, 2025-01-10:
Thanks for reporting, sorry for the slow turnaround, feel free to reopen.

closed 2025-01-10 by Alvin1Zhang

jdecodes, 2026-10-29:
Is there any solution for this? even if the mic is mono, it's providign the stereo data because of this? how to handle it gracefully?