espressif/esp-idf #14113: esp_wifi_get_config retruns station ssid and password as empty

imdahsiaria90, 2024-06-30:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, I have an old project that was complied using ESPIDF - d60ba848-dirty
The project stored Station wifi credentials using esp_wifi_set_config api and later recovered it using esp_wifi_get_config.

I am now building a new project using the espidf v5 where I plan to do a OTA update and recover the wifi credentails.
when I call esp_get_wifi_config, the ssid and password are empty

Question:
1 .Is this command no longer available  ?
2. where is the location of function esp_wifi_get_config located in esp-idf ?
3. Is there any other way i can recover the ssid and password contents ?

hansw123, 2024-09-09:
@imdahsiaria90 
sorry for late reply

can you provide more info,such as new idf version 、reproduce flow and more details
esp_wifi_get_config（） will be test in  idf  release test,so it still work well

hansw123, 2024-09-12:
@imdahsiaria90 
did you mean get config after OTA？

closed 2024-10-18 by Sherry616