espressif/esp-idf #14117: core_dump_elf.c error array-bounds

Massimiliano-solutiontech, 2024-07-01:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

That the project compiles

### What is the actual behavior?

```bash
In function 'elf_write_note_header',
    inlined from 'elf_add_wdt_panic_details' at /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/espcoredump/src/core_dump_elf.c:559:25,
    inlined from 'elf_write_core_dump_info' at /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/espcoredump/src/core_dump_elf.c:622:12,
    inlined from 'esp_core_dump_do_write_elf_pass' at /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/espcoredump/src/core_dump_elf.c:667:15:
/Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/espcoredump/src/core_dump_elf.c:210:5: error: 'memcpy' forming offset [18, 19] is out of the bounds [0, 18] [-Werror=array-bounds=]
  210 |     memcpy((void*)name_buffer, (void*)name, name_len);
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
```

### Steps to reproduce.

[sdkconfig.txt](https://github.com/user-attachments/files/16054725/sdkconfig.txt)

### Build or installation Logs.

```plain
Executing task: /opt/homebrew/bin/cmake -G=Ninja -DPYTHON_DEPS_CHECKED=1 -DESP_PLATFORM=1 -B=/Users/massimilianomontagni/projects/firmware/build -S=/Users/massimilianomontagni/projects/firmware -DCCACHE_ENABLE=1 

-- Found Git: /opt/homebrew/bin/git (found version "2.44.0") 
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /Users/massimilianomontagni/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /Users/massimilianomontagni/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /Users/massimilianomontagni/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin/riscv32-esp-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32c6
Processing 4 dependencies:
[1/4] espressif/esp_secure_cert_mgr (2.4.1)
[2/4] espressif/nghttp (1.58.0)
[3/4] espressif/sh2lib (1.0.5)
[4/4] idf (5.2.2)
[... 85 lines cut ...]
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Adding linker script /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/soc/esp32c6/ld/esp32c6.peripherals.ld
-- Bootloader project name: "bootloader" version: 1
-- Adding linker script /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.ld
-- Adding linker script /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.api.ld
-- Adding linker script /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.rvfp.ld
-- Adding linker script /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.newlib.ld
-- Adding linker script /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.phy.ld
-- Adding linker script /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.wdt.ld
-- Adding linker script /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.systimer.ld
-- Adding linker script /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_rom/esp32c6/ld/esp32c6.rom.version.ld
-- Adding linker script /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/bootloader/subproject/main/ld/esp32c6/bootloader.ld
-- Adding linker script /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/bootloader/subproject/main/ld/esp32c6/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table riscv soc spi_flash
-- Component paths: /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/bootloader /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/bootloader_support /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/efuse /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_app_format /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_bootloader_format /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_common /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_hw_support /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_rom /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esp_system /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/esptool_py /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/freertos /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/hal /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/log /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/bootloader/subproject/main /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/bootloader/subproject/components/micro-ecc /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/newlib /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/partition_table /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/riscv /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/soc /Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/spi_flash
-- Configuring done (2.4s)
-- Generating done (0.0s)
-- Build files have been written to: /Users/massimilianomontagni/projects/firmware/build/bootloader
ninja: build stopped: subcommand failed.
```

erhankur, 2024-07-01:
@Massimiliano-solutiontech Thanks for the report. I got the error reason. 

Because of the alignment to the up, name_len becomes 20 but name itself still sizeof 18. So compiler throws an error.  I will fix it.

Massimiliano-solutiontech, 2024-07-01:
@erhankur thanks for your quick reply, do you have a workaround or an ETA for the fix. Because right now I'm stucked with this project

erhankur, 2024-07-01:
Please try this patch.

[coredump_name_len.patch](https://github.com/user-attachments/files/16056203/coredump_name_len.patch)

Massimiliano-solutiontech, 2024-07-01:
@erhankur 

the scope of `name_len` is inside the `else if` but the variable is used also in the return `return ALIGN_UP(name_len + ALIGN_UP(self->note_data_size, 4) + sizeof(elf_note), 4);`

if I move `name_len` to the begging the project compiles

```c
static int elf_add_wdt_panic_details(core_dump_elf_t *self)
{
    core_dump_elf_opaque_t param = {
        .self = self,
        .total_size = 0,
        .size_only = false,
    };

    if (self->elf_stage == ELF_STAGE_CALC_SPACE) {
        param.size_only = true;
        esp_task_wdt_print_triggered_tasks(elf_write_core_dump_note_cb, &param, NULL);
        ELF_CHECK_ERR((param.total_size > 0), ELF_PROC_ERR_OTHER, "wdt panic message len is zero!");
        self->note_data_size = param.total_size;
    } else if (self->elf_stage == ELF_STAGE_PLACE_DATA) {
        esp_err_t err = elf_write_note_header(self,
                                        ELF_ESP_CORE_DUMP_PANIC_DETAILS_NOTE_NAME,
                                        strlen(ELF_ESP_CORE_DUMP_PANIC_DETAILS_NOTE_NAME),
                                        self->note_data_size,
                                        ELF_ESP_CORE_DUMP_PANIC_DETAILS_TYPE);
        if (err != ESP_OK) {
            return err;
        }
        uint32_t name_len = ALIGN_UP(strlen(ELF_ESP_CORE_DUMP_PANIC_DETAILS_NOTE_NAME), 4);
        esp_task_wdt_print_triggered_tasks(elf_write_core_dump_note_cb, &param, NULL);
        ELF_CHECK_ERR((param.total_size > 0), ELF_PROC_ERR_WRITE_FAIL, "Write ELF note data failure (%d)", err);
        const uint32_t mod = self->note_data_size & 3;
        if (mod != 0) {
            uint8_t pad_bytes[3] = {0};
            uint32_t pad_size = 4 - mod;
            ESP_COREDUMP_LOG_PROCESS("Core dump note needs %d bytes padding", pad_size);
            err = esp_core_dump_write_data(&self->write_data, pad_bytes, pad_size);
            ELF_CHECK_ERR((err == ESP_OK), ELF_PROC_ERR_WRITE_FAIL, "Write ELF note padding failure (%d)", err);
        }
    }

	return ALIGN_UP(name_len + ALIGN_UP(self->note_data_size, 4) + sizeof(elf_note), 4);
}
```

```bash
/Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/espcoredump/src/core_dump_elf.c: In function 'elf_add_wdt_panic_details':
/Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/espcoredump/src/core_dump_elf.c:566:18: warning: unused variable 'name_len' [-Wunused-variable]
  566 |         uint32_t name_len = ALIGN_UP(strlen(ELF_ESP_CORE_DUMP_PANIC_DETAILS_NOTE_NAME), 4);
      |                  ^~~~~~~~
/Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/espcoredump/src/core_dump_elf.c:579:25: error: 'name_len' undeclared (first use in this function)
  579 |         return ALIGN_UP(name_len + ALIGN_UP(self->note_data_size, 4) + sizeof(elf_note), 4);
      |                         ^~~~~~~~
/Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/espcoredump/src/core_dump_elf.c:103:27: note: in definition of macro 'ALIGN_UP'
  103 | #define ALIGN_UP(x, a) (((x) + (a) - 1) & ~((a) - 1))
      |                           ^
/Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/espcoredump/src/core_dump_elf.c:579:25: note: each undeclared identifier is reported only once for each function it appears in
  579 |         return ALIGN_UP(name_len + ALIGN_UP(self->note_data_size, 4) + sizeof(elf_note), 4);
      |                         ^~~~~~~~
/Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/espcoredump/src/core_dump_elf.c:103:27: note: in definition of macro 'ALIGN_UP'
  103 | #define ALIGN_UP(x, a) (((x) + (a) - 1) & ~((a) - 1))
      |                           ^
/Users/massimilianomontagni/esp/v5.2.2/esp-idf/components/espcoredump/src/core_dump_elf.c:580:1: error: control reaches end of non-void function [-Werror=return-type]
  580 | }
      | ^
```

erhankur, 2024-07-01:
Aah yes. Ok go ahead like this.  I will fix it and backport to the release branches. But it will take some time to sync.

Massimiliano-solutiontech, 2024-07-01:
Ok thanks, should I close this issue or is better to wait until the fix is published in a stable branch?

erhankur, 2024-07-01:
Corrected patch is here.  

[coredump_name_len.patch](https://github.com/user-attachments/files/16058646/coredump_name_len.patch)

I will keep this open until the fix is merged.

Massimiliano-solutiontech, 2024-07-01:
Confirm 

Thanks for the support

closed 2024-07-17 by espressif-bot