espressif/esp-idf #14119: BLE Mesh Nodes should not relay messages multiple times

jtmullen, 2024-07-01:
### Is your feature request related to a problem?

I have a BLE Mesh network where a high TTL is needed due to the large physical size, but there are often groups of nodes relatively close together. In this case, I find that these bunches of nodes relay the same messages over and over again to the point where they are unable to actually operate reliably. They are completely overwhelmed by the number of messages, but most of the messages are ones they have received before. Each node ends up seeing the message multiple times at each TTL level. 

I am still trying to fully understand the mesh code, but I see that the relay function is implemented at the network layer and the replay check is at the transport layer. So a message will already have been queued to relay independent of the replay check. Additionally, the nodes only seem to store sequence numbers for devices that send messages directly to their address or an address they subscribe to, not all nodes. So they couldn't be doing replay checks for messages from nodes that haven't sent them anything yet. 

### Describe the solution you'd like.

In their [BLE Mesh Networking Overview](https://www.bluetooth.com/wp-content/uploads/2019/03/Mesh-Technology-Overview.pdf) I see that the Bluetooth SIG specifies where relay should be implemented (Network Layer) but not where replay protection should be implemented. Maybe there is other documentation I have not found.  But this [application note](https://www.silabs.com/documents/public/application-notes/an1316-bluetooth-mesh-network-optimization.pdf) from Silicon Labs about Network Optimization notes that sequence numbers should be checked before Relaying to help prevent the issues I am seeing.  I would like this to possible with the ESP-IDF

### Describe alternatives you've considered.

The application note above also offers other optimizations that are possible. We are tuning the TTL based on network size as a first step, which has somewhat reduced but not eliminated the problem. 

Our product is installed in the field by 3rd parties in significantly varied environments so network topology information is unique for every install and difficult to capture with enough detail for some other optimizations.  We are looking into ways to map the network to reduce the number of relay nodes and/or further reduce TTL, however doing this mapping increases the cost of installation and may not be static for the life of the product meaning. These are far more difficult and less reliable solutions than just not relaying each message effectively [TTL]<sup>[Node Count]</sup> times by checking sequence number before relaying.

forx157, 2024-07-09:
Hi,  @jtmullen 
Thank you very much for your research, regarding what you said about not relaying RPL messages is already supported in IDF 4.4 and later versions, it can be enabled in Menuconfig via the following path `Top) → Component config → ESP BLE Mesh Support → Make BLE Mesh experimental features visible` and `Top) → Component config → ESP BLE Mesh Support → Not relay replayed messages in a mesh network`.

Additionally, the impact of message flooding can be reduced by increasing the RPL list size when there are too many messages forwarded by the network (`Top) → Component config → ESP BLE Mesh Support → Network message cache size
 `) to reduce the impact of message flooding.

chegewara, 2024-08-01:
Hi,
maybe not related and even it may be useless, but i hope it may help:
- try to decrease number of relays, to not have too many relays too close each other
- how about mesh v1.1 feature, direct path relaying? (or whatever it is called)

jtmullen, 2024-08-02:
Hi @forx157 thanks for the information and sorry for the slow response. This seems to be what I was looking for.  Not sure why it does not come up when searching in the documentation. 

Is there a reason this is experimental? Any limitations or known issues we should be aware of?

forx157, 2025-01-07:
@whatdtech
[image]
Are you currently using config server with relay enabled?
If it's already on, please share your sdkconfig and info level logs with me!

whatdtech, 2025-01-08:
Sorry for the late reply. Yes im using config server with relay enabled. Provisioner is esp32s3 with st7789 display and loaded vendor client model, other two are vendor server models with esp32 mcu. Communication between one client and server is fine also tested individual communication between client and relay , but when I put them apart and introduce relay node, client no more receiving. 

**### vendor client log**
> ``
I (27) boot: ESP-IDF v5.3.2 2nd stage bootloader
I (27) boot: compile time Jan  7 2025 21:41:48
I (27) boot: Multicore bootloader
I (30) boot: chip revision: v0.1
I (34) boot: efuse block revision: v1.2
I (38) boot.esp32s3: Boot SPI Speed : 80MHz
I (43) boot.esp32s3: SPI Mode       : DIO
I (48) boot.esp32s3: SPI Flash Size : 16MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (76) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00200000
I (91) boot:  3 storage1         Unknown data     01 82 00210000 00020000
I (99) boot:  4 storage2         Unknown data     01 82 00230000 00050000
I (106) boot:  5 storage3         Unknown data     01 82 00280000 00080000
I (114) boot: End of partition table
I (118) esp_image: segment 0: paddr=00010020 vaddr=3c0d0020 size=3e1e8h (254440) map
I (171) esp_image: segment 1: paddr=0004e210 vaddr=3fc9aa00 size=01e08h (  7688) load
I (173) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=c2f14h (798484) map
I (318) esp_image: segment 3: paddr=00112f3c vaddr=3fc9c808 size=02c2ch ( 11308) load
I (320) esp_image: segment 4: paddr=00115b70 vaddr=40374000 size=1696ch ( 92524) load
I (353) boot: Loaded app from partition at offset 0x10000
I (353) boot: Disabling RNG early entropy source...
I (365) cpu_start: Multicore app
I (374) cpu_start: Pro cpu start user code
I (374) cpu_start: cpu freq: 160000000 Hz
I (374) app_init: Application information:
I (377) app_init: Project name:     vendor_client
I (383) app_init: App version:      a5c9a02f-dirty
I (388) app_init: Compile time:     Jan  7 2025 22:45:37
I (394) app_init: ELF file SHA256:  a07ffb1b9...
I (399) app_init: ESP-IDF:          v5.3.2
I (404) efuse_init: Min chip rev:     v0.0
I (409) efuse_init: Max chip rev:     v0.99
I (414) efuse_init: Chip rev:         v0.1
I (419) heap_init: Initializing. RAM available for dynamic allocation:
I (426) heap_init: At 3FCA7790 len 00041F80 (263 KiB): RAM
I (432) heap_init: At 3FCE9710 len 00005724 (21 KiB): RAM
I (438) heap_init: At 3FCF0000 len 00008000 (32 KiB): DRAM
I (444) heap_init: At 600FE100 len 00001EE8 (7 KiB): RTCRAM
I (452) spi_flash: detected chip: gd
I (455) spi_flash: flash io: dio
I (459) sleep: Configure to isolate all GPIO pins in sleep state
I (466) sleep: Enable automatic switching of GPIO sleep configuration
I (473) coexist: coex firmware version: cbb41d7
I (501) coexist: coexist rom version e7ae62f
I (502) main_task: Started on CPU0
I (512) main_task: Calling app_main()
I (512) ST7789-BLEMESH: Initializing...
use esp timer !!!
E (522) esp_timer: Task is already initialized
I (522) gpio: GPIO[0]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:3 
I (522) ST7789-BLEMESH: Initializing SPIFFS
I (542) ST7789-BLEMESH: Mount /fonts to storage1 success
I (542) ST7789-BLEMESH: Partition size: total: 113201, used: 0
I (562) ST7789-BLEMESH: Mount /images to storage2 success
I (562) ST7789-BLEMESH: Partition size: total: 293921, used: 0
W (592) SPIFFS: mount failed, -10025. formatting...
I (3882) ST7789-BLEMESH: Mount /icons to storage3 success
I (3882) ST7789-BLEMESH: Partition size: total: 474641, used: 0
I (3902) BLE_INIT: BT controller compile version [d752dea]
I (3902) BLE_INIT: Bluetooth MAC: 34:85:18:93:b3:e6
I (3902) phy_init: phy_version 680,a6008b2,Jun  4 2024,16:41:10
W (3912) phy_init: failed to load RF calibration data (0x1102), falling back to full calibration
[... 5696 lines cut ...]
#
CONFIG_WIFI_PROV_SCAN_MAX_ENTRIES=16
CONFIG_WIFI_PROV_AUTOSTOP_TIMEOUT=30
# CONFIG_WIFI_PROV_BLE_BONDING is not set
# CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION is not set
# CONFIG_WIFI_PROV_BLE_NOTIFY is not set
# CONFIG_WIFI_PROV_KEEP_BLE_ON_AFTER_PROV is not set
CONFIG_WIFI_PROV_STA_ALL_CHANNEL_SCAN=y
# CONFIG_WIFI_PROV_STA_FAST_SCAN is not set
# end of Wi-Fi Provisioning Manager
# end of Component config

# CONFIG_IDF_EXPERIMENTAL_FEATURES is not set

# Deprecated options for backward compatibility
# CONFIG_APP_BUILD_TYPE_ELF_RAM is not set
# CONFIG_NO_BLOBS is not set
# CONFIG_ESP32_NO_BLOBS is not set
# CONFIG_ESP32_COMPATIBLE_PRE_V2_1_BOOTLOADERS is not set
# CONFIG_ESP32_COMPATIBLE_PRE_V3_1_BOOTLOADERS is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_NONE is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_ERROR is not set
# CONFIG_LOG_BOOTLOADER_LEVEL_WARN is not set
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
# CONFIG_LOG_BOOTLOADE

whatdtech, 2025-01-08:
I see certain macros and functions in API regarding relay like

>  * ESP_BLE_MESH_ADDR_RELAYS
>  * ESP_BLE_MESH_NET_PRIMARY
>  * ESP_BLE_MESH_RELAY_ENABLED
>  * ESP_BLE_MESH_RELAY_NOT_SUPPORTED
>  * ESP_BLE_MESH_FEATURE_RELAY
>  * ESP_BLE_MESH_TAG_RELAY
>  * ESP_BLE_MESH_MODEL_OP_RELAY_GET
>  * 
> Config Relay Get
> ESP_BLE_MESH_MODEL_OP_RELAY_SET
> Config Relay Set
> ESP_BLE_MESH_MODEL_OP_RELAY_STATUS
> esp_ble_mesh_cfg_client_get_state_t ESP_BLE_MESH_MODEL_OP_RELAY_GET
> esp_ble_mesh_cfg_client_set_state_t ESP_BLE_MESH_MODEL_OP_RELAY_SET
> esp_ble_mesh_cfg_relay_set_t relay_set ESP_BLE_MESH_MODEL_OP_RELAY_SET
> esp_ble_mesh_cfg_relay_status_cb_t relay_status The relay status value
> esp_ble_mesh_cfg_srv_t uint8_t relay uint8_t relay_retransmit
> esp_ble_mesh_cfg_relay_set_t uint8_t relay The relay value uint8_t relay_retransmit The relay retransmit value
> esp_ble_mesh_cfg_relay_status_cb_t uint8_t relay uint8_t retransmit
> ESP_BLE_MESH_TIME_RELAY Time Role - Mesh Time Relay
> 

does any of them needed for relay functioning

forx157, 2025-01-08:
@whatdtech 

I don't see the client loss message in the log you gave me, in the client log, at the end there is also a message received from the server。
[image]

whatdtech, 2025-01-08:
Thanks for the reply. That is a message is received  from the server when it is near the client. But later I moved the server more than 10 metrs then nothing is received on the client, so i switched on the relay node and put it in between them around 5 metrs apart even then nothing is received ont the client. It stays idle

whatdtech, 2025-01-08:
[main-code.zip](https://github.com/user-attachments/files/18341064/main-code.zip)
 I zipped all the codes of provisioner and nodes. Kindly have a look

chegewara, 2025-01-08:
> Thanks for the reply. That is a message is received from the server when it is near the client. But later I moved the server more than 10 metrs then nothing is received on the client, so i switched [...]

Did you add that relay to mesh with provisioning?

whatdtech, 2025-01-08:
@chegewara thank you for the reply . This is the part I'm missing sir. Can you guide me to a document or an example.

chegewara, 2025-01-08:
Relay node is like any other node. You just have to do provisioning like with client or server node to join mesh with network and app key.

whatdtech, 2025-01-08:
yeah it is provisioned like a normal node and I can communicate with the unicast address like 0x0006 both from the client and to the relay node. only relay functionality is missing

whatdtech, 2025-01-08:
These are the relay node logs.

### Relay node

> I (60) boot: compile time Jan 6 2025 19:02:49
> I (60) boot: Multicore bootloader
> I (65) boot: chip revision: v1.0
> I (69) boot.esp32: SPI Speed : 40MHz
> I (74) boot.esp32: SPI Mode : DIO
> I (78) [...]

whatdtech, 2025-01-09:
I found one mistake that I was publishing to unicast address of provisioner, Not yet working after publishing to group address 0xC000