espressif/esp-idf #14120: wifi_prov_mgr build  fatal error: nimble/storage_port.h: No such file or directory

MarkMendelsohn, 2024-07-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.2

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Clean build of wifi_prov_mgr

### What is the actual behavior?

error during build

### Steps to reproduce.

1. . esp-idf/export.sh
2. cd to examples/provisioning/wifi_prov_mgr
3. idf.py set-target esp32s3
4. idf.py build

### Build or installation Logs.

```plain
/Volumes/Untitled/mpm/esp-idf/components/bt/host/nimble/port/src/nvs_port.c:12:10: fatal error: nimble/storage_port.h: No such file or directory
   12 | #include "nimble/storage_port.h"
      |          ^~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
[883/1075] Building C object esp-idf/bt/CMakeFiles/__idf_bt.dir/host/nimble/nimble/porting/nimble/src/os_mbuf.c.obj
ninja: build stopped: subcommand failed.
```

Shreeyash-17, 2024-07-02:
I have followed these steps to build the project. Please ensure you have also completed them:

1 git submodule update --init --recursive
2 ./install.sh
3 . ./export.sh
4 cd examples/provisioning/wifi_prov_mgr/
5 Removed any existing build directory:
   rm -rf build
6 idf.py set-target esp32s3
7 idf.py build

   1st step is important step here
After executing the above steps, no errors have occurred.

Shreeyash-17, 2024-07-02:
hi @MarkMendelsohn

closed 2024-07-23 by Alvin1Zhang