espressif/esp-idf #14123: How to set a custom mac as a base mac

xmlhp, 2024-07-02:
### Is your feature request related to a problem?

we have burned a customized mac into efuse. I don't understand where the decision is made to use the customer's mac or the ESP's own MAC.

ESP-Marius, 2024-07-03:
Did you take a look at this doc: https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/system/misc_system_api.html#custom-interface-mac

Does it answer your question?

xmlhp, 2024-07-04:
Yes, I have read it.
I am going to follow Custom MAC Address in eFuse . First use esp_efuse_mac_get_custom to read the MAC in eFuse BLK3, and then use esp_base_mac_addr_set to set the base MAC.
But I read esp-idf, and I don't know where to initialize the wifi interface and ble interface.
Please help me guide.

xmlhp, 2024-07-04:
I found the modification, thank you.

KonstantinKondrashov, 2024-07-04:
Hi @xmlhp!
You can have a look at this example - https://github.com/espressif/esp-idf/blob/master/examples/system/base_mac_address/main/base_mac_address_example_main.c.
It demonstrates how to use a custom MAC stored in efuse or other places. 

> But I read esp-idf, and I don't know where to initialize the wifi interface and ble interface.

MACs for these interfaces are derived from the BASE MAC address which can be customized:
- If CONFIG_ESP_MAC_USE_CUSTOM_MAC_AS_BASE_MAC==y then pre-defined efuse fields are used.
- You can manually set it as `esp_iface_mac_addr_set(custom_base_mac, ESP_MAC_BASE)`.

If there are no other questions I close it. Please feel free to reopen.
Thanks.

closed 2024-07-04 by KonstantinKondrashov