espressif/esp-idf #14124: ADC after TRNG: Cannot use `ADC` after disabling "TRNG" functionality (ESP32-C6)

playfulFence, 2024-07-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-1308-gb793ffe71d-dirty

### Espressif SoC revision.

esp32c6 (revision v0.0)

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6 DevKitC-1 v1.1

### Power Supply used.

USB

### What is the expected behavior?

After disabling source of entropy with `bootloader_random_disable` function from [this file](https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/bootloader_random_esp32c6.c) ADC is expected to be usable.

Example (on `esp32`): 
[image]

### What is the actual behavior?

However, it shots the `Guru Meditation Error`
[image]

### Steps to reproduce.

1) Add `bootloader_random_enable` and then `bootloader_random_disable` at the start of [oneshot_read](https://github.com/espressif/esp-idf/blob/master/examples/peripherals/adc/oneshot_read/main/oneshot_read_main.c#L55) ADC example
2) Flash and execute example on `ESP32-C6`

### More Information.

Tested on @peter-marcisovsky's Linux machine. Same behaviour on my `macOS` one.

I was implementing the `TRNG` functionality in the `esp-rs` drivers when I noticed that the steps from `esp-idf` for this problem were causing the `ADC` peripheral to not work correctly. I decided to double check if the "original" works correctly in this edge case, when someone wants to drop `ADC` entropy source for `RNG` and normally use `ADC` afterwards.

Problematic file: https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/bootloader_random_esp32c6.c

0xjakob, 2024-07-03:
@playfulFence Thanks for letting us know. We could reproduce this and will check what the issue is here.

0xjakob, 2024-08-05:
@playfulFence Just wanted to let you know we haven't forgot about this issue. We're still working on fixing it.

closed 2024-08-14 by espressif-bot

playfulFence, 2024-08-14:
Thanks a lot!