espressif/esp-idf #14125: JITP using ESP32 for AWSIOT

Detective-Roy, 2024-07-02:
### Is your feature request related to a problem?

I am working on a IOT project where I need to upload the data using MQTT. Its a consumer electronic device and we need to secure it using provisioning provided by AWS JITP. Every device is associated with unique device ID which we are providing while manufacturing. I want to make it secure and want to provision the individual certificate for each and every device. From AWS account I have 100 certificates ready for 100 devices. I want to know how we can provision it inside while manufacturing the device. Do I need to compile the fw 100 times including individual certificate and then flash. Or can we get only one compiled bin file and then combine with individual and flash it. Or can we flash the bin file first and then provision it later like I am doing for assigning unique name and device ID for each and every device?

### Describe the solution you'd like.

I want to know the process and any example which is available for this. I already read the article which is provided by AWS but this provisioning is not clear or may be I missed. Please share me some article, example and description for the same?

Alvin1Zhang, 2024-07-03:
Thanks for reporting.

euripedesrocha, 2024-07-04:
Hi @Detective-Roy, from my reading of the AWS JITP documentation and the description of your problem, I would advise you to use [nvs API](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/storage/nvs_flash.html) to store the certificates in the devices. 
From the mqtt client aspect, you can take the `ssl_muthual_auth` example as a starting point. The change that you will need to make from that example is that you will need to load the client certificates to memory and pass it to the mqtt client.

rrtandler, 2024-07-09:
Hi @Detective-Roy ,

I think you are missing the namespace in your .csv file. 
The namespace row should appear just before lines with client_cert and client_key and should then be used as a parameter of the nvs initialisation function returning the `core_storage` value.
Please see the example `examples/storage/nvsgen/nvs_data.csv` and look at the line 3 defining namespace "storage" for all subsequent lines in that .csv. Then you can see the matching value "storage" in the call to the `nvs_open_from_partition`in `examples/storage/nvsgen/main/nvsgen_example_main.c:34

Detective-Roy, 2024-07-10:
Hi @rrtandler ,
Thanks for the guidelines. I am able to flash the nvs_partition.bin which is created using csv file. I have made changes in my shell script as suggested. 
[try.txt](https://github.com/user-attachments/files/16158418/try.txt)

Now I am able to read my certificate and key on device initialization.
Steps followed:-
1- flashed the application bin file with above mentioned partition table (https://github.com/espressif/esp-idf/issues/14125#:~:text=st_partition_table.csv). 
2- then flashed nvs bin file generated by "try.sh".
3- On Initialisation- in my ```app_main()`` I am calling ```nvs_storage_init()``` function mentioned below-
```
int nvs_storage_init(void)
{
    esp_err_t ret;

    ret = nvs_flash_init();
	if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
        ESP_LOGI(NVS_TAG, "nvs_flash_init error \n");
		nvs_flash_erase();
		ret = nvs_flash_init();
	}

    ret = nvs_open_from_partition("nvs", "storage", NVS_READWRITE, &core_storage);
    if (ret != ESP_OK) {
        ESP_LOGE(NVS_TAG, "Error (%s) opening NVS handle!\n", esp_err_to_name(ret));
        return ret;
    }

    return ESP_OK;
}
```
Where ```core_storage``` declared globally as nvs handle.

4- After this - I am calling https://github.com/espressif/esp-idf/issues/14125#:~:text=read_certificates_from_nvs function.
5- I am able to read the cert and key whatever I have written while flashing using try.sh.

Issues:-
Now I am writing some more variable in NVS after initialisation in runtime. 
like-
```
nvs_set_str(core_storage, "curr_ssid", "");
nvs_set_str(core_storage, "currwifi_pwd", "");
nvs_set_u32(core_storage, "logs_enable", 0);
nvs_set_u8(core_storage, "mnfctrng_config", 1);

and etc ......
```
But, I think these variables are not written to the nvs because whenever I am trying to read back it is giving me same error that variable is not available.

is there any need to create a custom nvs for the extra variable which I am writing in runtime or am I doing anything wrong in this?

rrtandler, 2024-07-10:
@Detective-Roy 
Some troubleshooting ideas:

1. Evaluate and ideally log the result code of each call to the `nvs_set_*` and then `nvs_get*` to learn control your program flow apropriately.
2. From the code fragments you have posted, the function `void read_certificates_from_nvs()`ends with the `nvs_close(core_storage);` This invalidates the handle `core_storage` and if you do not re-initialize it, it prevents subsequent calls like `nvs_set_*(core_storage, ...` to work at all. You would see it from the result code(s), when logged.

Detective-Roy, 2024-07-10:
Hi, Thanks for pointing out the error. It worked.