espressif/esp-idf #14126: Optimising MbedTLS for size and incorrect KCONFIG explanations

ammaree, 2024-07-02:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Moving from IDF 5.0 to Master we have run into problems that our application does not fit into a 1MB partition anymore.
In the process we tried to optimise the MbedTLS config to only support a minimal subset of cipher suites and using the suites supported by the hosts we need to access we found that the following 2 protocols are common:
TLS_ECDHE_RSA_WITH_AES_128_GCM_SHA256
TLS_ECDHE_RSA_WITH_AES_256_GCM_SHA384

We are using HTTPS only for a client connection to upload coredump info (very irregular) as well as to do IP geolocation to determine timezone and related info (very, very irregularly).

Whilst going through all the options for MbedTLS (only v1.2 required) in SDKCONFIG we found the same information/description for 4 different options being:
[image]

Questions we have are:
a) Can we assume option descriptions (1st line of each one) is correct and text in additional info (2nd line) is incorrect?
b) If not, can we please have more info so as to be able to ensure working config?
c) Since the HTTPS client is used very seldom speed is not a major concern for us. Most important is a) to minimise code size and b) stack and/or heap usage. Any suggestions for the optimal config to meet the requirement for the 2 specific cipher suites, TLS1.2 only, client only?

All help appreciated.

mahavirj, 2024-07-03:
@ammaree 

>  Can we assume option descriptions (1st line of each one) is correct and text in additional info (2nd line) is incorrect?

Yes. I have raised internal fix to correct the additional info text.

For size optimization to mbedTLS library, please refer to this [guide](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/performance/size.html#mbedtls-features).

For reducing the heap usage, you may consider enabling dynamic buffer feature (supported with only TLS 1.2) as per this [guide](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/mbedtls.html#reducing-heap-usage).

closed 2024-07-04 by espressif-bot

AxelLin, 2024-10-06:
> For reducing the heap usage, you may consider enabling dynamic buffer feature (supported with only TLS 1.2) as per this [guide](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/protocols/mbedtls.html#reducing-heap-usage).

Hi @mahavirj 

Currently MBEDTLS_DYNAMIC_BUFFER feature is not supported with TLS 1.3 yet.
Do you have plan to support enabling dynamic buffer feature with TLS1.3?

The thing is diabling MBEDTLS_DYNAMIC_BUFFER takes too much memory (about 20K) for ESP32.
So due to the limited memory size on esp32,  it's almost not possible to support TLS1.3 for existing products using esp32.

mahavirj, 2024-10-16:
@AxelLin 

>  Do you have plan to support enabling dynamic buffer feature with TLS1.3?

Previously when we looked at this, it was not straight forward to add support for it in 3.6.x version. Ideally, this optimization should be part of the upstream stack itself, maintenance becomes easier that way. But, we will re-evaluate this feature request and see if we can optimize the memory footprint for TLS1.3 use-case. CC @AdityaHPatwardhan

AxelLin, 2024-11-20:
> @AxelLin
>
>
> Previously when we looked at this, it was not straight forward to add support for it in 3.6.x version. Ideally, this optimization should be part of the upstream stack itself, maintenance [...]

@mahavirj @AdityaHPatwardhan 

Just wondering if you have any update about this?
Also do you consider contribute your changes to upstream?

kayzen9, 2024-11-20:
ESP had tried .. .. https://github.com/Mbed-TLS/mbedtls/pull/3889.  i think we need to push memory topic in mbedtls community. it will be great addition- if you have some time to spare on this @AxelLin

AxelLin, 2024-11-21:
@kayzen9 

I'm not a expert in mbedtls, just a normal user.
Above PR was closed because MBEDTLS_SSL_VARIABLE_BUFFER_LENGTH serves the purpose of dynamic buffers which is the main goal of this PR.
The MBEDTLS_SSL_VARIABLE_BUFFER_LENGTH is already available in mbedtls.
So I'm confused if the issue is on espressif side or mbedtls?

BTW, in my test, with MBEDTLS_DYNAMIC_BUFFER=y it use less memory than MBEDTLS_SSL_VARIABLE_BUFFER_LENGTH=y.
But MBEDTLS_SSL_PROTO_TLS1_3 depends on !MBEDTLS_DYNAMIC_BUFFER.

AdityaHPatwardhan, 2024-11-21:
HI @AxelLin 
We are tracking the feature request internally and will soon update regarding dynamic buffers with TLS 1.3. No work has been done as of yet.

AxelLin, 2025-02-12:
> HI [@AxelLin](https://github.com/AxelLin) We are tracking the feature request internally and will soon update regarding dynamic buffers with TLS 1.3. No work has been done as of yet.

@AdityaHPatwardhan I already wait for almost 3 Months, but I don't find any update regarding this in recent esp-idf.