espressif/esp-idf #14128: Socket，UDP Cann't get dada

henry1960, 2024-07-03:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF 5.2

### Espressif SoC revision.

ESP32C6

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-C6-DevKitC-1

### Power Supply used.

USB

### What is the expected behavior?

I what to get a answer from espressif.

### What is the actual behavior?

A udp client  project was normally completed ,but it can NOT receive data,.

### Steps to reproduce.

After the App is started, server can get data form the Kit, bet server  get nothing from Kit.

nopnop2002, 2024-07-03:
There are four ways to specify UDP addresses:
・Limited broadcast address
・Directed broadcast address
・Multicast address
・Unicast address

Reception is not possible if the sender and receiver are different.

henry1960, 2024-07-03:
It's no matter about address, because server can get data from udp client, but udp client can NOT.

henry1960, 2024-07-03:
It's no matter about address, because data server can get data from udp client, but udp client can not get data from data server.
The Kit get IP address from DHCP server , the data server too. they connect to same router.

henry1960, 2024-07-03:
It's no matter about IP address, because data server can get data from udp client, but udp client can not get data from data server.
The Kit gets IP address from DHCP server , the data server too. they connect to same router.

nopnop2002, 2024-07-03:
https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/udp_client

https://github.com/espressif/esp-idf/tree/master/examples/protocols/sockets/udp_server

henry1960, 2024-07-03:
Thank you, will try it tomorrow.

henry1960, 2024-07-03:
Thank you,I'll test the samples tomorrow.

Xiehanxin, 2024-09-05:
hi @henry1960 any update about this issue?

closed 2024-12-23 by Alvin1Zhang