espressif/esp-idf #14134: Unable to export environment

MaplestoryAlen, 2024-07-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

len@lee:~/esp-idf$ source export.sh 
Detecting the Python interpreter
Checking "python3" ...
Python 3.8.10
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
Adding ESP-IDF tools to PATH...
Using a supported version of tool cmake found in PATH: 3.16.3.
However the recommended version is 3.24.0.
Checking if Python packages are up to date...
Constraint file: /home/alen/.espressif/espidf.constraints.v5.1.txt
Requirement files:
 - /home/alen/esp-idf/tools/requirements/requirements.core.txt
Python being checked: /home/alen/.espressif/python_env/idf5.1_py3.8_env/bin/python
Python requirements are satisfied.
Added the following directories to PATH:
  /home/alen/esp-idf/components/espcoredump
  /home/alen/esp-idf/components/partition_table
  /home/alen/esp-idf/components/app_update
Done! You can now compile ESP-IDF projects.
Go to the project directory and run:

  idf.py build

Cannot import module "esp_idf_monitor". This usually means that "idf.py" was not spawned within an ESP-IDF shell environment or the python virtual environment used by "idf.py" is corrupted.
Please use idf.py only in an ESP-IDF shell environment. If problem persists, please try to install ESP-IDF tools again as described in the Get Started guide.

dobairoland, 2024-07-04:
Hi @MaplestoryAlen. Please run the following commands after you've sourced `export.sh` and share the received outputs:
```bash
which idf.py
idf.py --version
which python
/home/alen/.espressif/python_env/idf5.1_py3.8_env/bin/python -m pip freeze
```

MaplestoryAlen, 2024-07-12:
> Hi @MaplestoryAlen. Please run the following commands after you've sourced `export.sh` and share the received outputs:
> 
> ```shell
> which idf.py
> idf.py --version
> which python
> /home/alen/.espressif/python_env/idf5.1_py3.8_env/bin/python -m pip freeze
> ```

Thank you very much for resolving the issue

closed 2024-07-12 by MaplestoryAlen

lilian-lilifox, 2024-09-09:
Hi @dobairoland. I had encountered the same problem, and I followed your instructions, getting results below:

```
╭─    ~/Documents/Embedded/esp/esp-idf   master ··························· ✔  02:49:48 PM 
╰─ which idf.py                                                                          (base) 
/home/lilian/Documents/Embedded/esp/esp-idf/tools/idf.py

╭─    ~/Documents/Embedded/esp/esp-idf   master ··························· ✔  02:50:10 PM 
╰─ idf.py --version                                                                      (base) 
Cannot import module "esp_idf_monitor". This usually means that "idf.py" was not spawned within an ESP-IDF shell environment or the python virtual environment used by "idf.py" is corrupted.
Please use idf.py only in an ESP-IDF shell environment. If problem persists, please try to install ESP-IDF tools again as described in the Get Started guide.

╭─    ~/Documents/Embedded/esp/esp-idf   master ························· ✘ 1  02:50:13 PM 
╰─ which python                                                                          (base) 
/home/lilian/.espressif/python_env/idf5.4_py3.12_env/bin/python

╭─    ~/Documents/Embedded/esp/esp-idf   master ··························· ✔  02:50:25 PM 
╰─ /home/lilian/.espressif/python_env/idf5.4_py3.12_env/bin/python -m pip freeze         (base) 
annotated-types==0.7.0
argcomplete==3.5.0
bitarray==2.9.2
bitstring==4.2.3
certifi==2024.8.30
cffi==1.17.1
charset-normalizer==3.3.2
click==8.1.7
colorama==0.4.6
construct==2.10.70
cryptography==43.0.1
ecdsa==0.19.0
esp-coredump==1.11.0
esp-idf-kconfig==2.3.0
esp-idf-monitor==1.4.0
esp-idf-panic-decoder==1.2.1
esp-idf-size==1.5.0
esp_idf_nvs_partition_gen==0.1.3
esptool==4.8.dev5
freertos-gdb==1.0.3
idf-component-manager==2.0.3
idna==3.8
intelhex==2.3.0
jsonref==1.1.0
markdown-it-py==3.0.0
mdurl==0.1.2
packaging==24.1
pyclang==0.5.0
pycparser==2.22
pydantic==2.9.0
pydantic-settings==2.4.0
pydantic_core==2.23.2
pyelftools==0.31
pygdbmi==0.11.0.0
Pygments==2.18.0
pyparsing==3.1.4
pyserial==3.5
python-dotenv==1.0.1
PyYAML==6.0.2
reedsolo==1.7.0
requests==2.32.3
requests-file==2.1.0
requests-toolbelt==1.0.0
rich==13.8.0
setuptools==71.0.0
six==1.16.0
tqdm==4.66.5
typing_extensions==4.12.2
tzdata==2024.1
urllib3==2.2.2
```

dobairoland, 2024-09-09:
@lilian-lilifox In your case the esp-idf-monitor package is not installed. You have to re-run the install script.

lilian-lilifox, 2024-09-09:
@dobairoland Thank you for your reply. It has had 'esp-idf-monitor==1.4.0' in pip freeze already as you can see it above. And I tried to reinstall and to downgrade it, but it didn't work.

dobairoland, 2024-09-09:
Yes, I'm sorry. I must be blind.

@peterdragun Do you have an idea why the monitor cannot be imported? I think the failure is coming from here: https://github.com/espressif/esp-idf/blob/master/tools/idf_py_actions/tools.py#L25

peterdragun, 2024-09-09:
I cannot think of any reason why the import would not work.

@lilian-lilifox Can you try running `python -m esp_idf_monitor` to see if the package can at least work by itself?

lilian-lilifox, 2024-09-09:
@dobairoland Sorry, its not a bug. It should be the matter of environment. 
```
╭─    ~/Documents/Embedded/esp/esp-idf   master ··························· ✔  03:55:00 PM 
╰─ python3 tools/idf.py --version                                                        (base) 
ESP-IDF v5.4-dev-2744-g59e1838270

╭─    ~/Documents/Embedded/esp/esp-idf   master ··························· ✔  03:55:08 PM 
╰─ idf.py --version                                                                      (base) 
Cannot import module "esp_idf_monitor". This usually means that "idf.py" was not spawned within an ESP-IDF shell environment or the python virtual environment used by "idf.py" is corrupted.
Please use idf.py only in an ESP-IDF shell environment. If problem persists, please try to install ESP-IDF tools again as described in the Get Started guide.
```
By calling `idf.py` directly, the system used a `python3` different from the one `esp-idf/export.fish` set. QwQ...

lilian-lilifox, 2024-09-09:
It seems shebang '#!/usr/bin/env python' at the begin of `esp-idf/tools/idf.py` doesn't work.