espressif/esp-idf #14135: esp_wifi_beacon_monitor_configure does not apply early wakeup parameter

dariocorna, 2024-07-04:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Linux

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-C6 MINI

### Power Supply used.

USB

### What is the expected behavior?

in wifi_init.c there is the possibility to change delta_intr_early (even if it is not in configuration).
there reason to change it is that on some router (ie fritzbox 7530) the power consumption is very high due to an high percentage of beacon loss.
This happens because it seems like beacon is drifted by a few microseconds in advance.
What i would reach with this modification it to accept an average higher consumption in ideal situation by stay alive a few milliseconds more for every beacon window, but at the same time grant an average consumption acceptable in all situations.

### What is the actual behavior?

delta_intr_early parameter is not applyed at all

### Steps to reproduce.

I'm using power_save example in wifi folder.

sdkconfig differences:
CONFIG_EXAMPLE_POWER_SAVE_MAX_MODEM=y 
CONFIG_ESP_WIFI_SLP_BEACON_LOST_OPT=y
CONFIG_ESP_WIFI_SLP_BEACON_LOST_TIMEOUT=25
CONFIG_ESP_WIFI_SLP_BEACON_LOST_THRESHOLD=0

 modfying wifi_init.c beacon monitor parameters manually in source code.

[image]

dariocorna, 2024-07-04:
Attached the same test on esp32-S2, that anticipate wakeup by 2ms by default and does not miss a beacon, this would be perfecly ok to me, even if the wakeup time is longer

[image]
[image]

esp-lis, 2024-07-09:
recv beacon window = default tbtt interrupt early time (builtin) + delta_intr_early * loss_counter + delta_timeout * loss_counter

loss_counter++ if beacon is lost and loss_counter < loss_threshold

CONFIG_ESP_WIFI_SLP_BEACON_LOST_THRESHOLD=0 means loss_counter is always 0

dariocorna, 2024-07-09:
ok this mean that with the following parameters and a tbtt of 204800 considering listening interval 2 I should get a window anticipated up to 192ms, instead is always 204 as you can see, there are also some other parameters nbt clears to me, like tsf_time_sync_deviation

`# define WIFI_BEACON_MONITOR_CONFIG_DEFAULT(ena)   { \
    .enable = (ena), \
    .loss_timeout = 50, \
    .loss_threshold = 5, \
    .delta_intr_early = 99, \
    .delta_loss_timeout = 0, \
    .beacon_abort = 1, \
    .broadcast_wakeup = 0, \
    .tsf_time_sync_deviation = 5, \
    .modem_state_consecutive = 10, \
    .rf_ctrl_wait_cycle = 20 \
}`

[image]

esp-lis, 2024-07-10:
> ok this mean that with the following parameters and a tbtt of 204800 considering listening interval 2 I should get a window anticipated up to 192ms, instead is always 204 as you can see, there are [...]

pls refer to [wifi_types.h#L44-L62](https://gitlab.espressif.cn:6688/espressif/esp-idf/-/blob/master/components/esp_wifi/include/esp_private/wifi_types.h?ref_type=heads#L44-L62).

>> ok this mean that with the following parameters and a tbtt of 204800 considering listening interval 2 I should get a window  anticipated up to 192ms, instead is always 204 as you can see

I can not understand these description, Can you add more details describe?

dariocorna, 2024-07-10:
> pls refer to [wifi_types.h#L44-L62](https://github.com/espressif/esp-idf/blob/master/components/esp_wifi/include/esp_private/wifi_types.h#L44).
> 

Comments are useful for sure, it would be better to me to understand how those parameter impact on beacon window, ie. tsf deviation is a percentage or a value, if value is ms ms^2 or us?

> > > ok this mean that with the following parameters and a tbtt of 204800 considering listening interval 2 I should get a window  anticipated up to 192ms, instead is always 204 as you can see
> 
> I can not understand these description, Can you add more details describe?

I mean whatever i set the beacoun window does not move, it is always on tbtt time (102400 ms * LISTEN_INTERVAL).
I'd just need to anticipate it a bit to obtain an average wake up time waiting for the beacon up to 5ms with a strong confidence to don't miss it. This works on esp32-S2 actually.

It seems like the algorithm works better when the router grant a drift on beacon window of less than 100us, it doesn't works when it is higher.

closed 2024-09-10 by Sherry616