espressif/esp-idf #14141: ESP32-S3-Mini-1-N8 module - Heap memory free space is decreasing during program execution (IEP-1289)

MN000022, 2024-07-05:
I am working on ESP32-S3-Mini-1-N8 module. I am using GATT server example. In this application,
1. BLE power save mode is implemented. External 32KHz crystal is used.
2. ESP32-S3 light sleep mode is implemented.
3. RTC GPIO wake up is implemented.
4. Timer periodic interrupts for 100us, 1ms & 10ms are implemented - GPtimer

Initialization for timer is as follows:
    	/* GPtimer 2 - 10ms */
    		gptimer2 = NULL;
    		QueueHandle_t Timer_queue2 = xQueueCreate(10, sizeof(example_queue_element_t));
    	     if (!Timer_queue2) {
    	         return;
    	     }
    	     gptimer_config_t timer_config2 = {
    	         .clk_src = GPTIMER_CLK_SRC_DEFAULT,
    	         .direction = GPTIMER_COUNT_UP,
    	         .resolution_hz = 1000000, // 1MHz, 1 tick=1us
    	     };
    	     ESP_ERROR_CHECK(gptimer_new_timer(&timer_config2, &gptimer2));

    	     gptimer_event_callbacks_t cbs2 = {
    	         .on_alarm = example_timer_on_alarm_cb_v2,
    	     };
    	     cbs2.on_alarm = example_timer_on_alarm_cb_v2;
    	     ESP_ERROR_CHECK(gptimer_register_event_callbacks(gptimer2, &cbs2, Timer_queue2));
    	     ESP_ERROR_CHECK(gptimer_enable(gptimer2));

    	     gptimer_alarm_config_t alarm_config2 = {
    	         .alarm_count = 10000,//1000000, // period = 10ms
    	     };
    	     ESP_ERROR_CHECK(gptimer_set_alarm_action(gptimer2, &alarm_config2));
    	     ESP_ERROR_CHECK(gptimer_start(gptimer2));

The issue I observed is, heap memory free space is getting decreased during application execution and when heap memory is around 500 bytes, ESP32-S3 goes in to unidentified state where BLE and GPIO wake up are also not working. Only it starts working once power is reset.
I checked free heap memory using function 'xPortGetFreeHeapSize' during program execution.
I have enabled task watchdog and panic handler invoked on task watchdog timeout.
I have also checked without initializing BLE. I am facing same issue.
Can anyone elaborate on this issue?

Thanks

suda-morris, 2024-10-23:
sounds like your application is leaking memory, maybe you allocated some memory in the gptimer callback and never free them? @MN000022

Alvin1Zhang, 2024-11-06:
Thanks for reporting, will close due short of feedback, feel free to reopen with more updates.

closed 2024-11-06 by Alvin1Zhang