espressif/esp-idf #14143: Support of UVC as USB device

ohault, 2024-07-05:
### Is your feature request related to a problem?

Lack of UVC in https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/device

### Describe the solution you'd like.

For boards like ESP32-S3-CAM, with an implementation of UVC, these boards could be seen on an USB host as an USB Webcam device.

roma-jam, 2024-07-05:
Hi @ohault ,

Thanks for the feature request. 
For the current quarter features were already planned,  so we could start considering to implement new features not early than September.

Meanwhile, the esp-idf USB device examples are based on the TinyUSB library, which has couple of examples, which could be helpful. Here is the one, which implements video capture ([link](https://github.com/hathach/tinyusb/tree/master/examples/device/video_capture))

igrr, 2024-07-05:
There is also an example in esp-iot-solution repo which you can try: https://github.com/espressif/esp-iot-solution/tree/master/examples/usb/device/usb_webcam

ohault, 2024-07-05:
@igrr thank you but when I'm trying set-target to esp32s3, I get this error : 

CMake Error at C:/Espressif/frameworks/esp-idf-v5.2.2-2/tools/cmake/build.cmake:544 (message):
  ERROR: Because no versions of usb_device_uvc match >1.1.0,<1.2.0

   and usb_device_uvc (1.1.0) depends on espressif/tinyusb (>=0.15.0~10), usb_device_uvc (>=1.1.0,<1.2.0) requires espressif/tinyusb (>=0.15.0~10).

  So, because no versions of espressif/tinyusb match >=0.15.0~10

   and project depends on usb_device_uvc (1.1.*), version solving failed.

roma-jam, 2024-07-05:
@ohault , 
Could you share the full output of the command `idf.py set-target esp32s3` for the example from `esp-iot-solution` mentioned above?

ohault, 2024-07-05:
```
C:\esp-iot-solution\examples\usb\device\usb_webcam>idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Directory 'C:\esp-iot-solution\examples\usb\device\usb_webcam\build' doesn't seem to be a CMake build directory. Refusing to automatically delete files in this directory. Delete the directory manually to 'clean' it.

C:\esp-iot-solution\examples\usb\device\usb_webcam>idf.py add-dependency "espressif/tinyusb^0.15.0~10"
Executing action: add-dependency
ERROR: Dependency "espressif/tinyusb" already exists for in manifest "C:\esp-iot-solution\examples\usb\device\usb_webcam\main\idf_component.yml"

C:\esp-iot-solution\examples\usb\device\usb_webcam>idf.py set-target esp32s3
Adding "set-target"'s dependency "fullclean" to list of commands with default set of options.
Executing action: fullclean
Build directory 'C:\esp-iot-solution\examples\usb\device\usb_webcam\build' not found. Nothing to clean.
Executing action: set-target
Set Target to: esp32s3, new sdkconfig will be created.
Running cmake in directory C:\esp-iot-solution\examples\usb\device\usb_webcam\build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=C:\Espressif\python_env\idf5.2_py3.11_env\Scripts\python.exe -DESP_PLATFORM=1 -DIDF_TARGET=esp32s3 -DCCACHE_ENABLE=1 C:\esp-iot-solution\examples\usb\device\usb_webcam"...
-- Found Git: C:/Espressif/tools/idf-git/2.44.0/cmd/git.exe (found version "2.44.0.windows.1")
-- sdkconfig not exist
EYES_SHOW_TARGET is set to esp32s3
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc.exe - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++.exe - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Dependencies lock doesn't exist, solving dependencies.
Using component placed at C:\esp-iot-solution\examples\usb\device\usb_webcam\eyes_show for dependency eyes_show(*), specified in C:/esp-iot-solution/examples/usb/device/usb_webcam/main\idf_component.yml
Using component placed at C:\esp-iot-solution\components\usb\usb_device_uvc for dependency usb_device_uvc(1.1.*), specified in C:/esp-iot-solution/examples/usb/device/usb_webcam/main\idf_component.yml
....................Using component placed at C:\esp-iot-solution\examples\usb\device\usb_webcam\eyes_show for dependency eyes_show(*), specified in C:/esp-iot-solution/examples/usb/device/usb_webcam/main\idf_component.yml
Using component placed at C:\esp-iot-solution\components\usb\usb_device_uvc for dependency usb_device_uvc(1.1.*), specified in C:/esp-iot-solution/examples/usb/device/usb_webcam/main\idf_component.yml
CMake Error at C:/Espressif/frameworks/esp-idf-v5.2.2-2/tools/cmake/build.cmake:544 (message):
  ERROR: Because no versions of usb_device_uvc match >1.1.0,<1.2.0

   and usb_device_uvc (1.1.0) depends on espressif/tinyusb (>=0.15.0~10), usb_device_uvc (>=1.1.0,<1.2.0) requires espressif/tinyusb (>=0.15.0~10).

  So, because no versions of espressif/tinyusb match >=0.15.0~10

   and project depends on usb_device_uvc (1.1.*), version solving failed.

Call Stack (most recent call first):
  C:/Espressif/frameworks/esp-idf-v5.2.2-2/tools/cmake/project.cmake:605 (idf_build_process)
  CMakeLists.txt:25 (project)

....................-- Configuring incomplete, errors occurred!
See also "C:/esp-iot-solution/examples/usb/device/usb_webcam/build/CMakeFiles/CMakeOutput.log".
cmake failed with exit code 1, output of the command is in the C:\esp-iot-solution\examples\usb\device\usb_webcam\build\log\idf_py_stderr_output_16640 and C:\esp-iot-solution\examples\usb\device\usb_webcam\build\log\idf_py_stdout_output_16640

C:\esp-iot-solution\examples\usb\device\usb_webcam>
```

chipweinberger, 2024-07-05:
https://components.espressif.com/components/espressif/usb_host_uvc/versions/1.0.4

found this

ohault, 2024-07-06:
> https://components.espressif.com/components/espressif/usb_host_uvc/versions/1.0.4
> 
> 
> 
> found this

This components would be required to support a USB webcam on the ESP board (USB Host). I'm looking the opposite, by enabling an ESP board like ESP32-S3-CAM to become an USB webcam(UVC) device

roma-jam, 2024-07-08:
Hi @ohault, 

seems like some issue with building process and solving dependencies. 
Could you delete the `/build` folder manually and repeat the process? 

Also, there is no necessity to add the dependency manually, there is already present the tinyusb component in manifest with correct version. 

May I ask you to run the sequence again and share the output: `install` -> `export` -> `idf.py set-target  esp32s3` and `idf.py build`?

ohault, 2024-07-08:
> Hi @ohault,
>
> seems like some issue with building process and solving dependencies. Could you delete the `/build` folder manually and repeat the process?
>
> Also, there is no necessity to add the [...]

I deleted the build folder and `idf.py set-target esp32s3` gives me the same exact error/output.

roma-jam, 2024-07-08:
@ohault ,
You could try to remove `dependencies.lock` from the usb_webcam example to resolve dependencies again while run i`df.py set-target esp32s3.

roma-jam, 2024-07-08:
@ohault , 

_Last but not least there is also this directory in the usb_dongle example project C:\esp-iot-solution\examples\usb\device\usb_dongle\managed_components\espressif__tinyusb_

Yes, that is how the component manager works. When the dependencies are solved, the correct version of the component downloaded to the `managed_components` folder of the example or project folder. 

For `usb_webcam` the process should be absolutely the same, as for `usb_dongle`, the difference is only in the amount of the components.

ohault, 2024-07-08:
> @ohault , You could try to remove `dependencies.lock` from the `usb_webcam` example to resolve dependencies again while run `idf.py set-target esp32s3`.

I can see a generated dependencies.lock in `usb_dongle` but it is not the case for `usb_webcam`

ohault, 2024-07-08:
> For `usb_webcam` the process should be absolutely the same, as for `usb_dongle`, the difference is only in the amount of the components.

That's why I have just tried `usb_dongle` and it is working, but I don't know why `usb_webcam`  is not working.

roma-jam, 2024-07-08:
I see, that there is the one commit, related to the version parsing (3 weeks ago): https://github.com/espressif/esp-iot-solution/commit/1172c0066986e8a2453374b12a7629906ac68f2d

Just in case, are you using the latest master of `esp-iot-solution`? (`git checkout master` and `git pull`)

ohault, 2024-07-08:
@roma-jam, by July 5th, 2024 I have just done a `git clone https://github.com/espressif/esp-iot-solution.git`

leeebo, 2024-07-09:
The issue can only be reproduced on Windows OS with `idf-component-manager~=1.2`. The IDF component manager team is working on the fix.

The current workaround is https://github.com/espressif/esp-iot-solution/issues/383#issuecomment-2217162424

tore-espressif, 2024-12-02:
The dependency issues were solved, the USB UVC device feature is further developed in [usb_device_uvc](https://components.espressif.com/components/espressif/usb_device_uvc) component.

I'll close this issue, feel free to reopen if there are more questions

closed 2024-12-02 by tore-espressif