espressif/esp-idf #14146: Esp32 cam and I2c_lcd SCL and SDA pin

IenSignif, 2024-07-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi, I'm creating a project with shrinked opencv with I2c lcd. I use opencv Hough circle to count circle and display it on my I2c lcd, everything was going fine and finished the code. The problem is when i run the flashed device it shows an error of :

Btw I'm using the version ESP-IDF v4.4.8 because it's compatible with the opencv library file
Other info about my device:
ESP32-CAM (HW-818)
I2c LCD 2 x 16 0x27

`[0;32mI (774) cpu_start: Starting scheduler on PRO CPU.[0m
[0;32mI (0) cpu_start: Starting scheduler on APP CPU.[0m
[0;32mI (778) LCD: I2C initialized successfully[0m

[0;31mE (1828) i2c: i2c_set_pin(882): scl and sda gpio numbers are the same[0m
[0;32mI (1828) LCD: Error is sending command[0m

[0;31mE (2828) i2c: i2c_set_pin(882): scl and sda gpio numbers are the same[0m
[0;32mI (2828) LCD: Error is sending command[0m

[0;31mE (3828) i2c: i2c_set_pin(882): scl and sda gpio numbers are the same[0m
[0;32mI (3828) LCD: Error is sending command[0m`

even though my config in my main.cpp is correct which is :

`static esp_err_t i2c_master_init(void)
{
    int i2c_master_port = I2C_NUM_0;

    i2c_config_t conf = {
        .mode = I2C_MODE_MASTER,
        .sda_io_num = GPIO_NUM_13,
        .scl_io_num = GPIO_NUM_16,
        .sda_pullup_en = GPIO_PULLUP_ENABLE,
        .scl_pullup_en = GPIO_PULLUP_ENABLE,
        .master.clk_speed = 100000,
    };

    i2c_param_config(i2c_master_port, &conf);

    return i2c_driver_install(i2c_master_port, conf.mode, 0, 0, 0);
}`

I'm almost finish on my project, or so I thought. can someone help me, or any recommendations?

IenSignif, 2024-07-07:
I figured it out, i used an old I2c library which works fine but another problem came when I'm running the flash device:

`[0;33mW (968) LCD: I2C bus is busy, waiting...[0m
[0;33mW (968) LCD: I2C bus is busy, waiting...[0m
[0;33mW (978) LCD: I2C bus is busy, waiting...[0m
[0;33mW (978) LCD: I2C bus is busy, waiting...[0m
[0;33mW (978) LCD: I2C bus is busy, waiting...[0m
[0;33mW (988) LCD: I2C bus is busy, waiting...[0m
[0;33mW (988) LCD: I2C bus is busy, waiting...[0m
[0;33mW (988) LCD: I2C bus is busy, waiting...[0m
[0;33mW (998) LCD: I2C bus is busy, waiting...[0m
[0;33mW (998) LCD: I2C bus is busy, waiting...[0m
ESP_ERROR_CHECK failed: esp_err_t 0xffffffff (ESP_FAIL) at 0x40087700
file: "./main/main.cpp" line 90
func: void LCD_writeNibble(uint8_t, uint8_t)
expression: i2c_master_cmd_begin(I2C_NUM_0, cmd, 1000/portTICK_PERIOD_MS)

abort() was called at PC 0x40087703 on core 0`

and the backtrace ( I use addr2line to identify the backtraces ) :

esp/v4.4.8/esp-idf/components/esp_system/esp_err.c:42
esp/v4.4.8/esp-idf/components/esp_system/esp_err.c:43
esp/v4.4.8/esp-idf/components/esp_system/panic.c:408
esp/v4.4.8/esp-idf/components/esp_system/esp_system.c:137
esp/v4.4.8/esp-idf/components/newlib/abort.c:46
esp/v4.4.8/esp-idf/components/esp_system/esp_err.c:43
main/main.cpp:90 (discriminator 1)
main/main.cpp:173
main/main.cpp:199
esp/v4.4.8/esp-idf/components/freertos/port/port_common.c:141 (discriminator 2)
esp/v4.4.8/esp-idf/components/freertos/port/xtensa/port.c:142

IenSignif, 2024-07-07:
I fixed it, I just mistakenly put the SCL in a different pin,

IenSignif, 2024-07-07:
everythings fine now, now I'm stuck with this 

`[0;32mI (2553) camera: Allocating 1 frame buffers (3750 KB total)[0m
[0;32mI (3063) camera: Allocating 3750 KB frame buffer in OnBoard RAM[0m
[0;31mE (3573) gpio: esp_intr_alloc failed (0x105)[0m
[0;31mE (3573) camera: gpio_install_isr_service failed (105)[0m
[0;31mE (3573) camera: Camera init failed with error 0x105[0m
[0;31mE (3583) LCD: Camera init failed with error 0x105[0m`

suda-morris, 2024-07-08:
failed in GPIO interupt allocation, error code 0x105 means `#define ESP_ERR_NOT_FOUND           0x105   /*!< Requested resource not found */`

So check how your camera driver calls the `gpio_install_isr_service` function. Can the camera driver initialize on another CPU core?