espressif/esp-idf #14148: VolToPart use const in fatfs.

qwx, 2024-07-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

When I want to use two partition in one emmc, I have to change the code in diskio.c like this.
```
const PARTITION VolToPart[FF_VOLUMES] = {
    {0, 0},    /* Logical drive 0 ==> Physical drive 0, auto detection */
#if FF_VOLUMES > 1
    {1, 0},    /* Logical drive 1 ==> Physical drive 1, auto detection */
#endif
```
change to:
```
const PARTITION VolToPart[FF_VOLUMES] = {
    {0, 1},    /* Logical drive 0 ==> Physical drive 0, Partition 1 */
#if FF_VOLUMES > 1
    {0, 2},    /* Logical drive 1 ==> Physical drive 0, Partition 2 */
#endif
```
Everyone coding with me have to change them too. 
So is there a way not change the code in esp-idf but in user code?
Like make VolToPart not const or other beautiful way to do this?

adokitkat, 2024-07-15:
Yes, this is something which can be improved. Right now creating more partitions on the same storage device is not trivial. Will look into this. Thank you for your comment.

qwx, 2024-07-15:
Thank you for your reply. I use the code below to create more than one partition on emmc:
```
    BYTE work[FF_MAX_SS];         /* Working buffer */
    LBA_t plist[] = {90, 10, 0}; 

    FRESULT fres = f_fdisk(0, plist, work);            /* Divide the physical drive 0 */
    if (fres != FR_OK)
    {
        printf("fdisk error");
    }
    fres = f_mkfs("0:", 0, work, sizeof work); /* Create FAT volume on the logical drive 0 */
    if (fres != FR_OK)
    {
        printf("fdisk1 error");
    }
    fres = f_mkfs("1:", 0, work, sizeof work); /* Create FAT volume on the logical drive 1 */
    if (fres != FR_OK)
    {
        printf("fdisk2 error");
    }
```
I also create a pull request(#14150)  to do this change. I can provide infomation if you need. Thanks a lot.

closed 2025-04-20 by espressif-bot