espressif/esp-idf #14152: [ESP32S3] Reading from a SPIFFS filesystem causes interrupt jitter

higaski, 2024-07-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have noticed that reading from a SPIFFS filesystem via read or fread system calls introduces noticeable jitter to interrupts, even if those interrupts have been given the highest user priority (level 3) and are running from IRAM.

Given the architecture diagrams from the ESP32-S3 reference manuals, can someone explain to me why this happens
[image]

[image]

igrr, 2024-07-10:
This probably is related with the fact that code involved in performing Flash operations occasionally uses critical sections (portENTER/EXIT_CRITICAL), which disable interrupts up to level 3. The code inside those critical sections is relatively short, but it is still enough to cause a few microseconds of interrupt jitter.

higaski, 2024-07-10:
> This probably is related with the fact that code involved in performing Flash operations occasionally uses critical sections (portENTER/EXIT_CRITICAL), which disable interrupts up to level 3. The [...]

Any chance we could get a user level interrupt which does not get disabled?

There technically is no reason for my timer interrupt to get disabled, yet I'm paying the price.

igrr, 2024-07-10:
If the real ISR in your application is as simple as in the example project and doesn't call any RTOS functions, you can try to rewrite it in assembly and register it as a [high-priority interrupt](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/hlinterrupts.html) (level 5). These interrupts do not get disabled in RTOS critical sections and could be used for simple tasks, like toggling a GPIO.

You could also try making the ISR registration call from CPU 1, in which case the interrupt will be serviced there. This won't  completely eliminate jitter since RTOS critical sections are still occasionally used on CPU1, but might reduce jitter to an acceptable level.

higaski, 2024-07-10:
Sadly it's not that simple no. My real application calls functions from the GPTimer und UART driver and sends a task notification...

```cpp
// RMT interrupt, start timer
bool IRAM_ATTR rmt_callback(rmt_channel_handle_t,
                            rmt_tx_done_event_data_t const*,
                            void*) {
  gptimer_set_raw_count(gptimer, 0ull);
  gptimer_alarm_config_t const alarm_config{
    .alarm_count = static_cast<decltype(gptimer_alarm_config_t::alarm_count)>(
      TCSMin + offsets.tcs)};
  gptimer_set_alarm_action(gptimer, &alarm_config);
  return pdFALSE;
}

// Timer interrupt, disable outputs, read available UART bytes
bool IRAM_ATTR gptimer_callback(gptimer_handle_t timer,
                                gptimer_alarm_event_data_t const* edata,
                                void*) {
  static bool ch1{};
  bool const ch2{true};
  BaseType_t high_task_awoken{pdFALSE};

  // TCS
  if (edata->alarm_value < TTS1) {
    gptimer_set_raw_count(timer, TCSMin);

    // Pull tracks low
    gpio_set_level(bidi_en_gpio_num, 1u);

    // Reset alarm to TS2
    gptimer_alarm_config_t const alarm_config{.alarm_count = TTC1};
    gptimer_set_alarm_action(timer, &alarm_config);
  }
  // TS2
  else if (edata->alarm_value < TTS2) {
    // Reset alarm to TCE
    gptimer_alarm_config_t const alarm_config{.alarm_count = TCEMin};
    gptimer_set_alarm_action(timer, &alarm_config);

    // Check whether there has been data in channel 1
    ch1 = uart_ll_get_rxfifo_len(&UART1);
  }
  // TCE
  else {
    // Release tracks
    gpio_set_level(bidi_en_gpio_num, 0u);

    // Send a notification to the task
    xTaskNotifyIndexedFromISR(task.handle,
                              default_notify_index,
                              (ch2 << 1u) | ch1,
                              eSetValueWithOverwrite,
                              &high_task_awoken);

    gptimer_set_alarm_action(gptimer, NULL);
  }

  return high_task_awoken == pdTRUE;
}
```

igrr, 2024-07-10:
Could you try enabling CONFIG_SPI_FLASH_AUTO_SUSPEND and see if that improves the situation? It depends on the flash chip supporting auto-suspend, but if it does, flash reading process is simplified and there will probably be less critical sections involved.

> Another way to work around this issue would be to give users the ability to specify the priority level which should be disabled during flash reads, at least between >3 and >=3).

I'm afraid there are could be some risks with such a change. If the developer changes this setting during development and forgets about it, they could later run into race conditions which are hard to debug. It's not easy to check automatically that the ISR is not accessing anything that is not locked by a critical section with a matching interrupt masking level. Besides, currently all our critical sections disable interrupts up to the same level, so we would need to introduce a new critical section API with a configurable level. We'll discuss this internally and see what could be done.

higaski, 2024-07-10:
> Could you try enabling CONFIG_SPI_FLASH_AUTO_SUSPEND and see if that improves the situation? It depends on the flash chip supporting auto-suspend, but if it does, flash reading process is simplified and there will probably be less critical sections involved.

CONFIG_SPI_FLASH_AUTO_SUSPEND doesn't seem to be supported on the WROOM2 I'm using:
`E (1153) chip_generic: The flash you use doesn't support auto suspend, only 'XMC' is supported`

> I'm afraid there are could be some risks with such a change. If the developer changes this setting during development and forgets about it, they could later run into race conditions which are hard [...]

I can see how such a change might now be very impactful to the codebase, but I disagree that it is harder to debug than any of the peculiarities which already exist anyhow. My favorite example is still the **single** `constexpr variable` which gets accessed inside an IRAM function and causes some kind of cache panic... I think I myself opened multiple issues on such cases over the past few months.

In my opinion, such a feature could help the ESP32 to shed its image of not being "real-time" capable, at least a little. So far whenever I got asked about interrupt jitter I told people that they haven't setup their cores and peripherals correctly yet which admittedly is **very very hard**. But at least it's possible. Getting a peripherals interrupt disabled because of a critical section which would technically not be necessary though is something I currently can't change.

igrr, 2024-07-10:
> that it is harder to debug than any of the peculiarities which already exist anyhow. My favorite example is still the single constexpr variable which gets accessed inside an IRAM function and causes some kind of cache panic

If an issue leads to a panic, and the panic points to the function which needs to be changed, this is more straightforward to debug than if you accidentally corrupt RTOS data structures by not locking the interrupts and then (maybe indirectly) calling an RTOS function from an interrupt handler. In the latter case, the crash will not directly point to the problematic function, and will most likely result in a bug report similar to "IDF FreeRTOS crashes" — which IDF developers will then need to reproduce and debug.

I do agree that having the ISRs less affected by the system would be a nice improvement, we'll discuss this and see what can be done.

higaski, 2024-07-10:
Fair point. I'm looking forward to your reply.

sudeep-mohanty, 2024-08-09:
Hi @higaski,
Thank you for providing a very useful example to reproduce the problem. From our initial investigation, it looks like most of the jitter is caused by the `fopen()` and `fclose()` calls during the SPIFFS read operation. For a test, if your application allows it, would it be possible for you to maintain the file descriptor outside of your read loop so that you don't have to open/close it each time? Could you then assess its impact on the interrupt jitter you observe in your application? Thanks!

higaski, 2024-08-09:
> Hi @higaski, Thank you for providing a very useful example to reproduce the problem. From our initial investigation, it looks like most of the jitter is caused by the `fopen()` and `fclose()` calls [...]

My application would require more than just a single file though. I'm not terrible familiar with C file handlers, but I assume you'd need one per file.

sudeep-mohanty, 2024-08-10:
@higaski What I meant was that, you could open the file once with `fopen()`, perform all your operations on the file, and then close the file with one `fclose()`. Irrespective of the number of file descriptors you need, opening and closing files multiple times can result in increased interrupt jitter. I'd like to know, if indeed reducing the number of `fopen()` and `fclose()` operations reduces the jitter you observe. That could help us narrow down to an optimal solution.

higaski, 2024-08-10:
> @higaski What I meant was that, you could open the file once with `fopen()`, perform all your operations on the file, and then close the file with one `fclose()`. Irrespective of the number of file [...]

I've changed my example to use a global file descriptor which opens the file just once and never closes it again:
```cpp
// Read .png file from SPIFFS
void read_file() {
  static std::array<uint8_t, 1024uz> file_buffer;
  if (fd == NULL) {
    fd = fopen("/logo.png", "r");
    if (fd == NULL) {
      ESP_LOGE(TAG, "Failed to open file for reading");
      return;
    }
  }
  rewind(fd);
  size_t chunksize;
  do {
    if (!(chunksize = read(fileno(fd), data(file_buffer), size(file_buffer))))
      continue;
  } while (chunksize);
}
```

Estimated jitter looks about the same as before.

sudeep-mohanty, 2024-08-10:
Thanks for trying out @higaski. We shall continue the investigation and keep you posted.

habazut, 2024-10-17:
> If the real ISR in your application is as simple as in the example project and doesn't call any RTOS functions, you can try to rewrite it in assembly and register it as a [high-priority interrupt](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/hlinterrupts.html) (level 4). These interrupts do not get disabled in RTOS critical sections and could be used for simple tasks, like toggling a GPIO.

To be useful, some examples which show how to use high level interrupts would be very good to have. I bet most programmers do  not have any clue about the sliding register window architecture and how to handle that in an interrupt context and thus do not know how to call their code (probably a function written in C code) from assembly. So I think you should consider to make it possible to use high level interrupts even for "mere C coders". 

Regards,
Harald.

sudeep-mohanty, 2024-10-18:
> To be useful, some examples which show how to use high level interrupts would be very good to have. 

You could have a look at the [NMI interrupt example](https://github.com/espressif/esp-idf/tree/a97a7b09/examples/system/nmi_isr).

higaski, 2024-10-20:
This example doesn't call a C function though.

higaski, 2026-02-14:
I just realized that BT has a way to do "high level interrupts in C" using the `CONFIG_BTDM_CTRL_HLI` option. Any chance to make this a more general utility for ESP-IDF users?

sudeep-mohanty, 2026-02-16:
Hi @higaski,
Apologies for not following up on your previous query. ESP-IDF already supports high priority interrupts in a generic manner. Please have a look at the reference [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-guides/hlinterrupts.html). Unfortunately, there is no other update on this issue and we haven't had the time to investigate it further. However, we do have it in the plans and will get to the issue asap.

higaski, 2026-09-29:
Again revisiting this. Any updates?

Otherwise I might throw some money at an LLM to write me the necessary assembly to trigger a timer and read the UART FIFO threshold.

sudeep-mohanty, 2026-10-05:
Hello @higaski,
Apologies for the late reply but unfortunately, the state is the same. This hasn't been worked up on in a while but we will pick it up again immediately. Thank you for the prompt!

sudeep-mohanty, 2026-10-19:
Hello @higaski,
Sorry for the long silence on this one. We spent some time looking at the jitter in your example and wanted to share what we found.

## Observations

- We reproduced it on an `esp32s3` with your example (`higaski/esp_idf_issue_14152`). While `read_file()` runs, the interval between edges varies from 52.6 µs to 63.8 µs. This is true for the latest `master` branch of `esp-idf`.
- Your GPTimer interrupt is IRAM-safe, so it is not disabled while the cache is off and it keeps running during each flash operation. The jitter comes from the short critical sections in the code that starts and ends each flash operation:

  - taking and giving the flash operation lock
  - the IPC call that stops the other core
  - masking and unmasking the non-IRAM interrupts
  - suspending and resuming the cache

- Since the critical sections mask all level 3 interrupts, any one of them that becomes pending inside one of the critical sections waits until it ends, which in our measurements was up to 6–7 µs. SPIFFS reads flash in 256-byte pages, so one 1 KB read() performs a dozen or more flash operations, each with several of these critical sections.

- We profiled these paths and found no single slow spot that would remove the jitter. It is the total cost of turning the cache off and on through critical sections, and it is needed for flash access to be safe.

## Suggestions
- There is no obvious optimizations on the platform that we could do. So the question is if it would be possible to perhaps modify the architecture of your application to avoid reading the image from flash all the time?
    - Could you perhaps read once at boot time into a PSRAM buffer?
    - Or perhaps embed the image in the app partition via `EMBED_FILES`. This allows the image to be in the `.rodata` part of the app, so the CPU reads it from flash through the cache like any other constant data. Note that this makes the app image larger by the size of the file.

- If the timing must stay exact while flash is being accessed, the remaining option is to use highlevel (level 4 or 5) interrupts which do not get blocked by a critical section. Previously you did mention that you'd be open to trying a solution where such high level interrupts are writable in C for Xtensa targets. But as you may know, such interrupts would come with restrictions such as not being able to use FreeRTOS APIs and most other IDF peripheral driver APIs which take spinlocks. So, before heading down this direction, may I ask if it would work for your usecase?

higaski, 2026-10-19:
@sudeep-mohanty 
Thank you for the reply!

Your suggestion here 

> * Or perhaps embed the image in the app partition via `EMBED_FILES`. This allows the image to be in the `.rodata` part of the app, so the CPU reads it from flash through the cache like any other constant data. Note that this makes the app image larger by the size of the file.

is pretty much exactly what I’m currently doing. I basically package an entire Flutter web app into the `.rodata` section. But let me tell you, it's [not pretty](https://github.com/OpenRemise/Firmware/blob/dacd71ea4b133a659191b4e3d346d8e6770b69d6/CMakeLists.txt#L83-L126) 😆 

That eliminated the jitter caused by SPIFFS, but the underlying problem still persists. Imho users still want high-level interrupts. An example of why I personally need them can be found in another issue I opened: [ESP32S3 - Continuous RMT transmissions (IDFGH-11922)](https://github.com/espressif/esp-idf/issues/13003)
Several other users have also posted there who would likely benefit from high-level interrupts.

Not being able to use the FreeRTOS and ESP-IDF peripheral APIs is, of course, a "footgun", but I don’t think it’s any more or less so than with other microcontroller architectures or frameworks. 

And what would still remain usable in pure C is the entire `*_ll*` API. The difference between writing
```c
timer_ll_set_counter(0x42);
my_high_prio_irq_flag = true;
```

or
```asm
// Pseudo ASM
mov r1, 0x42
str CRYPTIC_REG, r1
ldr r1, &my_high_prio_irq_flag
mov r2, 1
str r1, r2
```

is huge.