espressif/esp-idf #14153: Nimble whitelist not working

Steven35700, 2024-07-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I started from the [bleprph example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/bleprph), I activated the `CONFIG_EXAMPLE_EXTENDED_ADV` flag over menuconfig and I added a line to activate the filter on the whitelist: `params.filter_policy = BLE_HCI_SCAN_FILT_USE_WL;` in the `ext_bleprph_advertise(void)` function.

After building and flashing the example on an ESP32S3, when I scan Bluetooth devices over NRFConnect, I still see the advertising and I can connect and interact with the device even though my phone is not in the whitelist.

I thought the problem might come from an empty whitelist. So I added a random address to the whitelist, but the problem persists.

There is no example dealing with this whitelist use case and the nimble documentation is not very complete on this subject.

I absolutely don't feel like this is normal, am I making a mistake ?

rahult-github, 2024-07-08:
Hi @Steven35700 , 

 Please try BLE_HCI_SCAN_FILT_USE_WL_INITA

Steven35700, 2024-07-08:
Hi @rahult-github 

Using `BLE_HCI_SCAN_FILT_USE_WL_INITA` doesn't work either

However, I just tried without using extended advertising I can get the whitelist work !
But I use extended advertising to be able to have multi advertising

rahult-github, 2025-02-03:
Hi @Steven35700 , 

Sorry for late reply . When the value BLE_HCI_SCAN_FILT_USE_WL_INITA is set, remote can still see the device. However, esp32 s3 will reject the incoming scan / connection request if the remote device is NOT in the whitelist. 

I tested s3 now for both extended adv and legacy and i see it works correctly. May i know what exactly is not working at your end ?. Please share console log and application code for reference.

finger563, 2025-02-13:
I'm having a possibly related issue where the remote device is able to connect, but the functionality of the peripheral is broken - e.g. the peripheral is a BLE HID input device and it sets the whitelist so that only a specific central can connect. The central does connect just fine, but does not properly discover the services and such. For the same device with the same bond, advertising without the whitelist address allows the central to connect and the HID device works just fine.

finger563, 2025-02-14:
I'm also slightly confused - you're talking about using the `BLE_HCI_SCAN_FILT_*` values, which (as I understand) apply to a scan and its parameters. I'm currently using `BLE_HCI_ADV_FILT_*` to try to configure the peripheral to reject remote device connections which are not in the white list - is that not the correct way to do it?

finger563, 2025-02-21:
@rahult-github thanks for that confirmation.

Is whitelist fully supported and expected to work for NimBLE host on ESP32-s3? I'm still seeing non-whitelisted devices connecting.

rahult-github, 2025-02-23:
Hi @finger563 ,  yes whitelist feature is fully supported. May i request you  to share with us your sample application (based on any IDF eample) and steps you are performing that we can try to replicate and check more ?

rahult-github, 2025-03-26:
Closing issue. Feel free to reopen with updates, if any.