espressif/esp-idf #14155: Improve UART VFS support for binary streams

AguileraG, 2024-07-09:
### Is your feature request related to a problem?

Both `uart_read()` and `uart_write()` treat end-of-line characters as special characters. Most of this specific behaviour can be disabled by configuring the VFS driver with the `ESP_LINE_ENDINGS_LF` line ending mode.

However, `uart_read()` always stops reading data when a `'\n'` character is received, even though more bytes may be available on the RX buffer (see [line 285](https://github.com/espressif/esp-idf/blob/5ca9f2a49aaabbfaf726da1cc3597c0edb3c4d37/components/esp_driver_uart/src/uart_vfs.c#L285)). 

This behaviour cannot be disabled in any way, and it makes it more difficult to work with binary protocols which treat `'\n'` as a regular byte.

### Describe the solution you'd like.

I would like for the UART VFS driver to be configurable so that all incoming characters are treated as regular bytes, so that when `uart_read()` is called, all data available in the RX buffer is read.

### Describe alternatives you've considered.

Right now, this issue can be worked around by calling `read()` and therefore `uart_read()` in a loop until the expected number of bytes is received. 

However, this approach makes it more difficult to perform non-blocking reads with `select()` or `poll()`, as the application must keep track of how long each call to `read()` takes.

igrr, 2024-07-09:
Thanks for raising this issue! Incidentally, we got another bug report about UART VFS reading behavior, and after fixing that other issue, early return on a newline character can actually be removed, fixing your issue as well.

We'll update this ticket once the fix is merged.

robin-twice, 2024-10-07:
@igrr I stumbled over the issue @AguileraG described here. Do you have any idea whether this is going to be resolved any time soon? Thanks in advance!

closed 2024-11-04 by espressif-bot