espressif/esp-idf #14156: I2S TDM is changing the sequence of Data in 32bit DMA value

matti122, 2024-07-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-1388-g5ca9f2a49a

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

own PCB

### Power Supply used.

USB

### What is the expected behavior?

The ESP32-S3 I2S can only handle 32bit values from DMA. I'm using 16bit Mono Audio format. So I expect that the ESP32-S3 will use the lower or the upper 16 bits reliable and will not switch the Sequence in runtime and will also provide valid data.

### What is the actual behavior?

The ESP32-S3 is switching the Sequence of the Data inside the 32bit DMA value. If the lower 16bits are used the Data is not valid anymore. So it is a gamble to get valid Audio Data.

### Steps to reproduce.

1. Use TDM Audio 16bit Mono mode 1 Slot

L-KAYA, 2025-01-03:
Hi @matti122 ! I guess you are using RX to receive audio, right?

Please correct me if my understanding is wrong. You mean the valid data sometimes appears on the low 16 bits and sometimes appears on the high 16 bits during the runtime?

Normally, the data you received should be all valid, there won't be an invalid data between two valid data. Aside of this, the data switching issue might caused by the incorrect `dma_fram_num`. 

Could you provide the I2S configuration to check if anything wrongly configured?

matti122, 2025-01-03:
Hi @L-KAYA 
I was able to find a configuration which is working.
There were some wrong configurations at the external codec and the I2S.

closed 2025-01-03 by matti122