espressif/esp-idf #14157: Component manager does not work for components that are not in COMPONENTS variable

MathyV, 2024-07-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Install the dependencies

### What is the actual behavior?

It does not process the dependencies

### Steps to reproduce.

1. Create two components:
* foo has an idf_component.yml that specifies other dependencies
* bar has no idf_component.yml but lists foo as a dependency in REQUIRES

2. Set COMPONENTS to 'bar'
3. Build and watch how it does not download the extra dependencies

### More Information.

In `build.cmake` I can see that the component manager is called before the expansion of components based on their REQUIRES and PRIV_REQUIRES properties happens, so it's obvious component manager only know about the top-level components, not the ones that are lower down in the hierarchy.

This is confirmed by the workaround: adding the dependency (foo in my example) directly to COMPONENTS makes everything work.

igrr, 2024-07-10:
I think this was fixed on the master branch in https://github.com/espressif/esp-idf/commit/74843a442bc0dcf2d979076627be09fd0302b2b2, but we don't have this change in release/v5.2 yet.

hfudev, 2024-07-10:
the change has been backported to https://github.com/espressif/esp-idf/commit/c77c218b86410ecef671b5455ab1c6bab03ce0b8, but we didn't release it yet.

MathyV, 2024-07-10:
I can confirm it works with those changes, thx!

closed 2024-07-10 by MathyV