espressif/esp-idf #14158: ESP-TLS issue

korderavi, 2024-07-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.1

### Espressif SoC revision.

ESP32C6

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32c6

### Power Supply used.

External 5V

### What is the expected behavior?

i am using https rest server in esp32c6 controller which have 16mb flash
while i am requsting files from browser i will get same files and some of the files getting failed while loading 
following is screenshot of it

[image]

### What is the actual behavior?

whilerequesting of files all files should upload on brower 

### Steps to reproduce.

1. Step :at the starting of code we are geeting wifi IP adress
2. Step: by using IP address we are requsting files from browser 
3. Step: while uploding files it geting failed
...

### Debug Logs.

```plain
I (67841) SPI_thread:  In Cmd 5 Master cmd 0 
E (68101) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (68111) esp_https_server: esp_tls_create_server_session failed
E (68111) httpd: httpd_accept_conn: session creation failed
W (68111) httpd: httpd_server: error accepting new connection
I (68121) esp_https_server: performing session handshake
I (68341) SPI_thread:  In Cmd 5 Master cmd 0 
I (68841) SPI_thread:  In Cmd 5 Master cmd 0 
E (69071) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (69071) esp_https_server: esp_tls_create_server_session failed
E (69081) httpd: httpd_accept_conn: session creation failed
W (69081) httpd: httpd_server: error accepting new connection
I (69091) esp_https_server: performing session handshake
I (69341) SPI_thread:  In Cmd 5 Master cmd 0 
I (69841) SPI_thread:  In Cmd 5 Master cmd 0 
E (70021) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (70021) esp_https_server: esp_tls_create_server_session failed
E (70021) httpd: httpd_accept_conn: session creation failed
W (70031) httpd: httpd_server: error accepting new connection
I (70041) esp_https_server: performing session handshake
I (70341) SPI_thread:  In Cmd 5 Master cmd 0 
I (70851) SPI_thread:  In Cmd 5 Master cmd 0 
E (70971) esp-tls-mbedtls: mbedtls_ssl_handshake returned -0x7780
E (70971) esp_https_server: esp_tls_create_server_session failed
E (70971) httpd: httpd_accept_conn: session creation failed
W (70981) httpd: httpd_server: error accepting new connection
I (71351) SPI_thread:  In Cmd 5 Master cmd 0
```

AdityaHPatwardhan, 2024-07-16:
Hi @korderavi It seems that the TLS handshake failed. I have following questions that may give us more insights.

1) Is it possible for you to enable mbedTLS debug logs and share the result here?
2) Does the error occur once and keeps occurring successively? or is the occurrence arbitrary?
3) Have you checked the enough free memory available before the session is to be established ?

AdityaHPatwardhan, 2024-08-07:
Hi Closing the issue due to lack of response.
Please feel free to re-open if it persists.
Thanks,
Aditya

closed 2024-08-07 by AdityaHPatwardhan