espressif/esp-idf #14162: Some of the HID  consumer codes are not working as expected using the ble_hid_device_demo example (PlayPause(0xCD) AC Home (0xDF))

rleython, 2024-07-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.5.5

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

What I am trying to do is to emulate an STB Remote Controller that uses the BLE connection to control a STB, when I connect the original remote control to my Android Phone I can detect the commands, and I can control i.e. the youtube player; Using the BLE example  now I am able to use most of the commands defined there, but for _HID_CONSUMER_PLAY_PAUSE_  that is defined in hid_dev.h file, it is not working as expected, neither in the STB or in the Youtube player, and the same is happening with the _HID_CONSUMER_AC_HOME_ but this last one is not defined in hid_dev.h but I guess the code is the rigth one.

Most of the commands are working as expected using the _HID_, and _HID consumer_ codes, exept for codes:  

`#define HID_CONSUMER_PLAY_PAUSE     205 // Play/Pause`
and one more code not defined but acoording to some search should be :
`#define HID_CONSUMER_AC_HOME        223 // AC Home`

### What is the actual behavior?

When I connect the BLE device to an application that identifies the keypressed, this is doing nothing, Is like the user is sending anything.

### Steps to reproduce.

Basically I just modied the example to hardcode the commands:

1. I am sending the keycode in **true**,
2. Then I wait small period of time to simulate the **keypress**
3. Then I sent the keycode in false to **release** the key

```
            esp_hidd_send_consumer_value(hid_conn_id, HID_CONSUMER_PLAY_PAUSE, true);
            vTaskDelay(100 / portTICK_PERIOD_MS);
            esp_hidd_send_consumer_value(hid_conn_id, HID_CONSUMER_PLAY_PAUSE, false);
```

### More Information.

1. I have tried all the codes from 0x00 to 0xFF in both HID and HID Consumer codes and none of them are doing what is expected. I did it to discard that maybe there is another code expected.
2. I have tried adding the    ` 0x09, 0xCD,   //   Usage (Play/Pause) `to the descriptor _hid_device_le_prf.c_ file
3. Also I noticed that the case to build the buffer  when calling the function: _hid_consumer_build_report_  has all the cases exept for the commands that I was looking for, so I have tried adding the case like:

```
case HID_CONSUMER_PLAY_PAUSE:
            HID_CC_RPT_SET_BUTTON(buffer, HID_CONSUMER_PLAY_PAUSE);
            break; 
```

but none of the above solutions are working as what I was expecting.

Any help or advice is really apreciated

weiyuhannnn, 2024-07-15:
@rleython Thanks for choosing Espressif product! May I know which chip do you use?

rleython, 2024-07-15:
Thanks for your reply @weiyuhannnn , I am using the ESP32-WROOM-32D

gcstr, 2024-08-21:
I have the same issue. 

```
esp_hidd_send_consumer_value(hid_conn_id, HID_CONSUMER_PLAY_PAUSE, true);
```

has no effect

esp-xzk, 2024-11-12:
Hello,
Please find the attached patch for instructions on sending the commands you asked for.
To send more custom commands, you need to adjust the report map in `hid_device_le_prf.c` as well as the `hid_dev.h` file and the `hid_consumer_build_report` function. 
[0001-patch-add-support-for-sending-PLAY-PAUSE-and-AC-home.patch](https://github.com/user-attachments/files/17712324/0001-patch-add-support-for-sending-PLAY-PAUSE-and-AC-home.patch)

rleython, 2025-05-21:
Hello @Alvin1Zhang @esp-xzk , 

Sorry for late reply but I miss the fix and just notice the notification for the ticket to be closed, I have tried the patch but the issues still present, the Home,  and Play/Pause action still doing nothing :/ 

I dont have the permissions to reopen this ticket.