espressif/esp-idf #14164: esp-eye board camera timeout

leehakgun, 2024-07-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

A function that measures the voice dB and captures the image when it exceeds a certain dB and outputs it to the web server

### What is the actual behavior?

현재 i2s_channel_read 이 부분에서 제대로 작동하지 않아 버그가 발생하고 있습니다,

### Steps to reproduce.

#include <esp_system.h>
#include <nvs_flash.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "driver/gpio.h"
#include "esp_camera.h"
#include "esp_http_server.h"
#include "esp_timer.h"
#include "connect_wifi.h"
#include "esp_log.h"
#include <math.h>
#include "driver/i2s_std.h"
#include "esp_heap_caps.h"

// 카메라 핀 설정
#define PWDN_GPIO_NUM    -1
#define RESET_GPIO_NUM   -1
#define XCLK_GPIO_NUM    4
#define SIOD_GPIO_NUM    18
#define SIOC_GPIO_NUM    23
#define Y9_GPIO_NUM      36
#define Y8_GPIO_NUM      37
#define Y7_GPIO_NUM      38
#define Y6_GPIO_NUM      39
#define Y5_GPIO_NUM      35
#define Y4_GPIO_NUM      14
#define Y3_GPIO_NUM      13
#define Y2_GPIO_NUM      34
#define VSYNC_GPIO_NUM   5
#define HREF_GPIO_NUM    27
#define PCLK_GPIO_NUM    25

static const char *TAG = "esp32-cam Webserver";

camera_fb_t *last_fb = NULL;
bool capture_flag = false;
static i2s_chan_handle_t rx_chan; 

[... 258 lines cut ...]

I (1513) esp32-cam Webserver: Starting to measure decibel
I (1523) esp32-cam Webserver: Reading I2S data with RX channel handle: 0x3f808154
I (1523) esp32-cam Webserver: Before i2s_channel_read...
I (1633) esp32-cam Webserver: After i2s_channel_read...
E (1633) esp32-cam Webserver: I2S read error: ESP_ERR_TIMEOUT
E (1633) esp32-cam Webserver: Error details: ret=263, bytes_read=0
I (1633) esp32-cam Webserver: Decibel measurement done
W (1753) cam_hal: NO-SOI
W (1803) cam_hal: NO-SOI
W (1853) cam_hal: NO-SOI
W (1903) cam_hal: NO-SOI
W (1953) cam_hal: NO-SOI
W (2003) cam_hal: NO-SOI
W (2053) cam_hal: NO-SOI
W (2103) cam_hal: NO-SOI
W (2153) cam_hal: NO-SOI
W (2203) cam_hal: NO-SOI
W (2253) cam_hal: NO-SOI
W (2303) cam_hal: NO-SOI
W (2353) cam_hal: NO-SOI
W (2403) cam_hal: NO-SOI
W (2453) cam_hal: NO-SOI
W (2503) cam_hal: NO-SOI
```

Sandra-lol, 2024-07-11:
Hi, we've several suggestions for your reference:

1）May use examples/camera from esp-iot-solution repo, to validate that the devboard and camera sensor are working fine.
2）DVP camera on esp32-eye is implemented with I2S0 as default, so if you're using I2S, better to avoid I2S0 when camera is enabled.

Have a nice day~

leehakgun, 2024-07-12:
I thought it wouldn't matter if I set auto mode, but it wasn't.
Thanks to you, the problem has been solved.

closed 2024-07-12 by Alvin1Zhang