espressif/esp-idf #14167: openthread/ot_rcp build failing

pavel808, 2024-07-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-1388-g5ca9f2a49a

### Espressif SoC revision.

v2.8

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP-IDF 

### Power Supply used.

USB

### What is the expected behavior?

build ok

### What is the actual behavior?

Fails with logs captured below

### Steps to reproduce.

cd examples/openthread/ot_rcp
idf.py set-target esp32h2
idf.py build

### Debug Logs.

```plain
FAILED: esp-idf/mbedtls/mbedtls/3rdparty/p256-m/CMakeFiles/p256m.dir/p256-m/p256-m.c.obj 
/home/user/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20240530/riscv32-esp-elf/bin/riscv32-esp-elf-gcc -DESP_PLATFORM -DIDF_VER=\"v5.4-dev-1388-g5ca9f2a49a-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DSOC_XTAL_FREQ_MHZ=CONFIG_XTAL_FREQ -D_GLIBCXX_HAVE_POSIX_SEMAPHORE -D_GLIBCXX_USE_POSIX_SEMAPHORE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/home/user/esp-idf/examples/openthread/ot_rcp/build/config -I/home/user/esp-idf/components/mbedtls/port/include -I/home/user/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -I/home/user/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -I/home/user/esp-idf/components/mbedtls/mbedtls/include -I/home/user/esp-idf/components/mbedtls/mbedtls/library -I/home/user/esp-idf/components/newlib/platform_include -I/home/user/esp-idf/components/freertos/config/include -I/home/user/esp-idf/components/freertos/config/include/freertos -I/home/user/esp-idf/components/freertos/config/riscv/include -I/home/user/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/user/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include -I/home/user/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include/freertos -I/home/user/esp-idf/components/freertos/esp_additions/include -I/home/user/esp-idf/components/esp_hw_support/include -I/home/user/esp-idf/components/esp_hw_support/include/soc -I/home/user/esp-idf/components/esp_hw_support/include/soc/esp32h2 -I/home/user/esp-idf/components/esp_hw_support/dma/include -I/home/user/esp-idf/components/esp_hw_support/ldo/include -I/home/user/esp-idf/components/esp_hw_support/port/esp32h2/. -I/home/user/esp-idf/components/esp_hw_support/port/esp32h2/include -I/home/user/esp-idf/components/esp_hw_support/port/esp32h2/private_include -I/home/user/esp-idf/components/heap/include -I/home/user/esp-idf/components/log/include -I/home/user/esp-idf/components/soc/include -I/home/user/esp-idf/components/soc/esp32h2 -I/home/user/esp-idf/components/soc/esp32h2/include -I/home/user/esp-idf/components/hal/platform_port/include -I/home/user/esp-idf/components/hal/esp32h2/include -I/home/user/esp-idf/components/hal/include -I/home/user/esp-idf/components/esp_rom/include -I/home/user/esp-idf/components/esp_rom/esp32h2/include -I/home/user/esp-idf/components/esp_rom/esp32h2/include/esp32h2 -I/home/user/esp-idf/components/esp_rom/esp32h2 -I/home/user/esp-idf/components/esp_common/include -I/home/user/esp-idf/components/esp_system/include -I/home/user/esp-idf/components/esp_system/port/soc -I/home/user/esp-idf/components/esp_system/port/include/riscv -I/home/user/esp-idf/components/esp_system/port/include/private -I/home/user/esp-idf/components/riscv/include -I/home/user/esp-idf/components/lwip/include -I/home/user/esp-idf/components/lwip/include/apps -I/home/user/esp-idf/components/lwip/include/apps/sntp -I/home/user/esp-idf/components/lwip/lwip/src/include -I/home/user/esp-idf/components/lwip/port/include -I/home/user/esp-idf/components/lwip/port/freertos/include -I/home/user/esp-idf/components/lwip/port/esp32xx/include -I/home/user/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/user/esp-idf/components/lwip/port/esp32xx/include/sys -march=rv32imac_zicsr_zifencei  -Wall -Wextra -Wwrite-strings -Wformat=2 -Wno-format-nonliteral -Wvla -Wlogical-op -Wshadow -Wformat-signedness -Wformat-overflow=2 -Wformat-truncation -Werror -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-error=extra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -nostartfiles -Os -freorder-blocks -msave-restore -fmacro-prefix-map=/home/user/esp-idf/examples/openthread/ot_rcp=. -fmacro-prefix-map=/home/user/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/mbedtls/mbedtls/3rdparty/p256-m/CMakeFiles/p256m.dir/p256-m/p256-m.c.obj -MF esp-idf/mbedtls/mbedtls/3rdparty/p256-m/CMakeFiles/p256m.dir/p256-m/p256-m.c.obj.d -o esp-idf/mbedtls/mbedtls/3rdparty/p256-m/CMakeFiles/p256m.dir/p256-m/p256-m.c.obj -c /home/user/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m/p256-m.c
In file included from /home/user/esp-idf/components/mbedtls/mbedtls/include/psa/crypto_builtin_composites.h:28,
                 from /home/user/esp-idf/components/mbedtls/mbedtls/include/psa/crypto_driver_contexts_composites.h:28,
                 from /home/user/esp-idf/components/mbedtls/mbedtls/include/psa/crypto_struct.h:109,
                 from /home/user/esp-idf/components/mbedtls/mbedtls/include/psa/crypto.h:4678,
                 from /home/user/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m/p256-m.c:11:
/home/user/esp-idf/components/mbedtls/port/include/mbedtls/gcm.h:25:28: error: unknown type name 'mbedtls_gcm_context_soft'; did you mean 'mbedtls_gcm_context'?
   25 | void mbedtls_gcm_init_soft(mbedtls_gcm_context_soft *ctx);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~
      |                            mbedtls_gcm_context
/home/user/esp-idf/components/mbedtls/port/include/mbedtls/gcm.h:28:29: error: unknown type name 'mbedtls_gcm_context_soft'; did you mean 'mbedtls_gcm_context'?
   28 | int mbedtls_gcm_setkey_soft(mbedtls_gcm_context_soft *ctx,
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~
      |                             mbedtls_gcm_context
/home/user/esp-idf/components/mbedtls/port/include/mbedtls/gcm.h:33:29: error: unknown type name 'mbedtls_gcm_context_soft'; did you mean 'mbedtls_gcm_context'?
   33 | int mbedtls_gcm_starts_soft(mbedtls_gcm_context_soft *ctx,
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~
      |                             mbedtls_gcm_context
[... 3 lines cut ...]
      |                                mbedtls_gcm_context
/home/user/esp-idf/components/mbedtls/port/include/mbedtls/gcm.h:40:29: error: unknown type name 'mbedtls_gcm_context_soft'; did you mean 'mbedtls_gcm_context'?
   40 | int mbedtls_gcm_update_soft(mbedtls_gcm_context_soft *ctx,
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~
      |                             mbedtls_gcm_context
/home/user/esp-idf/components/mbedtls/port/include/mbedtls/gcm.h:45:29: error: unknown type name 'mbedtls_gcm_context_soft'; did you mean 'mbedtls_gcm_context'?
   45 | int mbedtls_gcm_finish_soft(mbedtls_gcm_context_soft *ctx,
      |                             ^~~~~~~~~~~~~~~~~~~~~~~~
      |                             mbedtls_gcm_context
/home/user/esp-idf/components/mbedtls/port/include/mbedtls/gcm.h:51:36: error: unknown type name 'mbedtls_gcm_context_soft'; did you mean 'mbedtls_gcm_context'?
   51 | int mbedtls_gcm_crypt_and_tag_soft(mbedtls_gcm_context_soft *ctx,
      |                                    ^~~~~~~~~~~~~~~~~~~~~~~~
      |                                    mbedtls_gcm_context
/home/user/esp-idf/components/mbedtls/port/include/mbedtls/gcm.h:64:35: error: unknown type name 'mbedtls_gcm_context_soft'; did you mean 'mbedtls_gcm_context'?
   64 | int mbedtls_gcm_auth_decrypt_soft(mbedtls_gcm_context_soft *ctx,
      |                                   ^~~~~~~~~~~~~~~~~~~~~~~~
      |                                   mbedtls_gcm_context
[... cut ...]
   75 | void mbedtls_gcm_free_soft(mbedtls_gcm_context_soft *ctx);
      |                            ^~~~~~~~~~~~~~~~~~~~~~~~
      |                            mbedtls_gcm_context
[19/1067] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/bignum.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/user/esp-idf/examples/openthread/ot_rcp/build/log/idf_py_stderr_output_43933 and /home/user/esp-idf/examples/openthread/ot_rcp/build/log/idf_py_stdout_output_43933
```

zwx1995esp, 2024-07-15:
Did you update the submodule? And could you please delete the `sdkconfig` and try anagin?

```
cd examples/openthread/ot_rcp
git submodule update --init --recursive
rm sdkconfig
idf.py set-target esp32h2
idf.py build
```

pavel808, 2024-07-16:
@zwx1995esp Thanks for your reply. I have done as you mentioned.

The main problem that I am facing is, on trying to access the serial terminal of the RCP ESP32-H2 for writing, I get the following issue : 

`Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.`

I believe that I must access the serial terminal here to configure the border router as mentioned here ? : 
https://docs.espressif.com/projects/esp-thread-br/en/latest/dev-guide/build_and_run.html#build-and-run-the-thread-cli-device

I am struggling to get this working now. I can't commission any devices through the TBR.

chshu, 2024-07-17:
The ot_rcp example doesn't provide the console interface, it only with a host SoC over the Spinel protocol.

If you want to use the CLI console, [ot_cli](https://github.com/espressif/esp-idf/tree/master/examples/openthread/ot_cli) is the correct example.

pavel808, 2024-07-17:
@chshu Thanks. I had just realized this. 

Is this _ot_cli_ example actually required to get the Thread network working? I don't fully understand. If we flash the ESP32-S3 chip with the right image, the ESP32-H2 will get set up and configured automatically according to the documentation. 

This is the issue that I am having here :  

https://github.com/espressif/esp-thread-br/issues/81

chshu, 2024-07-17:
`ot_cli` example is not mandatory to get the Thread network working, as long as the Thread BR is up and running, `ot_cli` is only used to test the Thread network features.

Looks like you have got the Thread BR working, but encountered some issue with Matter commissioning. Please close this issue if you don't have any question on rcp build, let's continue with https://github.com/espressif/esp-thread-br/issues/81

closed 2024-07-17 by pavel808