espressif/esp-idf #14169: ESP32S3 - PDM (Master Rx) - Over 104kHz not working

timsterc, 2024-07-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.4

### Espressif SoC revision.

ESP32-S3

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

I've been experimenting with PDM Rx up over 100kHz, and it appears that as soon as the MCLK_DIV goes from 3 to 2, that the MCLK goes to 10MHz, and that's it. That happens for anything over 104000 Hz Sampling Frequency.

I'm using ESP-IDF-5.1.4

I'm thinking (hoping) that it's something in the driver w/ all the clock setup - but when I look at all the underling calculations, they seem ok.

I do see this bit of code in the driver - so am thinking it might be an underlying hardware issue - but *maybe* with (another) workaround?

/**
* @brief Configure I2S RX module clock divider
*
* @param hw Peripheral I2S hardware instance address.
* @param mclk_div The mclk division coefficients
*/
static inline void i2s_ll_rx_set_mclk(i2s_dev_t *hw, const i2s_ll_mclk_div_t *mclk_div)
{
/* Workaround for inaccurate clock while switching from a relatively low sample rate to a high sample rate
* Set to particular coefficients first then update to the target coefficients,
* otherwise the clock division might be inaccurate.
* the general idea is to set a value that impossible to calculate from the regular decimal */
i2s_ll_rx_set_raw_clk_div(hw, 7, 317, 7, 3, 0);

Thanks!
-Tim

### What is the actual behavior?

Should be able to work above 104kHz

### Steps to reproduce.

Build and program the PDM Microphone Recording Example - then set sampling frequency over 104,000Hz.

timsterc, 2024-11-23:
@espressif-bot -- any update on this -- it's been nearly 6 months!

L-KAYA, 2024-12-10:
Hi @timsterc, so sorry that I missed this issue, may I ask the 104 KHz is the clock rate on the line or the sample rate configured while initializing the PDM?

Normally, the recommended sample rate of PDM mode is below 48 KHz.

As for the workaround in the code, it is used for the bug that sample rate sometimes can be half of the configured. Now it has been confirmed a hardware limitation, which can be solved by setting the clock divisions in a specific sequence (see the [fix commit on v5.1 branch](https://github.com/espressif/esp-idf/commit/eedf7b24ec792696de81ddb33a8451da1064292e))

timsterc, 2024-12-10:
@L-KAYA -- would you be able to try on one of your dev boards, seeing if it is possible for the PDM microphone example to work about 96kHz?  I'd ideally like to be able to have it working at 192kHz (for detecting bats).  If it isn't possible, that's OK, just would like to have someone at Espressif verify that is indeed the case.

Thanks!

L-KAYA, 2024-12-12:
Hi @timsterc, I've tried locally, the good news is. 

## Conclusion

Yes! The clock can indeed reach that high for 192KHz sample rate.

## Current Limitation 

However, the [default limitation](https://github.com/espressif/esp-idf/blob/a2f420a36ecf8cd19b443adde1d75e5eaecc6309/components/esp_driver_i2s/i2s_pdm.c#L348C3-L348C4) in the driver limits the minimum mclk. So, the mclk will be too large to be generated from the source clock.

### Why There Is Such Limitation?

To guarantee the a high SNR (signal noise ratio).

### How to Remove The Limitation?

1. Modify the [default limitation](https://github.com/espressif/esp-idf/blob/a2f420a36ecf8cd19b443adde1d75e5eaecc6309/components/esp_driver_i2s/i2s_pdm.c#L348C3-L348C4) to `4` for example;
2. Set the `i2s_pdm_rx_config_t::clk_cfg.bclk_div = 4`;

Alvin1Zhang, 2025-01-10:
Thanks for reporting and sorry for slow turnaround, will close due to short of updates, feel free to reopen with more updates.

closed 2025-01-10 by Alvin1Zhang

timsterc, 2025-01-19:
I have made your suggested change to the pdm driver, but the audio still doesn't seem to be working correctly.  

Any chance you could try this on one of your development boards that includes a PDM microphone?  Maybe you could use the PDM Microphone Recording example and modify it to work at 192,000Hz ?

Thanks!
-Tim