espressif/esp-idf #14170: Modbus TCP Master Example crashes

tomlinyama, 2024-07-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Execute normally like example

### What is the actual behavior?

I use the example to only modify the IP address, but the example will crash.

### Steps to reproduce.

1. modify ip address
[image]

...

### Build or installation Logs.

```plain
I (39) boot.esp32c3: SPI Speed      : 80MHz
I (43) boot.esp32c3: SPI Mode       : DIO
I (48) boot.esp32c3: SPI Flash Size : 4MB
I (53) boot: Enabling RNG early entropy source...
I (58) boot: Partition Table:
I (62) boot: ## Label            Usage          Type ST Offset   Length
I (69) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (77) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (84) boot:  2 factory          factory app      00 00 00010000 00100000
I (91) boot: End of partition table
I (96) esp_image: segment 0: paddr=00010020 vaddr=3c0b0020 size=259c0h (154048) map
I (128) esp_image: segment 1: paddr=000359e8 vaddr=3fc93000 size=02b18h ( 11032) load
I (131) esp_image: segment 2: paddr=00038508 vaddr=40380000 size=07b10h ( 31504) load
I (140) esp_image: segment 3: paddr=00040020 vaddr=42000020 size=a25b8h (665016) map
I (246) esp_image: segment 4: paddr=000e25e0 vaddr=40387b10 size=0b3c4h ( 46020) load
I (261) boot: Loaded app from partition at offset 0x10000
I (261) boot: Disabling RNG early entropy source...
I (273) cpu_start: Unicore app
I (282) cpu_start: Pro cpu start user code
I (282) cpu_start: cpu freq: 160000000 Hz
I (282) cpu_start: Application information:
I (285) cpu_start: Project name:     modbus_tcp_master
I (291) cpu_start: App version:      v5.2.1-dirty
I (296) cpu_start: Compile time:     Jul 12 2024 15:08:41
I (302) cpu_start: ELF file SHA256:  4f76b17c5...
I (307) cpu_start: ESP-IDF:          v5.2.1-dirty
I (313) cpu_start: Min chip rev:     v0.3
I (318) cpu_start: Max chip rev:     v1.99 
I (322) cpu_start: Chip rev:         v0.3
I (327) heap_init: Initializing. RAM available for dynamic allocation:
[... 120 lines cut ...]

3fc9d740: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9d760: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9d780: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9d7a0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9d7c0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9d7e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9d800: 0x00000000 0x00000000 0x00000000 0x00000600 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9d820: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9d840: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9d860: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9d880: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9d8a0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9d8c0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9d8e0: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9d900: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9d920: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9d940: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9d960: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5
3fc9d980: 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5

ELF file SHA256: 4f76b17c5

Rebooting...
```

alisitsyn, 2024-07-12:
@tomlinyama, the log says that your configuration for the master is incorrect. Please add here the configuration of data dictionary and the changes in IP address table.

tomlinyama, 2024-07-15:
@alisitsyn This issue has been resolved: the system would crash when unable to connect, although sometimes it could connect successfully.
 (5453) uart: ESP_INTR_FLAG_IRAM flag not set while CONFIG_UART_ISR_IN_IRAM is enabled, flag updated
I (5463) MASTER_TEST: Leave IP(0) = [192.168.1.124] set manually.
I (5473) MASTER_TEST: IP(1) is not set in the table.
I (5483) MASTER_TEST: Configured 1 IP addresse(s).
I (5483) MASTER_TEST: Modbus master stack initialized...
I (5483) MB_TCP_MASTER_PORT: TCP master stack initialized.
I (5503) MB_TCP_MASTER_PORT: Host[IP]: "192.168.1.124"[192.168.1.124]
I (5513) MB_TCP_MASTER_PORT: Add slave IP: 192.168.1.124
I (5513) MB_TCP_MASTER_PORT: Connecting to slaves...
-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.-.E (25513) MB_CONTROLLER_MASTER: mbc_tcp_master_start(178): mb stack could not connect to slaves for 20 seconds.
E (25513) MB_CONTROLLER_MASTER: mbc_master_start(175): Master start failure, error=(0x103) (ESP_ERR_INVALID_STATE).
-E (25523) MASTER_TEST: master_init(700): mb controller start fail, returns(0x103).
ESP_ERROR_CHECK failed: esp_err_t 0x103 (ESP_ERR_INVALID_STATE) at 0x4200701c
0x4200701c: app_main at /home/xsza/esp-idf/examples/protocols/modbus/tcp/mb_tcp_master/main/tcp_master.c:736 (discriminator 1)

file: "/IDF/examples/protocols/modbus/tcp/mb_tcp_master/main/tcp_master.c" line 736
func: app_main
expression: master_init(&comm_info)

abort() was called at PC 0x40387739 on core 0
0x40387739: _esp_error_check_failed at /home/xsza/esp-idf/components/esp_system/esp_err.c:50

Core  0 register dump:
MEPC    : 0x403806d0  RA      : 0x40387744  SP      : 0x3fc9d5a0  GP      : 0x3fc93800  
0x403806d0: panic_abort at /home/xsza/esp-idf/components/esp_system/panic.c:472
0x40387744: __ubsan_include at /home/xsza/esp-idf/components/esp_system/ubsan.c:313

TP      : 0x3fc78db4  T0      : 0x37363534  T1      : 0x7271706f  T2      : 0x33323130  
S0/FP   : 0x00000004  S1      : 0x3fc9d604  A0      : 0x3fc9d5cc  A1      : 0x3fc9d602  
A2      : 0x00000000  A3      : 0x3fc9d5f9  A4      : 0x00000001  A5      : 0x3fc9a000  
A6      : 0x7a797877  A7      : 0x76757473  S2      : 0x00000000  S3      : 0x00000000  
S4      : 0x00000000  S5      : 0x00000000  S6      : 0x00000000  S7      : 0x00000000  
S8      : 0x00000000  S9      : 0x00000000  S10     : 0x00000000  S11     : 0x00000000  
T3      : 0x6e6d6c6b  T4      : 0x6a696867  T5      : 0x66656463  T6      : 0x62613938  
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000  
0x40380001: _vector_table at ??:?

MHARTID : 0x00000000  

Stack memory:
3fc9d5a0: 0x3c0b505c 0x42007020 0x3fc9d600 0x4038eed6 0x00000000 0x00000000 0x00000000 0x3fc94c3c
0x42007020: mbc_master_init_iface at /home/xsza/esp-idf/examples/protocols/modbus/tcp/mb_tcp_master/managed_components/espressif__esp-modbus/freemodbus/common/esp_modbus_master.c:21
0x4038eed6: abort at /home/xsza/esp-idf/components/newlib/abort.c:35

3fc9d5c0: 0x3fc9d604 0x3fc94c58 0x3fc9d600 0x726f6261 0x20292874 0x20736177 0x6c6c6163 0x61206465
3fc9d5e0: 0x43502074 0x34783020 0x37383330 0x20393337 0x63206e6f 0x2065726f 0x00000030 0x00000000
3fc9d600: 0x3c0b0030 0x38333034 0x39333737 0x00000000 0x00000000 0x00000000 0x3c0b2000 0x4038773c
0x4038773c: esp_system_abort at /home/xsza/esp-idf/components/esp_system/port/esp_system_chip.c:92

3fc9d620: 0x3fc9e028 0x00000000 0x3c0b2000 0x42007020 0x00000000 0x00000000 0x00000002 0x01f60000
0x42007020: mbc_master_init_iface at /home/xsza/esp-idf/examples/protocols/modbus/tcp/mb_tcp_master/managed_components/espressif__esp-modbus/freemodbus/common/esp_modbus_master.c:21

3fc9d640: 0x00000000 0x3fc9303c 0x3fca90dc 0x00000000 0x3c0b1eec 0x0000017f 0xfe7fffff 0x4209e8fe
0x4209e8fe: main_task at /home/xsza/esp-idf/components/freertos/app_startup.c:209 (discriminator 13)

3fc9d660: 0x00000000 0x00001388 0x00000001 0x00000000 0x00000000 0x00000000 0x00000000 0x403887e8
0x403887e8: vPortTaskWrapper at /home/xsza/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:258

3fc9d680: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fc9d6a0: 0x00000000 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0xa5a5a5a5 0x00000150 0x3fc9d4b0
3fc9d6c0: 0x000009d1 0x3fc95cd8 0x3fc95cd8 0x3fc9d6bc 0x3fc95cd0 0x00000018 0x3fcaa6c0 0x3fcaa6c0
3fc9d6e0: 0x3fc9d6bc 0x00000000 0x00000001 0x3fc9c6b8 0x6e69616d 0x00000000 0x00000000 0x00000000
3fc9d700: 0x3fc9d6b0 0x00000001 0x00000000 0x3fca13d8 0x4200043e 0x00000000 0x3fc9af1c 0x3fc9af84
0x4200043e: pthread_cleanup_thread_specific_data_callback at /home/xsza/esp-idf/components/pthread/pthread_local_storage.c:126

3fc9d720: 0x3fc9afec 0x00000000 0x00000000 0x00000001 0x00000000 0x00000000 0x00000000 0x42004d3e
0x42004d3e: esp_cleanup_r at /home/xsza/esp-idf/components/newlib/newlib_init.c:60

3fc9d740: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
[... 113 lines cut ...]
I (3393) wifi:pm start, type: 1

I (3403) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
[... cut ...]
I (5473) MB_TCP_MASTER_PORT: TCP master stack initialized.
I (5473) MB_TCP_MASTER_PORT: Host[IP]: "192.168.1.124"[192.168.1.124]
I (5493) MB_TCP_MASTER_PORT: Add slave IP: 192.168.1.124
I (5493) MB_TCP_MASTER_PORT: Connecting to slaves...
-.-.-.I (5803) MB_TCP_MASTER_PORT: Connected 1 slaves, start polling...
I (5853) MASTER_TEST: Start modbus test...
I (5853) MASTER_TEST: Characteristic #0 Data_channel_0 (Volts) value = 0.000000 (0x640000) read successful.
I (6363) MASTER_TEST: Alarm triggered by cid #0.

alisitsyn, 2024-07-15:
@tomlinyama, your original issue happened due to incorrect configuration. I propose you to close the issue then.

closed 2024-07-15 by tomlinyama