espressif/esp-idf #14172: ESP32-S3 - Combination Secure boot+Flash Encryption Failed on IDF 4.4.8

dzungpv, 2024-07-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a project success enable Secure boot+Flash Encryption for ESP32, but when I change to ESP32-S3 it failed,
After I flash the boot loader and the application image, it show in the log:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0x8 (SPI_FAST_FLASH_BOOT)
Saved PC:0x40048d46
invalid header: 0x20a36bad
invalid header: 0x20a36bad
invalid header: 0x20a36bad
invalid header: 0x20a36bad
invalid header: 0x20a36bad
invalid header: 0x20a36bad
invalid header: 0x20a36bad
```
This is the part of sdkconfig to enable flash encryption and secure boot:
```
#
# Security features
#
CONFIG_SECURE_SIGNED_ON_BOOT=y
CONFIG_SECURE_SIGNED_ON_UPDATE=y
CONFIG_SECURE_SIGNED_APPS=y
CONFIG_SECURE_BOOT_SUPPORTS_RSA=y
CONFIG_SECURE_TARGET_HAS_SECURE_ROM_DL_MODE=y
CONFIG_SECURE_SIGNED_APPS_RSA_SCHEME=y
CONFIG_SECURE_BOOT=y
CONFIG_SECURE_BOOT_V2_ENABLED=y
CONFIG_SECURE_BOOT_BUILD_SIGNED_BINARIES=y
CONFIG_SECURE_BOOT_SIGNING_KEY="sbv2_private.pem"
# CONFIG_SECURE_BOOT_ENABLE_AGGRESSIVE_KEY_REVOKE is not set
# CONFIG_SECURE_BOOT_INSECURE is not set
CONFIG_SECURE_FLASH_ENC_ENABLED=y
CONFIG_SECURE_FLASH_ENCRYPTION_AES128=y
# CONFIG_SECURE_FLASH_ENCRYPTION_AES256 is not set
# CONFIG_SECURE_FLASH_ENCRYPTION_MODE_DEVELOPMENT is not set
CONFIG_SECURE_FLASH_ENCRYPTION_MODE_RELEASE=y
CONFIG_SECURE_FLASH_CHECK_ENC_EN_IN_APP=y
# CONFIG_SECURE_DISABLE_ROM_DL_MODE is not set
CONFIG_SECURE_ENABLE_SECURE_ROM_DL_MODE=y
# CONFIG_SECURE_INSECURE_ALLOW_DL_MODE is not set
# end of Security features
```
I does not manual burn any efuse key, just let it does automatically by the chip. I flash the boot loader first then flash all later. The guide too complex and I have bricked many ESP32 chip before it work, this time occur with the S3 too, QEMU not working with the S3, so I have some esp32s3 to try, but before try the second one, I need some more clear guide.

dzungpv, 2024-07-13:
Updated: After check buy esptool.py --chip esp32s3 --no-stub get_security_info   I see the the flash encryption is not enable, it may failed on the second stage boot so the bootloader still keep it disable.

I try multiple way to make the chip back, and finally disable flash encryption and re-flash all the parts exclude boot loader with `idf.py flash` and it working with secure boot enable, logs:
```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
Valid secure boot key blocks: 0
secure boot verification succeeded
load:0x3fce3980,len:0x2618
load:0x403c9700,len:0x4
load:0x403c9704,len:0xa48
load:0x403cc700,len:0x47e8
entry 0x403c9894
I (435) cpu_start: Multicore app
I (436) cpu_start: Pro cpu up.
I (436) cpu_start: Starting app cpu, entry
```
**After that I retry enable flash encryption again** and reflash (because I choose secure download mode), this time it show meaning log here:

```
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
Octal Flash Mode Enabled
For OPI Flash, Use Default Flash Boot Mode
mode:SLOW_RD, clock div:1
Valid secure boot key blocks: 0
secure boot verification succeeded
load:0x3fce3970,len:0x3850
load:0x403c9700,len:0xbc0
load:0x403cc700,len:0x53a4
entry 0x403c9950
I (82) boot: ESP-IDF v4.4.8-dirty 2nd stage bootloader
I (82) boot: compile time 00:57:30
I (82) boot: Multicore bootloader
I (85) boot: chip revision: v0.2
I (89) boot.esp32s3: Boot SPI Speed : 80MHz
I (93) boot.esp32s3: SPI Mode       : SLOW READ
I (99) boot.esp32s3: SPI Flash Size : 8MB
I (103) boot: Enabling RNG early entropy source...
I (109) boot: Partition Table:
I (113) boot: ## Label            Usage          Type ST Offset   Length
I (120) boot:  0 nvs              WiFi data        01 02 00011000 0000c000
I (127) boot:  1 otadata          OTA data         01 00 0001d000 00002000
I (135) boot:  2 phy_init         RF data          01 01 0001f000 00001000
I (143) boot:  3 app0             OTA app          00 10 00020000 001e0000
I (150) boot:  4 app1             OTA app          00 11 00200000 001e0000
I (158) boot:  5 spiffs           Unknown data     01 82 003e0000 00011000
I (165) boot:  6 coredump         Unknown data     01 03 003f1000 00008000
I (173) boot:  7 esp_secure_cert  WiFi data        01 02 003f9000 00002000
I (180) boot:  8 fctry            WiFi data        01 02 003fb000 00004000
I (188) boot:  9 nvs_keys         NVS keys         01 04 003ff000 00001000
I (195) boot: End of partition table
I (200) esp_image: segment 0: paddr=00200020 vaddr=3c100020 size=3f61ch (259612) map
I (269) esp_image: segment 1: paddr=0023f644 vaddr=3fc9bd10 size=009d4h (  2516) load
I (270) esp_image: segment 2: paddr=00240020 vaddr=42000020 size=f01c8h (983496) map
I (506) esp_image: segment 3: paddr=003301f0 vaddr=3fc9c6e4 size=03c08h ( 15368) load
I (511) esp_image: segment 4: paddr=00333e00 vaddr=40374000 size=17d08h ( 97544) load
I (539) esp_image: segment 5: paddr=0034bb10 vaddr=00000000 size=044c0h ( 17600) 
I (543) esp_image: Verifying image signature...
I (543) secure_boot_v2: Verifying with RSA-PSS...
I (549) secure_boot_v2: Signature verified successfully!
I (561) boot: Loaded app from partition at offset 0x200000
I (562) secure_boot_v2: enabling secure boot v2...
[... 51 lines cut ...]
> I (561) boot: Loaded app from partition at offset 0x200000

And

> E (987) flash_encrypt: Invalid key state, check read&write protection for key and keypurpose(if exists)
I (998) efuse: Batch mode of writing fields is cancelled
E (1003) boot: Flash encryption check failed (259).

Oh, why the boot-loader load app at offset 0x200000? Because my app partition at the 0x20000?

This is the partition table:

```
# Name,   Type, SubType, Offset,  Size, Flags
nvs, data, nvs, 0x11000, 0xC000
otadata, data, ota, 0x1D000, 0x2000
phy_init, data, phy, 0x1F000, 0x1000
app0, app, ota_0, 0x20000, 0x1E0000
app1, app, ota_1, 0x200000, 0x1E0000
spiffs, data, spiffs, 0x3E0000, 0x11000
coredump, data, coredump, 0x3F1000, 0x8000
esp_secure_cert, data, nvs, 0x3F9000, 0x2000
fctry, data, nvs, 0x3FB000, 0x4000
nvs_keys, data, nvs_keys, 0x3FF000, 0x1000
```

mahavirj, 2024-07-15:
@dzungpv 

Please see the docs guide [here](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/security/host-based-security-workflows.html#enable-flash-encryption-and-secure-boot-v2-externally) for enabling the secure boot and flash-encryption features in step-by-step manner.

The order for enabling these features must be flash encryption first and then secure boot, the reason is specified in the doc guide pointer above. 

For your case, since the secure boot is enabled first, read protection of the flash-encryption key is no more possible and hence the error that you are getting here.

dzungpv, 2024-07-15:
> @dzungpv
>
> Please see the docs guide [...]

I solved problem myself, the key is the order of the flash.
The secure boot loader bin must be flash **LAST**.
Or a better way is merged all the binary include boot loader then flash only 1 time.

Please add to the document.

closed 2024-07-15 by dzungpv

aygh4266, 2024-08-15:
Hi, how did you solve this ? I have enabled only the secure boot, ran the command idf.py bootloader and then idf.py flash but the output looks like this after running idf.py monitor :

ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
invalid header: 0x020150aa
invalid header: 0x020150aa
invalid header: 0x020150aa
invalid header: 0x020150aa
invalid header: 0x020150aa
invalid header: 0x020150aa
invalid header: 0x020150aa
invalid header: 0x020150aa
invalid header: 0x020150aa

dzungpv, 2025-05-06:
> Hi [@dzungpv](https://github.com/dzungpv), Can you please specify the flashing order if you remember? cause the merged binaries still have the same issue for me.

Make sure you disable all cache, run 'idf.py fullclean' command or even delete all build folder before build the release. I often using 'rm -rf build*' along with fullclean and it work fine. Also merge all release files to 1 bin file and you will make the failed smallest. Check this guide: [https://docs.espressif.com/projects/esptool/en/latest/esp32/esptool/basic-commands.html?highlight=merge#merge-binaries-for-flashing-merge-bin it is for esp32 but ](https://docs.espressif.com/projects/esptool/en/latest/esp32s3/esptool/basic-commands.html#merge-binaries-for-flashing-merge-bin). If you want to flash file separately, the order I do below:
1, partition
2, factory if you have 
3, application
4, bootloader
5, efuse
6, reset or unplug/replug, waiting some minutes for encryption.
Remember do not reboot it during flash (check flash command), and full erase it first.

lekshith, 2025-05-07:
@dzungpv, 
You are a legend, Thank you!