espressif/esp-idf #14173: Clang builds of IDF are broken

KaeLL, 2024-07-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Manually patched 5ca9f2a49a it as below to build my project. Isn't esp-clang mature enough to add builds using it to the CI workflow? The complaints raised which prompted the changes seem sensible enough to me.
<details>
<summary>diff</summary>

```diff
diff --git a/components/bt/host/nimble/nimble b/components/bt/host/nimble/nimble
--- a/components/bt/host/nimble/nimble
+++ b/components/bt/host/nimble/nimble
@@ -1 +1 @@
-Subproject commit 74f81e10d8911e6206dfd75a2459dbebe5886d10
+Subproject commit 74f81e10d8911e6206dfd75a2459dbebe5886d10-dirty
diff --git a/components/esp_eth/src/mac/esp_eth_mac_esp.c b/components/esp_eth/src/mac/esp_eth_mac_esp.c
index dddf0d24b1..8bf39495af 100644
--- a/components/esp_eth/src/mac/esp_eth_mac_esp.c
+++ b/components/esp_eth/src/mac/esp_eth_mac_esp.c
@@ -584,6 +584,7 @@ static esp_err_t emac_esp_config_data_interface(const eth_esp32_emac_config_t *e
     esp_err_t ret = ESP_OK;
     switch (esp32_emac_config->interface) {
     case EMAC_DATA_INTERFACE_MII:
+    {
         /* MII interface GPIO initialization */
 #if SOC_EMAC_MII_USE_GPIO_MATRIX
         ESP_GOTO_ON_ERROR(emac_esp_gpio_matrix_init_mii(&esp32_emac_config->emac_dataif_gpio.mii), err, TAG, "failed to initialize EMAC MII GPIO Matrix");
@@ -599,7 +600,9 @@ static esp_err_t emac_esp_config_data_interface(const eth_esp32_emac_config_t *e
             emac_hal_clock_enable_mii(&emac->hal);
         }
         break;
+    }
     case EMAC_DATA_INTERFACE_RMII:
+    {
         /* RMII interface GPIO initialization */
         const eth_mac_rmii_gpio_config_t *rmii_data_gpio = NULL;
 #if SOC_EMAC_USE_MULTI_IO_MUX
@@ -640,6 +643,7 @@ static esp_err_t emac_esp_config_data_interface(const eth_esp32_emac_config_t *e
             ESP_GOTO_ON_FALSE(false, ESP_ERR_INVALID_ARG, err, TAG, "invalid EMAC clock mode");
         }
         break;
+    }
     default:
         ESP_GOTO_ON_FALSE(false, ESP_ERR_INVALID_ARG, err, TAG, "invalid EMAC Data Interface:%i", esp32_emac_config->interface);
     }
diff --git a/components/wifi_provisioning/src/scheme_ble.c b/components/wifi_provisioning/src/scheme_ble.c
index 6b594f22c4..60e19ad271 100644
--- a/components/wifi_provisioning/src/scheme_ble.c
+++ b/components/wifi_provisioning/src/scheme_ble.c
@@ -199,10 +199,11 @@ static esp_err_t set_config_endpoint(void *config, const char *endpoint_name, ui
 /* Used when both BT and BLE are not needed by application */
 void wifi_prov_scheme_ble_event_cb_free_btdm(void *user_data, wifi_prov_cb_event_t event, void *event_data)
 {
+    esp_err_t err;
     switch (event) {
         case WIFI_PROV_INIT:
+        {
 #ifdef CONFIG_BT_CONTROLLER_ENABLED
-            esp_err_t err;
             /* Release BT memory, as we need only BLE */
             err = esp_bt_mem_release(ESP_BT_MODE_CLASSIC_BT);
             if (err != ESP_OK) {
@@ -212,8 +213,10 @@ void wifi_prov_scheme_ble_event_cb_free_btdm(void *user_data, wifi_prov_cb_event
             }
 #endif
             break;
+        }
 
         case WIFI_PROV_DEINIT:
+        {
 #ifndef CONFIG_WIFI_PROV_KEEP_BLE_ON_AFTER_PROV
 #ifdef CONFIG_BT_CONTROLLER_ENABLED
             /* Release memory used by BLE and Bluedroid host stack */
@@ -226,6 +229,7 @@ void wifi_prov_scheme_ble_event_cb_free_btdm(void *user_data, wifi_prov_cb_event
 #endif
 #endif
             break;
+        }
 
         default:
             break;
@@ -235,10 +239,10 @@ void wifi_prov_scheme_ble_event_cb_free_btdm(void *user_data, wifi_prov_cb_event
 /* Used when BT is not needed by application */
 void wifi_prov_scheme_ble_event_cb_free_bt(void *user_data, wifi_prov_cb_event_t event, void *event_data)
 {
+    esp_err_t err;
     switch (event) {
         case WIFI_PROV_INIT:
 #ifdef CONFIG_BT_CONTROLLER_ENABLED
-            esp_err_t err;
             /* Release BT memory, as we need only BLE */
             err = esp_bt_mem_release(ESP_BT_MODE_CLASSIC_BT);
             if (err != ESP_OK) {
@@ -257,11 +261,11 @@ void wifi_prov_scheme_ble_event_cb_free_bt(void *user_data, wifi_prov_cb_event_t
 /* Used when BLE is not needed by application */
 void wifi_prov_scheme_ble_event_cb_free_ble(void *user_data, wifi_prov_cb_event_t event, void *event_data)
 {
+    esp_err_t err;
     switch (event) {
         case WIFI_PROV_DEINIT:
 #ifndef CONFIG_WIFI_PROV_KEEP_BLE_ON_AFTER_PROV
 #ifdef CONFIG_BT_CONTROLLER_ENABLED
-            esp_err_t err;
             /* Release memory used by BLE stack */
             err = esp_bt_mem_release(ESP_BT_MODE_BLE);
             if (err != ESP_OK) {

```

</details>

igrr, 2024-07-13:
Thanks for notifying! We just noticed this as well, the fix is in review.

There is a CI build, but it was smaller in scope and was missing a few components, hence the regression. That will be fixed as well.

mahavirj, 2024-08-08:
Fixed with d146fb5b8428ae522a7e4721f48de54663b7ef4b

closed 2024-08-08 by mahavirj

KaeLL, 2024-09-20:
espressif/esp-nimble#76