espressif/esp-idf #14176: The use of the FCS field in IEEE 802.15.4 frames is not documented and it doesn't affect transmission

igormichalak, 2024-07-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

According to the comment associated with the `esp_ieee802154_transmit` function, the FCS field should be present after the payload: https://github.com/espressif/esp-idf/blob/master/components/ieee802154/include/esp_ieee802154.h#L133

I'm trying to figure out how CRC can be calculated correctly based on MHR and payload, but no matter what it is set to, the frame is always received successfully, even with promiscuous mode turned off. 

I also tried using nRF52840 MDK USB dongle for sniffing, but according to it, the FCS is not present, but it may just be stripped away.

Is FCS calculated by the hardware and all I have to do is to allocate space for it?

```c
#include <stdio.h>
#include <string.h>
#include <stdint.h>
#include <inttypes.h>

#include "esp_attr.h"
#include "esp_err.h"
#include "esp_log.h"
#include "nvs_flash.h"
#include "esp_ieee802154.h"
#include "esp_ieee802154_types.h"
#include "esp_random.h"
#include "esp_crc.h"
#include "esp_mac.h"
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "portmacro.h"

#define IEEE802154_FRAME_TYPE_BEACON    (0b000 << 0)
#define IEEE802154_FRAME_TYPE_DATA      (0b001 << 0)
#define IEEE802154_FRAME_TYPE_ACK       (0b010 << 0)
#define IEEE802154_FRAME_TYPE_CMD       (0b011 << 0)
#define IEEE802154_FRAME_TYPE_MASK      0x0007

#define IEEE802154_SEC_ENABLED          (1 << 3)
#define IEEE802154_FRAME_PENDING        (1 << 4)
#define IEEE802154_ACK_REQ              (1 << 5)
#define IEEE802154_PAN_ID_COMPRESSION   (1 << 6)

#define IEEE802154_DST_ADDR_MODE_NONE           (0b00 << 10)
#define IEEE802154_DST_ADDR_MODE_ONLY_SHORT     (0b10 << 10)
#define IEEE802154_DST_ADDR_MODE_ONLY_EXT       (0b11 << 10)
#define IEEE802154_DST_ADDR_MODE_MASK           0x0C00

#define IEEE802154_FRAME_VER_2003       (0b00 << 12)
#define IEEE802154_FRAME_VER_2006       (0b01 << 12)
#define IEEE802154_FRAME_VER_MASK       0x3000

#define IEEE802154_SRC_ADDR_MODE_NONE           (0b00 << 14)
#define IEEE802154_SRC_ADDR_MODE_ONLY_SHORT     (0b10 << 14)
#define IEEE802154_SRC_ADDR_MODE_ONLY_EXT       (0b11 << 14)
#define IEEE802154_SRC_ADDR_MODE_MASK           0xC000

#define IEEE802154_DST_PAN_ID_BROADCAST         0xFFFF
#define IEEE802154_DST_ADDR_BROADCAST           0xFFFF

#define MHR_SIZE 11
#define PAYLOAD_SIZE 96
#define FRAME_SIZE (MHR_SIZE + PAYLOAD_SIZE + 2)

typedef struct __attribute__((packed)) {
    struct __attribute__((packed)) {
        uint16_t frame_control;
        uint8_t seq_num;
        uint16_t dst_pan_id;
        uint16_t dst_addr;
        uint16_t src_pan_id;
        uint16_t src_addr;
    } mhr;
    uint8_t payload[PAYLOAD_SIZE];
    uint16_t fcs;
} ieee802154_frame_t;

const static char *TAG = "MESH";

void calculate_crc(ieee802154_frame_t *frame) {
    static uint8_t buf[MHR_SIZE+PAYLOAD_SIZE];

    memcpy(buf, (uint8_t *) &frame->mhr, MHR_SIZE);
    memcpy(buf + MHR_SIZE, frame->payload, PAYLOAD_SIZE);

    uint16_t crc = 0x0000;

    crc = esp_crc16_le(crc, buf, MHR_SIZE+PAYLOAD_SIZE);

    frame->fcs = crc;
}

static void mesh_proto_worker(void *pvParams) {
    uint8_t seq_num = 0;

    while (true) {
        vTaskDelay(5000 / portTICK_PERIOD_MS);

        ieee802154_frame_t frame = {
            .mhr = {
                .frame_control = IEEE802154_FRAME_VER_2006 |
                                 IEEE802154_FRAME_TYPE_DATA |
                                 IEEE802154_DST_ADDR_MODE_ONLY_SHORT |
                                 IEEE802154_SRC_ADDR_MODE_ONLY_SHORT,
                .seq_num = seq_num++,
                .dst_pan_id = IEEE802154_DST_PAN_ID_BROADCAST,
                .dst_addr = IEEE802154_DST_ADDR_BROADCAST,
                .src_pan_id = *(uint16_t *) pvParams,
                .src_addr = 0x0001,
            },
            .payload = { 0 },
        };

        const char *msg = "Hello, World!";
        strncpy((char *) frame.payload, msg, PAYLOAD_SIZE);

        calculate_crc(&frame);

        static uint8_t tx_frame[FRAME_SIZE+1];
        tx_frame[0] = FRAME_SIZE;
        memcpy(tx_frame + 1, (uint8_t *) &frame, FRAME_SIZE);

        esp_ieee802154_transmit(tx_frame, true);
    }
}

void IRAM_ATTR esp_ieee802154_transmit_done(const uint8_t *frame, const uint8_t *ack, esp_ieee802154_frame_info_t *ack_frame_info) {
    if (ack != NULL) {
        ESP_ERROR_CHECK(esp_ieee802154_receive_handle_done(ack));
    }
}

void IRAM_ATTR esp_ieee802154_transmit_failed(const uint8_t *frame, esp_ieee802154_tx_error_t error) {
    switch (error) {
    case ESP_IEEE802154_TX_ERR_NONE:
        ESP_DRAM_LOGE(TAG, "Transmit: unknown error");
        break;
    case ESP_IEEE802154_TX_ERR_CCA_BUSY:
        ESP_DRAM_LOGE(TAG, "Transmit: channel is busy");
        break;
    case ESP_IEEE802154_TX_ERR_ABORT:
        ESP_DRAM_LOGE(TAG, "Transmit: abort");
        break;
    case ESP_IEEE802154_TX_ERR_NO_ACK:
        ESP_DRAM_LOGE(TAG, "Transmit: no ack");
        break;
    case ESP_IEEE802154_TX_ERR_INVALID_ACK:
        ESP_DRAM_LOGE(TAG, "Transmit: invalid ack");
        break;
    case ESP_IEEE802154_TX_ERR_COEXIST:
        ESP_DRAM_LOGE(TAG, "Transmit: rejected by the coexist system");
        break;
    case ESP_IEEE802154_TX_ERR_SECURITY:
        ESP_DRAM_LOGE(TAG, "Transmit: invalid sec config");
        break;
    }
}

void IRAM_ATTR esp_ieee802154_receive_done(uint8_t *rx_frame, esp_ieee802154_frame_info_t *frame_info) {
    ieee802154_frame_t *frame = (ieee802154_frame_t *) (rx_frame + 1);

    char seq_num_str[4];
    snprintf(seq_num_str, 4, "%" PRIu8, frame->mhr.seq_num);

    ESP_DRAM_LOGI(TAG, "Frame received, seq_num: %s", seq_num_str);

    ESP_ERROR_CHECK(esp_ieee802154_receive_handle_done(rx_frame));
}

void app_main(void) {
    esp_err_t err = nvs_flash_init();
    if (err == ESP_ERR_NVS_NO_FREE_PAGES || err == ESP_ERR_NVS_NEW_VERSION_FOUND) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        err = nvs_flash_init();
    }
    ESP_ERROR_CHECK(err);

    ESP_ERROR_CHECK(esp_ieee802154_enable());
    ESP_ERROR_CHECK(esp_ieee802154_set_channel(26));
    ESP_ERROR_CHECK(esp_ieee802154_set_txpower(10));
    ESP_ERROR_CHECK(esp_ieee802154_set_promiscuous(false));
    ESP_ERROR_CHECK(esp_ieee802154_set_coordinator(false));
    ESP_ERROR_CHECK(esp_ieee802154_set_pending_mode(ESP_IEEE802154_AUTO_PENDING_DISABLE));
    ESP_ERROR_CHECK(esp_ieee802154_set_cca_mode(ESP_IEEE802154_CCA_MODE_ED));
    ESP_ERROR_CHECK(esp_ieee802154_set_cca_threshold(-70));

    uint16_t pan_id;
    esp_fill_random(&pan_id, sizeof(pan_id));

    ESP_LOGI(TAG, "PAN ID: %" PRIx16, pan_id);
    ESP_ERROR_CHECK(esp_ieee802154_set_panid(pan_id));
    ESP_ERROR_CHECK(esp_ieee802154_set_short_address(0x0001));

    ESP_ERROR_CHECK(esp_ieee802154_set_rx_when_idle(true));
    ESP_ERROR_CHECK(esp_ieee802154_receive());

    xTaskCreate(mesh_proto_worker, "mesh_proto_worker", 20480, &pan_id, 5, NULL);

    fflush(stdout);
}
```

chshu, 2024-07-17:
@igormichalak In ESP 802.15.4 SoCs, the FCS field is calculated by hardware:
- During transmision, the hardware calculates the FCS, and send it over the air right after the MAC payload, so you just need to prepare the length, mac header and mac payload content.
- During receiving, the hardware calculates the FCS of the received frame, and may drop it if the FCS doesn't match, only the valid frames will be received and notified by `esp_ieee802154_receive_done()`. Please note that the FCS field is replaced by RSSI and LQI value of the received frame.

We will enhance the API docs with these details.

igormichalak, 2024-07-17:
I really appreciate the help, thank you!

closed 2024-07-17 by igormichalak