espressif/esp-idf #14177: ESP32-P4 benchmarks

marchingband, 2024-07-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I asked this question on reddit and an esp employee suggested I move it here...

1) Will the P4 have an improved FPU? Are there any benchmarks available?
2) Will the P4 have an improved ADC? Are there any test results available?

thanks!

ginkgm, 2024-07-17:
Hi @marchingband ,

We haven't finish the calibration of P4, so there is no data I can provide for now. After the datasheet is released, you can find the results there. 

@ESP-Marius could you help answer the FPU question?

ESP-Marius, 2024-07-17:
Wasn't able to find any benchmark for the FPUs, but we'll see if we can get some.

At least floating point division performance should be improved now compared to earlier xtensa chips with FPU, since as far as I know it is now supported in HW on P4

ginkgm, 2024-07-22:
It's very likely the ADC accuracy is the same as C6 v0.2.

ESP-Marius, 2024-07-23:
@marchingband here are some numbers from a quick test we did internally:

esp32p4: cpu 360MHz, flash 80MHz
esp32s3: cpu 240MHz, flash 80MHz

### CPU Cycles

| Operation | ESP32-P4 | ESP32-S3 |
|-----------|----------|----------|
| add/sub   | 4        | 5        |
| mul       | 4        | 5        |
| div       | 19       | 53       |
| sqrt()    | 22       | 63       |
| pow()     | 503      | 610      |
| sin()     | 62       | 83       |

### CoreMark-Pro (USE_FP32) (workloads/sec)

| Workload    | ESP32-P4 | ESP32-S3 |
|-------------|----------|----------|
| linear      | 33       | 11       |
| nnet        | 100      | 31       |
| radix_fft   | 1538     | 645      |
| loops       | 80       | 29       |

### Linpack (SF)

| Metric | ESP32-P4 | ESP32-S3 |
|--------|----------|----------|
| kflops | 67405    | 24262    |

Additional Notes
1. float operation is the cpu cycle to evaluate a single floating-point operation. The test is to iterate multiple times and take the average value.
2. For the coremark-pro test item, select several typical applications of floating-point types and specify USE_FP32. Since the original data is large, the data size is reduced in this test to exclude the impact of memory access. It is only for comparing floating-point performance.
3. linpack only chosen to test single-float type.

Bruce297, 2025-01-14:
Hi, P4 calibration have finish on chip v1.0, no improved for now.

ESP-Marius, 2025-02-13:
Closing this one now as the initally requested benchmarks have all been provided.