espressif/esp-idf #14178: ESP32-s3 error w5500 with PSRAM

tresler, 2024-07-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

release/v5.1 bd2b9390ef

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Linux

### How did you build your project?

Other (please specify in More Information)

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-S3-WROOM

### Power Supply used.

External 3.3V

### What is the expected behavior?

ESP32-S3 

### What is the actual behavior?

After few moment I have error:
`w5500.mac: received frame was truncated`
and few more second:
`w5500.mac: emac_w5500_transmit(577): free size (1) < send length (350)`

### Steps to reproduce.

1. Step: Using ESP32-S3 with Arduino 3.0.2, WiFi, Ethernet SPI [W5500](https://docs.wiznet.io/Product/ioModule/W5500-io) and, [sACN](https://github.com/forkineye/ESPAsyncE131) signal and smart LED strip by [FastLED](https://github.com/FastLED). Signal by ethernet running well more then 5 hours. 
2. Step: Use [I2SClockLessLedDriveresp32s3](https://github.com/hpwit/I2SClockLessLedDriveresp32s3) with PSRAM to OPI for quicker parallel output of FastLED and now you have after few moments error of w5500.mac. 
3. Step: Comment ethernet and use only WiFi and everything working fine. 

### Debug Logs.

```plain
w5500.mac: received frame was truncated
w5500.mac: emac_w5500_transmit(577): free size (1) < send length (350)
```
```

### More Information.

I'm trying on Linux with ESP32-Arduino 3.0.2 with IDF v5.1 and ESP32s3-wroom. 

here is my [test code](https://github.com/tresler/test-W5500-with-PSRAM/tree/main).

kostaond, 2024-07-18:
@tresler, do you have your custom designed board with W5500 or do you use some evaluation board?

tresler, 2024-07-18:
> @tresler, do you have your custom designed board with W5500 or do you use some evaluation board?

Thank for response. As I wrote I use [RJ-45 transformer jack with transformer](https://www.tme.eu/Document/8117c2b2a705c42e7c945f12ec20f30f/LMJTAB881243ML.PDF) connected to [W5500-io WIZnet modul](https://docs.wiznet.io/Product/ioModule/W5500-io) connected to ESP32-S3-WROOM.  
[image]

kostaond, 2024-07-18:
Is ESP32 connected to the W5500 module via wires?

tresler, 2024-07-18:
> Is ESP32 connected to the W5500 module via wires?

I have everything connected using my own PCB board.

kostaond, 2024-07-18:
The reason why I asked is it looks like the SPI message is corrupted.... It's usually caused by problems with signal integrity. Long, non-matching length wires are usual suspect. However, it can also happen on PCB when signal path is not properly designed. What is the SPI frequency you use? If it's higher than 16 MHz, try to decrease it.

tresler, 2024-07-18:
I understand. I was try it with 10 MHz, 4 MHz and 1 MHz, unfortunately without big effect. It just moved the error message a few packets further. I still think it has to be something with the memory (not free(buffer);) and with the PSRAM and using the i2sdriver for the led. As I wrote if I using same code only with FastLED without i2cleddriver and PSRAM, everything seems to work. When I use driver and send some sACN packet, I have error message.

kostaond, 2024-07-19:
`w5500.mac: received frame was truncated` should never occur under normal conditions. Therefore, I presume the SPI frame is corrupted. It can be caused by HW (PCB) or the message could theoretically be corrupted by the DMA. Do you observe the issue when you use PSRAM but without i2cleddriver? Or vice versa, do you observe the issue if you use i2cleddriver but without PSRAM? We need to isolate issue as much as possible. Thanks for your help!

tresler, 2024-07-19:
All right. Unfortunately led driver need PSRAM, so I can not run program with driver a without PSRAM. When I set PSRAM to OPI and don't use driver everything seems to work without a problem. When I use i2cledriver, after few packet I have error. If I use driver without eth.h only with wifi or Ethernet_Generic library, I don't see the problem there again.

kostaond, 2024-07-23:
The SPI Ethernet module should work along with other device on the same SPI bus when both drivers use **the same SPI driver**. This is the reason why we recently introduced SPI layer abstraction for our SPI Ethernet module drivers.

@me-no-dev, could you please check the issue if this is the case?

kostaond, 2024-09-04:
@tresler any update about the issue? Do you use the same SPI driver for both SPI devices?

tresler, 2024-09-04:
I cut paths on the board to SD card reader and now ethernet works fine. Unfortunately now I cannot test two separately SPI bus because I don't have any devices for test.

kostaond, 2024-09-04:
I see, so not much to do right now... :/  

I think the following could be the issues:
1) If you use Arduino, be sure the W5500 and SD card reader use the same SPI driver.
2) Make sure that CS is correctly handled for both devices.

As such I'm closing the issue. Feel free to reopen or open a new one if you have more detailed information.

closed 2024-09-04 by kostaond

etleli, 2024-11-27:
This is kinda funny but i have the exact same problem, just a few minor differences. 
I try using artnet to drive leds.
I use the same library, esp32-s3 with psram dev board and a w5500 dev board.

But i dont have a second spi client so that previous solution doesn't work for me.

the w5500 is connected via jumper cables.

BUT 

all works fine and seemingly forever until i plug in the power supply for the leds (12V). After a few seconds i get mostly: received frame was truncated, and some: invalid frame lenght, with a random number and afterwards a unexpected error 0x104.

I have connected the gnd everywhere but i assume the power supply introduces some noise on the gnd and that the w5500 is more sensitive since the esp is not affected or the spi communication is somehow disturbed, but wouldn't that be a problem from the start and not just after a few seconds?

kostaond, 2024-11-28:
@etleli how long are your wires? Have you tried to decrease SPI frequency?

etleli, 2024-11-28:
@kostaond no not yet, i will try that today, they are ca 20 cm, the once that came with the w5500 dev module. I will try shorten them/ solder them to a dev pcb to get them as short as possible too