espressif/esp-idf #14179: My complain about: esp-idf/examples/peripherals/lcd/i2c_oled

sonictl, 2024-07-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

esp-idf version: v5.2.2

// complain about: esp-idf/examples/peripherals/lcd/i2c_oled

// 这个example有几个问题

//   对于使用ssd1306显示的人来讲，其实不需要 lvgl, 但给的例子不够精简，引入了lvgl的支持。这对初学者并不友好。example的设置，首先就是为初学者服务的，专家不需要example也能开发。

//   具体问题：

//   对初学者来说，要使用这个example代码就需要了解ESP-IDF自带的组件和外部组件。esp_lcd是本example需要的ESP-IDF内部组件，已经包含。lvgl却是外部组件，需要初学者学会如何添加 lvgl 这个外部组件。甚至还要求使用者理解什么是port的概念，什么又是 esp_lvgl_port。

//   既然要弃用以往的esp_ssd1306组件，就应该提供一个新的、与之对等的、从精简到复杂的example，而不是另起炉灶，甚至引入外部库。

//   这样操作肯定会增加项目的学习难度，最终于公司产品推广不利。

//   真不知道是怎么策划的这个demo...

suda-morris, 2024-07-15:
Thanks for your valuable suggestion!

Yes, I agree the current LCD example are focus on too many things, including the promotion of the esp_bsp project. Which is good when the esp_bsp is still a new thing to the community. But now, it's a different story, we can see esp-bsp in many projects. So it's time to remove unnecessary dependencies in the esp-idf, and keep the examples there as clean as possible. In fact, the first release of the i2c_oled example does want to illustrate the usage of the "draw_bitmap" function, but later, this function was moved into the lvgl_port...

WRT the deprecation of esp_ssd1306 component, IMO we can refactor it and keep the help functions like "draw_string" "draw_rectangle". @espzav 

BTW, if you want a simpler usage of the SSD1306 usage, you can find some in: https://github.com/espressif/esp-idf/blob/master/components/esp_lcd/test_apps/i2c_lcd/main/test_i2c_lcd_panel.c

sonictl, 2024-07-15:
Thank you for your prompt response. In fact, there is a misunderstanding.

After further study and understanding, I have identified the reason for my confusion.

When I saw i2c_oled, I found that it is a folder within an ESP-IDF internal example, located at examples/peripherals/lcd/i2c_oled.

This path is easy to find, but this i2c_oled (essentially a project) can only be found when creating a new project through the example by searching for 'i2c_oled'. When I was creating a new project, I tried searching with 'peripherals' and 'lcd' but found nothing. So, I was very confused as to why the example project available on GitHub could not be found in the ESP-IDF Examples panel in VSCode.

I suggest that your Examples search should include the names of parent folders. In other words, users should be able to search not only by the example project name but also by its parent elements. For instance, the keyword 'i2c_oled' works, but 'peripherals' and 'lcd' should also be searchable.

sonictl, 2024-07-15:
By the way, i have tested the esp_port_lvgl on both 128x32 and 128x64 oled ssd1306. They does not support 128x32 oled well. Content will be squashed.

LearnerFranky, 2024-09-02:
I have the same idea as you.

I am using the `esp_lcd` component to implement my ssd1306 driver and there is no existing font file that I can use with the `drawbitmap` function.

I think the structure of the different types of LCDs is tightly coupled in the `esp_lcd` component and i have try to remove parts not useful for me but it failed.

Hopefully espressif will fix these experience “bugs” soon.

closed 2024-11-04 by espressif-bot