espressif/esp-idf #14185: esp_http_client_init() gives "HTTP_CLIENT: Error parse url"

serdartoren, 2024-07-14:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.4-dev-1388-g5ca9f2a49a

### Espressif SoC revision.

esp32s3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

esp_http_client_init() function can parse links in this structure `http://api_v2.torenelektronik.xyz/yedek/hello_world.bin`

### What is the actual behavior?

But esp_http_client_init() function works properly only with such url's `http://torenelektronik.com.tr/yedek/hello_world.bin`

### Steps to reproduce.

1. Open the native_ota_example example
2. Adjust your configs to allow http
3.Before working with esp32, verify that link 2 can be downloaded via your PC browser.
4. You will see that you get a parse url error in the link containing ".xyz"

### Debug Logs.

```plain
I (3754) wifi:new:<10,0>, old:<1,0>, ap:<255,255>, sta:<10,0>, prof:1, snd_ch_cfg:0x0

I (3754) wifi:state: init -> auth (0xb0)

I (3764) wifi:state: auth -> assoc (0x0)

I (3774) wifi:state: assoc -> run (0x10)

I (3784) wifi:<ba-add>idx:0 (ifx:0, 5c:63:bf:b5:fa:7c), tid:0, ssn:0, winSize:64

I (3814) wifi:connected with TurkTelekom_T88BA, aid = 1, channel 10, BW20, bssid = 5c:63:bf:b5:fa:7c

I (3814) wifi:security: WPA2-PSK, phy: bgn, rssi: -80

I (3814) wifi:pm start, type: 1

I (3824) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us

I (3874) wifi:AP's beacon interval = 102400 us, DTIM period = 1

I (4894) example_connect: Got IPv6 event: Interface "example_netif_sta" address: fe80:0000:0000:0000:26dc:c3ff:fea7:1f68, type: ESP_IP6_ADDR_IS_LINK_LOCAL

I (5334) esp_netif_handlers: example_netif_sta ip: 192.168.1.101, mask: 255.255.255.0, gw: 192.168.1.1

I (5334) example_connect: Got IPv4 event: Interface "example_netif_sta" address: 192.168.1.101

I (5344) example_common: Connected to example_netif_sta

I (5344) example_common: - IPv4 address: 192.168.1.101,

I (5354) example_common: - IPv6 address: fe80:0000:0000:0000:26dc:c3ff:fea7:1f68, type: ESP_IP6_ADDR_IS_LINK_LOCAL

I (5364) wifi:Set ps type: 0, coexist: 0

I (5364) native_ota_example: Starting OTA example task

I (5374) native_ota_example: Running partition type 0 subtype 0 (offset 0x00010000)

E (5384) HTTP_CLIENT: Error parse url https://api_v2.torenelektronik.xyz/yedek/hello_world.bin

E (5394) HTTP_CLIENT: Failed to set URL

E (5394) native_ota_example: Failed to initialise HTTP connection

E (5404) native_ota_example: Exiting task due to fatal error...

I (5404) main_task: Returned from app_main()
```

### More Information.

I think the esp_http_client_init() function should be able to parse these links as well.

Have a nice day!
-Serdar

mahavirj, 2024-07-15:
Relevant https://github.com/espressif/esp-idf/issues/13459

nileshkale123, 2024-07-23:
Closing this issue. Please feel free to reopen if the issue persists.

closed 2024-07-23 by nileshkale123