espressif/esp-idf #14187: simple_ota_example  (esp_https_ota: Mismatch chip id, expected 0, found 18242)

leehakgun, 2024-07-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf v5.2.2

### Espressif SoC revision.

ESP32 Revision 1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP_EYE

### Power Supply used.

USB

### What is the expected behavior?

Firmware expected an update as described in the example

### What is the actual behavior?

E (8826) esp_https_ota: Mismatch chip id, expected 0, found 18242
E (8836) simple_ota_example: Firmware upgrade failed
This error has occurred

### Steps to reproduce.

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_event.h"
#include "esp_log.h"
#include "esp_ota_ops.h"
#include "esp_http_client.h"
#include "esp_https_ota.h"
#include "protocol_examples_common.h"
#include "string.h"
#ifdef CONFIG_EXAMPLE_USE_CERT_BUNDLE
#include "esp_crt_bundle.h"
#endif

#include "nvs.h"
#include "nvs_flash.h"
#include "protocol_examples_common.h"
#include <sys/socket.h>
#if CONFIG_EXAMPLE_CONNECT_WIFI
#include "esp_wifi.h"
#endif

#define HASH_LEN 32

[... 655 lines cut ...]
I (8706) mbedtls: ssl_msg.c:2318 ssl->f_recv(_timeout)() returned 4424 (-0xffffeeb8)

I (8716) mbedtls: ssl_msg.c:2340 <= fetch input

I (8756) mbedtls: ssl_msg.c:1510 => decrypt buf

I (8766) mbedtls: ssl_msg.c:2121 <= decrypt buf

I (8796) mbedtls: ssl_msg.c:4261 <= read record

I (8796) mbedtls: ssl_msg.c:5942 <= read

I (8796) esp_https_ota: Starting OTA...
I (8796) esp_https_ota: Writing to partition subtype 16 at offset 0x110000
I (8806) mbedtls: ssl_msg.c:5782 => read

I (8816) mbedtls: ssl_msg.c:5942 <= read

I (8816) mbedtls: ssl_msg.c:5782 => read

I (8816) mbedtls: ssl_msg.c:5942 <= read

E (8826) esp_https_ota: Mismatch chip id, expected 0, found 18242
E (8836) simple_ota_example: Firmware upgrade failed
```

nileshkale123, 2024-07-23:
Could you please verify if the OTA image is downloadable from a browser on a device within the same network?

closed 2024-08-05 by nileshkale123

AxelLin, 2024-08-05:
@nileshkale123 What was fixed?

nileshkale123, 2024-08-05:
@AxelLin , we were unable to reproduce this issue on our end, and we have not received the requested files from @leehakgun . To confirm, we tested this example ourselves and it executed successfully. Therefore, we have closed this issue.

If you encounter this issue again, please feel free to reopen it and provide the necessary files and details about your work environment.