espressif/esp-idf #14188: ESP32-S3 in USB mode can't use FTP to save/create file to it's flash or SD Card

TransistorTun, 2024-07-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

esp-idf v5.2.2

### Espressif SoC revision.

ESP32 Revision 1

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-S3

### Power Supply used.

USB

### What is the expected behavior?

I use my computer as a master and create/save files to ESP32's flash/SD Card by FTP. And I can download this files to the secondary device or save these files on flash/ SD Card.

### What is the actual behavior?

When I don't connect the ESP(USB mode) to the secondary, I can use FTP to save/create files on esp's flash/ SD Card.  But when I connect the esp (USB mode) to the secondary device, I can't use FTP to save/create files on esp's flash/ SD Card anymore.

### Steps to reproduce.

/*********************
 *      INCLUDES
 *********************/

#include <stdio.h>
#include <inttypes.h>
#include <string.h>
#include <sys/unistd.h>
#include <sys/stat.h>

#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "freertos/event_groups.h"

#include "esp_system.h"
#include "esp_log.h"
#include "esp_event.h"
#include "esp_err.h"
#include "esp_sntp.h"
#include "esp_mac.h" // for MACSTR
#include "esp_partition.h"
#include "esp_check.h"

#include "lwip/dns.h"

#include "mdns.h"
[... 522 lines cut ...]
I (34695) [Ftp]: open_child: / + -a
I (34695) [Ftp]: open_child, New pwd: /-a
I (34695) [Ftp]: ftp_open_dir_for_listing path=[/-a] MOUNT_POINT=[/data]
I (34695) [Ftp]: ftp_open_dir_for_listing: /data/-a
I (34705) [Ftp]: Send reply: [550 ]
I (34725) [Ftp]: Send reply: OK (6)
I (34725) [Ftp]: remove_fname_from_path: /-a - -a
I (34725) [Ftp]: remove_fname_from_path: New pwd: /
I (34735) [Ftp]: CMD: TYPE
I (34735) [Ftp]: Send reply: [200 ]
I (34755) [Ftp]: Send reply: OK (6)
I (34775) [Ftp]: CMD: PASV
I (34775) [Ftp]: Data socket created
I (34775) [Ftp]: Send reply: [227 (192,168,1,21,7,232)]
I (34795) [Ftp]: Send reply: OK (26)
I (34825) [Ftp]: Data socket connected
I (35425) [Ftp]: CMD: LIST
I (35425) [Ftp]: open_child: / +
I (35425) [Ftp]: open_child, New pwd: /
I (35425) [Ftp]: ftp_open_dir_for_listing path=[/] MOUNT_POINT=[/data]
I (35425) [Ftp]: ftp_open_dir_for_listing: /data/
I (35435) [Ftp]: Send reply: [550 ]
I (35455) [Ftp]: Send reply: OK (6)
I (35455) [Ftp]: remove_fname_from_path: / -
```

peter-marcisovsky, 2024-07-15:
Hi @TransistorTun Thank you for reporting the issue.

I am trying to understand your setup.
Which development board are you using?
What is the "secondary" device in your setup?
You are using you PC as a host, from which you are sending files over FTP to an esp32s3, and the esp32s3 is receiving those data and saving them to it's flash memory, to the SD card. The esp32s3 has some sort of "secondary" device connected to it over USB. 
If the "secondary' device is not connected to you esp32s3, you are able to send and store some data, you have sent over FTP from you PC to the esp32s3's flash or SD card.
But the problem occurs, once you connect your "secondary" device to the the esp32s3. Then you are not able to save the data, you have sent over FTP from your host PC to the esp32s3.

Did I get it it right?

TransistorTun, 2024-07-15:
Yes, I am having this problem

peter-marcisovsky, 2024-07-15:
Could you please answer to the questions I have stated? 
I would really help me determine the problem you are facing. Thank you.

TransistorTun, 2024-07-15:
1. I am using ESP32-S3 N8R2 Dual Type-C.
2. I am using this like a USB MSC Wireless Disk. For example, I am connecting esp32-s3 (USB mode) in my CNC machine, and I want to send my file from my computer to this USB wireless by FTP. 
When I built this project, I had the problem that I described.

TransistorTun, 2024-07-15:
Here is the console when I dont't connect to the USB 
[image]
Here is when I connect the USB to the device, I can see the disk E
[image]
But when I create new file by FTP, I have error
[image]

D (707) esp_netif_objects: esp_netif_add_to_list_unsafe netif added successfully (total netifs: 1)
D (717) esp_netif_lwip: call api in lwip: ret=0x0, give sem
I (727) pp: pp rom version: e7ae62f
I (727) net80211: net80211 rom version: e7ae62f
D (737) nvs: nvs_open_from_partition misc 1
I (747) wifi:wifi driver task: 3fcabd24, prio:23, stack:6656, core=0
D (747) efuse: In EFUSE_BLK1__DATA1_REG is used 8 bits starting with 8 bit
D (747) efuse: In EFUSE_BLK1__DATA1_REG is used 8 bits starting with 0 bit
D (757) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 24 bit
D (767) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 16 bit
D (767) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 8 bit
D (777) efuse: In EFUSE_BLK1__DATA0_REG is used 8 bits starting with 0 bit
D (787) nvs: nvs_open_from_partition nvs.net80211 1
D (787) nvs: nvs_get opmode 1
D (797) nvs: nvs_get_str_or_blob sta.ssid
D (797) nvs: nvs_get sta.authmode 1
D (797) nvs: nvs_get_str_or_blob sta.pswd
D (807) nvs: nvs_get_str_or_blob sta.pmk
D (807) nvs: nvs_get sta.chan 1
D (817) nvs: nvs_get auto.conn 1
D (817) nvs: nvs_get bssid.set 1
D (817) nvs: nvs_get_str_or_blob sta.bssid
D (827) nvs: nvs_get sta.lis_intval 2
D (827) nvs: nvs_get sta.phym 1
D (827) nvs: nvs_get sta.phybw 1
D (837) nvs: nvs_get_str_or_blob sta.apsw
D (837) nvs: nvs_get_str_or_blob sta.apinfo
D (847) nvs: nvs_get sta.scan_method 1
D (847) nvs: nvs_get sta.sort_method 1
D (847) nvs: nvs_get sta.minrssi 1
D (857) nvs: nvs_get sta.minauth 1
D (857) nvs: nvs_get sta.pmf_e 1
D (857) nvs: nvs_get sta.pmf_r 1
D (867) nvs: nvs_get sta.btm_e 1
D (867) nvs: nvs_get sta.rrm_e 1
D (867) nvs: nvs_get sta.mbo_e 1
D (877) nvs: nvs_get_str_or_blob ap.ssid
D (877) nvs: nvs_get_str_or_blob ap.passwd
D (877) nvs: nvs_get_str_or_blob ap.pmk
D (887) nvs: nvs_get ap.chan 1
D (887) nvs: nvs_get ap.authmode 1
D (887) nvs: nvs_get ap.hidden 1
D (897) nvs: nvs_get ap.max.conn 1
D (897) nvs: nvs_get bcn.interval 2
D (897) nvs: nvs_get ap.phym 1
D (907) nvs: nvs_get ap.phybw 1
D (907) nvs: nvs_get ap.sndchan 1
D (907) nvs: nvs_get ap.pmf_e 1
D (917) nvs: nvs_get ap.pmf_r 1
D (917) nvs: nvs_get ap.p_cipher 1
D (917) nvs: nvs_get lorate 1
D (927) nvs: nvs_get_str_or_blob country
D (927) nvs: nvs_get ap.ftm_r 1
D (927) nvs: nvs_get sta.ft 1
D (937) nvs: nvs_get sta.owe 1
D (937) nvs: nvs_get sta.trans_d 1
D (937) nvs: nvs_get sta.sae_h2e 1
D (947) nvs: nvs_get sta.sae_pk_mode 1
D (947) nvs: nvs_get sta.bss_retry 1
D (957) nvs: nvs_get_str_or_blob sta.owe_data
D (957) nvs: nvs_get sta.he_dcm 1
D (957) nvs: nvs_get sta.he_dcm_c_tx 1
D (967) nvs: nvs_get sta.he_dcm_c_rx 1
[... 899 lines cut ...]
D (284567) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (284567) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (284567) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (284667) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (284767) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (284867) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (285487) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
I (291177) [Ftp]: CMD: TYPE
I (291177) [Ftp]: Send reply: [200 ]
I (291197) [Ftp]: Send reply: OK (6)
D (293477) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (295217) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (295517) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (307507) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (308447) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (309657) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (309767) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (310577) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (310667) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (313547) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
D (319507) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358
I (321387) [Ftp]: CMD: TYPE
I (321387) [Ftp]: Send reply: [200 ]
I (321407) [Ftp]: Send reply: OK (6)
D (321427) esp_netif_lwip: esp_netif_get_ip_info esp_netif:0x3fca9358

peter-marcisovsky, 2024-07-15:
From what you have provided, it looks like that your problem is related to a HW issue. 
Have you done any modification on the development board you are using?

TransistorTun, 2024-07-15:
No, I haven't done any modification on my development board. I have just buy that board to build this project

peter-marcisovsky, 2024-07-15:
Okay then, what is happening here is: 
There is a mutual exclusivity of owning the File System (FS) between the USB Host (your CNC machine) and the application (code running in your the esp32s3). Only one of them can have an access to the File system at the time.

With esp32s3 board NOT connected to the USB Host, the owner of the File System is your application, thus you can access the File system on the esp32s3 over the FTP.

Once you connect the esp32s3 board to the USB Host, the owner of the File System changes from your application, to the USB Host. Thus you (your code) are not allowed to access the File System on the esp32s3 over the FTP.

In your log, you can see the following lines:
`[usb]: Storage mounted to application: Yes`  - The FS is owned by the app
`[usb]: Storage mounted to application: No` - The FS is owned by the USB Host
Indicating, who owns the File system

To be able to monitor who owns the FS by hardware, one must implement a so called [VBUS Monitor](https://docs.espressif.com/projects/esp-idf/en/stable/esp32s3/api-reference/peripherals/usb_device.html#self-powered-device) on the USB Device side (on your devkit) And update the `tinyusb_config_t`, according to the documentation. After that you will be able to determine who owns the FS.

You can check the functionality of the Vbus monitor by running the `tusb_msc` example (with the changes from the documentation) and simply plug and unplug the USB Device. You should see a following output:

```
I (9948) example_main: Storage mounted to application: Yes
I (12358) example_main: Storage mounted to application: No
I (14628) example_main: Storage mounted to application: Yes
I (16708) example_main: Storage mounted to application: No

```

Now, you will be able to determine who owns the FS. Based on this information, you can follow up and change the logic in your code accordingly. Thus you should access the FS (over the FTP) Only when the File System is owned by the App, not by the USB Host

TransistorTun, 2024-07-16:
Thank you very much for helping and replying me. I'll try it and If it has problem, I would ask you later.

TransistorTun, 2024-07-19:
Hi @peter-marcisovsky , after debugging and fixing, I realized that when I don't plug the USB (esp32), the SD Card is mounted at an address (example /data). But the address is changed when I plug the USB (my esp32) into the device. I don't know how to fix it, but I know that when I plug in the USB, there is an interrupt that makes the USB unmount from esp32 and mount the USB to the device. So that, esp32 can't connect and make changes on the SD Card.

peter-marcisovsky, 2024-08-27:
Hi @TransistorTun have you resolved your problem? 
Feel free to close the issue, if you don't have any other questions, or if the problem has been resolved.
Thank you.