espressif/esp-idf #14189: Finding out the firmware version from a coredump elf

0xFEEDC0DE64, 2024-07-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We have collected hundrets of millions of coredump elf files from our ESP32 products around the globe and every time we receive another crash report, we have a hard time finding out which firmware this resulted from (we do some heuristics on the database on what the device reported earlier using websockets).

Is there some easy way to find out the exact project_name and version of the firmware from inside the coredump elf?

Or can we somehow embed custom information like customer name, serial number, firmware project_name and version directly into the elf?

erhankur, 2024-07-15:
Core dump ELF files contain the SHA-256 sum of the IDF applications. You can get it using the `esp_core_dump_get_summary` API. Does this work for you?

erhankur, 2025-01-20:
Closed due to inactivity.

closed 2025-01-20 by erhankur

mifarelight, 2026-08-20:
> > Or can we somehow embed custom information like customer name, serial number, firmware project_name and version directly into the elf?
> 
> You can also define a variable with `COREDUMP_DRAM_ATTR`, then this variable will be available from the coredump. (You would still need to run GDB to extract it.)
> 
> For example, something like this might work:
> 
> ```
> static COREDUMP_DRAM_ATTR esp_app_desc_t app_desc;
> 
> void app_main() {
>    app_desc = *esp_app_get_description();
> }
> ```
> 
> This would store an `app_desc` variable in the core dump, and initialize it to the actual app description on startup.

Sorry for disturbing you.

I am using VSCode+PlatformIO+Arduino to develop some ESP32 things.

I want to use the "Compile time" and "Compile date" to indicate the firmware version of my ESP32 Application. It returns a very old date and time when I use "const esp_app_desc_t *app_desc = esp_app_get_description();" to get the descriptions.

So I opened the firmware.bin and searched through the whole file then found two Data/Time informations. It is clear that esp_app_get_description() returns an outdated timestamp (19:50:46 Feb 11 2026 here). Is there any function to get the up-to-date timestamp (Software Info: 15:17:57 Aug 20 2026)?

[image]
[image]