espressif/esp-idf #14190: BLE Mesh: Send Data at High Data Rate (Error: Out of network buffers)

Florian-ger, 2024-07-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello everyone,

I want to send data at a high data rate over BLE mesh. Theoretically, BLE mesh supports a data rate of 2 Mbps when using the 2M PHY. The practical rate is, of course, lower, but in my application, I can't even achieve a data rate of 1 kbps. The data packets with a size of 7 bytes are sent in the following loop with a delay of 10 ms, which corresponds to a data rate of approximately 0.7-1 kbps. (The transmission works with a delay of 100 ms and there is no error message)

**Code Example:**

```c
uint8_t data[5] = {1, 2, 3, 4, 5};

for (int cycle = 0; cycle < 300; cycle++) {
    esp_ble_mesh_msg_ctx_t ctx = {0};
    ctx.net_idx = prov_key.net_idx;
    ctx.app_idx = prov_key.app_idx;
    ctx.addr = 0x00BE;
    ctx.send_ttl = 5;
    esp_ble_mesh_client_model_send_msg(vendor_client.model, &ctx, MODEL_OP_SEND, sizeof(data), data, 0, false, ROLE_NODE);
    vTaskDelay(10 / portTICK_PERIOD_MS);
}
```

**Issue:**

When sending 300 packets, I receive the error message after 256 packets: `BLE_MESH: Out of network buffers`. I have set the network buffers to the maximum value of 256 in the sdkconfig. This means that the network buffer is not being emptied quickly enough or the data is not being sent quickly enough. How can I configure my sdkconfig or code to achieve a higher data rate? What is the maximum transmission rate of the ESP32-C6?

**Environment:**

- **ESP-IDF version:** v5.3 (also tested with v5.1.4)
- **ESP board:** ESP32-C6-Dev-Kit-N8

[sdkconfig.txt](https://github.com/user-attachments/files/16234394/sdkconfig.txt)

forx157, 2024-07-17:
Hi, @Florian-ger
Evaluating the BLE rate with BLE Mesh is not accurate because BLE Mesh by default retransmits a broadcast packet twice at 20-30ms intervals, which means a broadcast packet will take up about 60-90ms, which is also the reason why when you use 10ms interval to advertising the data, you will get the message `BLE_MESH: Out of network buffers`.

Florian-ger, 2024-07-17:
Hi @forx157,

Thank you for your  response.

By retransmission of packets, do you mean the `net_transmit` setting in the configuration client? I have set the network retransmit count to 0 there. Additionally, I only send the packets to a unicast address (0x00BE, the server). Shouldn't it be quicker then?

I understand that achieving a transmission rate of 100 kbps is not possible, but I thought that sending packets would be much faster than a 100 ms delay.

forx157, 2024-07-18:
@Florian-ger 
I have tested it as you explained and there is no abnormality, can you share your test log with us after changing the log level of BLE Mesh to INFO level?

You can change the log level of BLE Mesh in menuconfig by following the path below:
`(Top) → Component config → ESP BLE Mesh Support → BLE Mesh STACK DEBUG LOG LEVEL`

Florian-ger, 2024-07-19:
If I set the BLE Mesh Stack Debug Log Level to Info, the program stops after 38 sent messages and does not print any further information on the serial monitor:

```
I (23) boot: ESP-IDF v5.3-rc1-dirty 2nd stage bootloader
I (24) boot: compile time Jul 19 2024 10:39:53
I (24) boot: chip revision: v0.0
I (27) boot.esp32c6: SPI Speed      : 80MHz
I (32) boot.esp32c6: SPI Mode       : DIO
I (37) boot.esp32c6: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (47) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (58) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (65) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (73) boot:  2 factory          factory app      00 00 00010000 00177000
I (80) boot: End of partition table
I (84) esp_image: segment 0: paddr=00010020 vaddr=420e8020 size=3b204h (242180) map
I (143) esp_image: segment 1: paddr=0004b22c vaddr=40800000 size=04dech ( 19948) load
I (148) esp_image: segment 2: paddr=00050020 vaddr=42000020 size=e32d8h (930520) map
I (341) esp_image: segment 3: paddr=00133300 vaddr=40804dec size=0d244h ( 53828) load
I (354) esp_image: segment 4: paddr=0014054c vaddr=40812030 size=02894h ( 10388) load
I (361) boot: Loaded app from partition at offset 0x10000
I (362) boot: Disabling RNG early entropy source...
I (374) cpu_start: Unicore app
I (383) cpu_start: Pro cpu start user code
I (383) cpu_start: cpu freq: 160000000 Hz
I (384) app_init: Application information:
I (386) app_init: Project name:     ble-mesh
I (392) app_init: App version:      1
I (396) app_init: Compile time:     Jul 19 2024 10:52:00
I (402) app_init: ELF file SHA256:  147875c38...
I (407) app_init: ESP-IDF:          v5.3-rc1-dirty
I (413) efuse_init: Min chip rev:     v0.0
I (418) efuse_init: Max chip rev:     v0.99 
I (422) efuse_init: Chip rev:         v0.0
I (427) heap_init: Initializing. RAM available for dynamic allocation:
I (434) heap_init: At 4081E800 len 0005DE10 (375 KiB): RAM
I (441) heap_init: At 4087C610 len 00002F54 (11 KiB): RAM
I (447) heap_init: At 50000000 len 00003FE8 (15 KiB): RTCRAM
I (454) spi_flash: detected chip: generic
I (458) spi_flash: flash io: dio
W (462) spi_flash: Detected size(8192k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (475) sleep: Configure to isolate all GPIO pins in sleep state
I (482) sleep: Enable automatic switching of GPIO sleep configuration
I (489) main_task: Started on CPU0
I (503) main_task: Calling app_main()
I (503) MAIN: Initializing...
I (515) BLE_INIT: Using main XTAL as clock source
I (520) BLE_INIT: ble controller commit:[392c824]
I (520) phy_init: phy_version 310,dde1ba9,Jun  4 2024,16:38:11
I (583) phy: libbtbb version: 04952fd, Jun  4 2024, 16:38:26
I (597) BT: Tx power level: -15 dBm
I (597) BT: Bluetooth address: ca:44:63:32:20:00
Bluetooth Mesh v1.1 commit: [10f7fdc1a9]
I (600) BLE_MESH: Relay not supported
I (603) BLE_MESH: Friend not supported
W (663) BLE_MESH: bt_mesh_init, Already
I (668) MESH_CLIENT: ESP_BLE_MESH_NODE_PROV_ENABLE_COMP_EVT, err_code 0
I (669) MESH_CLIENT: BLE Mesh Node initialized
I (670) main_task: Returned from app_main()
I (3446) MESH_CLIENT: ESP_BLE_MESH_NODE_PROV_LINK_OPEN_EVT, bearer PB-GATT
I (5056) BLE_MESH: Algorithm:   0x00
I (5057) BLE_MESH: Public Key:  0x00
I (5057) BLE_MESH: Auth Method: 0x00
I (5058) BLE_MESH: Auth Action: 0x00
I (5063) BLE_MESH: Auth Size:   0x00
I (5327) BLE_MESH: Remote Random: 2cfa5ad3284ee02c4794c5971c5ee3ea
I (5420) BLE_MESH: Primary Element: 0x0175
I (5421) BLE_MESH: net_idx 0x0000 flags 0x00 iv_index 0x0000
I (5421) BLE_MESH: DevKey 707bab896b6fdd2793dc143aa4ad9bf1
[... 334 lines cut ...]
E (26977) BLE_MESH: Out of network buffers
E (26977) BLE_MESH: Failed to send client message 0x00c002e5
E (26987) BLE_MESH: Out of network buffers
E (26988) BLE_MESH: Failed to send client message 0x00c002e5
E (27008) BLE_MESH: Out of network buffers
E (27008) BLE_MESH: Failed to send client message 0x00c002e5
E (27018) BLE_MESH: Out of network buffers
E (27021) BLE_MESH: Failed to send client message 0x00c002e5
E (27028) BLE_MESH: Out of network buffers
E (27033) BLE_MESH: Failed to send client message 0x00c002e5
E (27045) BLE_MESH: Out of network buffers
E (27050) BLE_MESH: Failed to send client message 0x00c002e5
E (27062) BLE_MESH: Out of network buffers
E (27067) BLE_MESH: Failed to send client message 0x00c002e5
E (27079) BLE_MESH: Out of network buffers
E (27083) BLE_MESH: Failed to send client message 0x00c002e5
E (27096) BLE_MESH: Out of network buffers
E (27100) BLE_MESH: Failed to send client message 0x00c002e5
E (27107) BLE_MESH: Out of network buffers
I (27108) MESH_SEND: Stop sending.
E (27115) BLE_MESH: Failed to send client message 0x00c002e5
E (27128) BLE_MESH: Out of network buffers
E (27132) BLE_MESH: Failed to send client message 0x00c002e5

```

forx157, 2024-07-22:
Hi, @Florian-ger 
Since I can't reproduce your results on my end, I need to trouble you to do some more testing:
You can use the code in example/bluetooth/esp_ble_mesh/vendor_model to test if you can complete the advertising of 300 messages, if the test fails, please share the INFO level log with me.

Secondly can you easily tell me what you did in your test code? Or is there any way in which I can reproduce the problem you are having.

Florian-ger, 2024-07-24:
Hi @forx157 ,
Thank you for you help.
It works now with the vendor example. I don't know what was wrong with my code

closed 2024-07-24 by Florian-ger