espressif/esp-idf #14191: Idf issue

guavuavcycayfayqvqu, 2024-07-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

How to fix it.  
[image]

louielauanu, 2024-07-16:
Hi @guavuavcycayfayqvqu, can you please provide more information about your goal, and the issue that you're facing? If you are failing on building an ESP-IDF project on Windows, have you tried to build it with IDF tool (see https://docs.espressif.com/projects/esp-idf/en/stable/esp32/get-started/windows-setup.html)?

ESP-Marius, 2024-08-08:
Closing due to lack of activity, feel free to reopen if there is still any outstanding esp-idf related issues.

closed 2024-08-08 by ESP-Marius