espressif/esp-idf #14192: Non-Volatile Storage Library implementation crashes randomically when call functions nvs_get_blob and nvs_get_str.

guilhermep-git, 2024-07-15:
Hello.
I’m facing the same problem with nvs_get_blob and nvs_get_str, related in #8271.
I’m using the workaround proposed by Igrr, that is used in API Application Example also: https://docs.espressif.com/projects/esp-idf/en/v5.0.6/esp32/api-reference/storage/nvs_flash.html#application-example

## Environment:
Target: ESP32
IDF version v5.0.6
Build System: Eclipse IDE
Operating System: Windows 10
Python 3.11.1

# Debug Logs
## Backtrace: 
0x4008b3eb:0x3fff2410 0x4008c227:0x3fff2440 0x4008c22f:0x3fff2460 0x4008fcc9:0x3fff2480 0x4008bdfd:0x3fff24a0 0x4010c8d9:0x3fff24e0 0x40110926:0x3fff2510 0x40194141:0x3fff2530 0x401114fe:0x3fff2550 0x4010fbbd:0x3fff25b0 0x4010ff86:0x3fff2600 0x401100a5:0x3fff2680 0x40110799:0x3fff26f0 0x4010f509:0x3fff2710 0x4010f8d1:0x3fff2750 0x400d74ee:0x3fff2770 0x400df6e0:0x3fff2830

## Files: 
0x4008b3eb: spi_flash_disable_interrupts_caches_and_other_cpu at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/spi_flash/cache_utils.c:173 (discriminator 1)
0x4008c227: cache_disable at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/spi_flash/spi_flash_os_func_app.c:74
0x4008c22f: spi1_start at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/spi_flash/spi_flash_os_func_app.c:114
0x4008fcc9: spiflash_start_default at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/spi_flash/esp_flash_api.c:123
0x4008bdfd: esp_flash_read at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/spi_flash/esp_flash_api.c:831
0x4010c8d9: esp_partition_read at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/esp_partition/partition_target.c:38
0x40110926: nvs::NVSPartition::read(unsigned int, void*, unsigned int) at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/nvs_flash/src/nvs_partition.cpp:45
 (inlined by) nvs::NVSPartition::read(unsigned int, void*, unsigned int) at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/nvs_flash/src/nvs_partition.cpp:39
0x40194141: nvs::Page::readEntry(unsigned int, nvs::Item&) const at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/nvs_flash/src/nvs_page.cpp:879
0x401114fe: nvs::Page::findItem(unsigned char, nvs::ItemType, char const*, unsigned int&, nvs::Item&, unsigned char, nvs::VerOffset) at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/nvs_flash/src/nvs_page.cpp:929
 (inlined by) nvs::Page::findItem(unsigned char, nvs::ItemType, char const*, unsigned int&, nvs::Item&, unsigned char, nvs::VerOffset) at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/nvs_flash/src/nvs_page.cpp:886
0x4010fbbd: nvs::Storage::findItem(unsigned char, nvs::ItemType, char const*, nvs::Page*&, nvs::Item&, unsigned char, nvs::VerOffset) at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/nvs_flash/src/nvs_storage.cpp:163
0x4010ff86: nvs::Storage::readMultiPageBlob(unsigned char, char const*, void*, unsigned int) at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/nvs_flash/src/nvs_storage.cpp:463
0x401100a5: nvs::Storage::readItem(unsigned char, nvs::ItemType, char const*, void*, unsigned int) at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/nvs_flash/src/nvs_storage.cpp:550
0x40110799: nvs::NVSHandleSimple::get_typed_item(nvs::ItemType, char const*, void*, unsigned int) at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/nvs_flash/src/nvs_handle_simple.cpp:36
 (inlined by) nvs::NVSHandleSimple::get_typed_item(nvs::ItemType, char const*, void*, unsigned int) at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/nvs_flash/src/nvs_handle_simple.cpp:32
0x4010f509: nvs_get_str_or_blob(unsigned long, nvs::ItemType, char const*, void*, unsigned int*) at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/nvs_flash/src/nvs_api.cpp:510
0x4010f8d1: nvs_get_blob at C:/Users/guilhermepiedade/esp-idf-v5.0.6/esp-idf/components/nvs_flash/src/nvs_api.cpp:520
0x400d74ee: getPushIdStorageNVS at D:/Projetos/.../main/storageNVS.c:313
0x400df6e0: taskNotification at D:/Projetos/.../main/notification.c:118

The last two files refer to my application.

This problem happens randomically. 
Is there some solution?

guilhermep-git, 2024-09-19:
Hi @rrtandler 
Thanks for your attention.
Ok. I will share a part of my code with required information.
[image]

Recently the Espressif released 5.0.7 version, so I upgraded my IDF from 5.0.6 to 5.0.7. I realized some tests and apparently the problem was solved.

rrtandler, 2024-09-19:
Hi @guilhermep-git  - at the first look, the code looks good. To make things even more robust, I would just add one condition after
`uint8_t *data_flash = malloc(required_size);` checking the pointer for NULL and continuing only in case the malloc really was able to allocate the memory.

guilhermep-git, 2024-09-19:
Hi @rrtandler,
Thank you for your analysis and suggestion.
But about API bug, according to debug logs and backtrace above, it was possible to identify some cause of the crash?
The Changelog for 5.0.7 version point to as Major Bug Fixes:
SPI Flash: Fixed stuck during spi_flash operation when a task was not pinned to a certain CPU. (f0c2da6)
This bug fix is the solution for crashes in 5.0.6 version?

rrtandler, 2024-09-20:
Hi @guilhermep-git 
The [f0c2da6](https://github.com/espressif/esp-idf/commit/f0c2da6dc49c08052164ad3044bab85b809a7e29) had changed the way the spi_flash driver uses CPU on multi-core variants of ESP32. While the aim was to increase performance, the side effect was also the bug fixing of stuck spi_flash operations and subsequent interference with watchdog. Sorry for delayed back-port to the 5.0.x branch.

guilhermep-git, 2024-09-20:
Hi @rrtandler 
Thank you for your support.

closed 2024-09-20 by guilhermep-git