espressif/esp-idf #14194: NO OPTION TO CHANGE THE TX POWER OF BLE GATTs IN ESP32H2

MR-VENOM, 2024-07-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am unable to find the CONFIG_BT_CTRL_DFT_TX_POWER_LEVEL_ config to change the TX power in H2 menu config and there is no mention on the documentation showing how to chamge the db level in H2.

zhaoweiliang2021, 2024-07-17:
@MR-VENOM  Hi,
You can change the value of BLE_LL_TX_PWR_DBM_N in the esp32h2/esp_bt_cfg.h file to modify the default tx power level. 
The specific levels can be found in the definition of esp_power_level_t in esp32h2/include/esp_bt.h. 
As for the function of configuring the default TX power level through config options, we will support and integrate it as soon as possible. 
Thanks.

zhaoweiliang2021, 2024-08-01:
@MR-VENOM Hi,
The commit ceda7d04e4412c25ef83afb000b7108d2864e6dc in master branch provides this function， please check it out , thanks.

closed 2024-09-06 by weiyuhannnn