espressif/esp-idf #14195: Trying to do Bootloader update OTA

ammaree, 2024-07-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

We have a large number of devices deployed during late 2017 and early 2018 using very old boot loaders.
Our firmware and hardware has been very reliable and the last upgrade of these devices was done in early 2019.
We are currently investigating a process to try and upgrade these boot loaders remotely using a modified OTA.
None of these devices have any level of flash encryption enabled

Steps we believe are required are as follows:
1. Check if new bootloader is available
2. If available, download as "bootloader.bin" to LittleFS storage
3. If download and file write successful, read existing flash to LittleFS as "bootloader.old"
4. Write "bootloader.bin" to bootloader at address 0x1000
5. If write successful, reboot using new boot loader
6. If not successful, try restore "bootloader.old" to 0x1000 then reboot

At a later stage, when all devices have been upgraded to latest bootloader and firmware we would want to start a process to enable bootloader checksum on these device and automatically calculate and update the initial checksum in the field

Steps 1 to 3 work successfully, we can check the boot loaders files on LittleFS, correct size and contents.
Steps 4 and 5 run correctly, no errors reported. When we check read the flash with esptool and check the new bootloader written all looks correct, offsets and lengths.

Before the bootloader update ROM startup output was as follows:
```
ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6156
ho 0 tail 12 room 4
load:0x40078000,len:16112
load:0x40080400,len:4
load:0x40080404,len:3712
entry 0x400805dc
```

After bootloader update ROM startup shows the following repeatedly:
```
ets Jul 29 2019 12:21:46
rst:0xc (SW_CPU_RESET),boot:0x1e (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:6144
load:0x00000000,len:1634885484
```

Based on the very different 'load' information I assume that I have failed to update some pointers and/or sizes somewhere.

Any suggestions.

KonstantinKondrashov, 2024-07-17:
Hi @ammaree!
Could you clarify some points.
- If you read out the newly written bootloader and then run `image_info` for this file do you see that it is correct and the same as on the host?

```
esptool.py read_flash 0x1000  <SIZE>  path/to/readout_new_bootloader.bin
esptool.py image_info path/to/readout_new_bootloader.bin
```
- Does this new bootloader work well if you write it directly to 0x1000 using esptool?
- Do not you forget to erase the 0x1000 to CONFIG_PARTITION_TABLE_OFFSET, right?

ammaree, 2024-07-17:
1. Will read back and check, let you know. 
2. New bootloader, like all previous, are 100% standard, nothing modified so should work if written by esptool
3. Just so I understand areas to be erased before, are you saying:
0x0000 to 0x0FFF, I believe unused?
and 0x1000 to 0x7FFF, normal bootloader?

KonstantinKondrashov, 2024-07-17:
> 3. Just so I understand areas to be erased before, are you saying:
0x0000 to 0x0FFF, I believe unused?
and 0x1000 to 0x7FFF, normal bootloader?

Normally we consider the bootloader size as = (CONFIG_PARTITION_TABLE_OFFSET - ESP_BOOTLOADER_OFFSET). The bootloader must not exceed the partition table location. If signature block(s) are appended, they should be erased too. To guarantee that you have to erase 0x1000 - CONFIG_PARTITION_TABLE_OFFSET.

ammaree, 2024-07-17:
@KonstantinKondrashov You are a star!!!!

Erase did the trick, I "assumed" it would be part of the flash write operation.

Thanks for quick response.

KonstantinKondrashov, 2024-07-17:
I am glad to hear that it was resolved.

FYI, I am working on Bootloader OTA. It will have two modes:
- Power-unsafe OTA bootloader update. I guess it could be similar to your implementation.  I have expanded the OTA APIs to download the new bootloader.
- Power-safe OTA bootloader update. A new chip version will have some changes in the ROM bootloader which will be able to select a recovery bootloader in case the primary bootloader fails to load.

KonstantinKondrashov, 2024-07-17:
I close this issue. Thanks.

closed 2024-07-17 by KonstantinKondrashov

ammaree, 2024-07-23:
@KonstantinKondrashov 

Last question. In order to change the spiffs and/or coredump partition sizes the MD5 checksum becomes a problem.

Currently I am scanning for the checksum indicator (right at the end) and clearing the values to all 0xFF before writing the changed partition table back.

Ideally I would like to recalculate and update the checksum going forward, especially for old devices where the bootloader has been updated.

Can you please provide some info on the area to be included in the checksum and the algorithm used, hopefully a standard IDF API?

Thanks

KonstantinKondrashov, 2024-07-24:
Hi @ammaree!
MD5 for the partition table is used if the CONFIG_PARTITION_TABLE_MD5 is enabled.
There are ROM APIs that are used to calculate MD5. You can have a look code in IDF:
- https://github.com/espressif/esp-idf/blob/master/components/esp_partition/partition.c#L88-L91
- https://github.com/espressif/esp-idf/blob/master/components/bootloader_support/src/flash_partitions.c#L41-L54

```c
#include "esp_rom_md5.h"
            
md5_context_t context;
unsigned char digest[16];
esp_rom_md5_init(&context);
esp_rom_md5_update(&context, (unsigned char *) partition_table, num_parts * sizeof(esp_partition_info_t));
esp_rom_md5_final(digest, &context);
```

ammaree, 2024-07-25:
@KonstantinKondrashov 
Sorry, hopefully last question...

Been looking for documentation on the esp_rom_md5_???? APIs but cannot find much, any link that will help?

esp_rom_md5__init is self explanatory.
esp_rom_md5_update will update the MD5 context with the calculated value from the range specified, either the whole partition table as 1, of in steps of an entry at a time.
esp_rom_md5_final I would prefer not to make assumptions on...

Primarily interested in verifying a) calculating a digest and b) verifying an existing digest...

Thanks

KonstantinKondrashov, 2024-07-25:
@ammaree
The description of ROM MD5 functions. An example of their usage for the partition table was in the previous message.
https://github.com/espressif/esp-idf/blob/master/components/esp_rom/include/esp_rom_md5.h#L74-L106

You may look at the mbedtls APIs for MD5, here is an example:
https://github.com/espressif/esp-idf/blob/master/examples/storage/spiffsgen/main/spiffsgen_example_main.c#L38-L79

Sorry, I guess that is all that we have for now. There is not a particular doc for that. You can try to search for more examples in the esp-idf repo by "md5".

ammaree, 2024-08-07:
Hi @KonstantinKondrashov 

Thanks for all the info above, got the process for auditing, changing and updating partition table working.
Also got safe & reliable download and updating of bootloader working.

We have however now come across the complication of matching
#versions/configs of partition tables, essentially with or without MD5; with
#versions of boot loaders, essentially again whether they ignore and/or handle and/or expect an MD5 record in the partition table.

The majority of old devices were deployed Feb to May 2018 and those are the ones where we are unsure of the bootloader & partition table versions and configurations. We assume they all use IDF v2.1.1 based bootloaders with a couple of later replacements using IDF 3.0 and 3.1 bootloaders

To get all devices to a common minimum standard we foresee the following steps.
1. Deploy "audit" firmware to report( via syslog) all info on the existing partition table as well as the existing bootloader.

2. Armed with this info develop and deploy firmware to update all partition tables to a similar structure, primarily coredump and littlefs related. If MD5 is present, update it. If not present leave as is, do not add. This functionality we already have, tested and working.

3. The develop and deploy firmware to do the bootloader upgrade. This step we need help on since the mixture of devices with/without MD5 must be accommodated transparently. Probably deploy a new bootloader with MD5 support but at same time time, immediately after OTA of new bootloader, also add MD5 to partition table if not there, the reboot.

Regarding 1 above:
Our current thought is build and deploy new "audit" firmware version that does an audit of the partition table and logs that info to our syslog host. We would like to also audit the installed boot-loader and report all info relating to the version and options it was build with, specifically MD5 support but also other items relevant to later doing a boot-loader upgrade.

Auditing the partition table we have done and working, what remains is the bootloader audit.

Core questions are:
A: Is there any information, hints, APIs or similar that we can use to extract this info programmatically from existing bootloaders?
B: If not, what are the safest build options to use for building this special "audit" firmware and an associated new bootloader that will work on the widest range (all combinations) of partition tables?

Thanks again for your assistance.

André

ammaree, 2024-09-03:
@KonstantinKondrashov 

May I again ask for some advice on the questions (especially A) in the message above, really need to try finalise the bootloader audit.