espressif/esp-idf #14197: icmp_echo func:cmd_ping_on_ping_end() get wrong ipv6 address

gl-liuqingbo, 2024-07-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

1. I use my router to enable ipv6,then conncet esp32 by eth way.
2. esp32 run esp-idf/examples/protocols/icmp_echo
3. ping ipv6 address
4. Log display another ipv6 address, I'm confused.
Here's log message:
[image]

espressif-abhikroy, 2024-07-23:
@gl-liuqingbo Thanks for reporting the issue.
It's happening because of an issue in ping_sock.c.
The periodic Router Advertisement packet sent by your router was overwriting the destination address.
This can be verified as the updated address is EUI-64 link local address of your router.

I am working on the fix and it will be available soon.

gl-liuqingbo, 2024-07-24:
@espressif-abhikroy Thank you for your help, I will wait for the follow-up update

AxelLin, 2024-08-12:
> I am working on the fix and it will be available soon.

@espressif-abhikroy 
I still cannot find this fix. Please also fix it in stable releases.

closed 2024-08-21 by espressif-bot

espressif-abhikroy, 2024-08-21:
The fix has been applied and is now available on the master branch. You can refer to the commit with the hash 53ba6e97cde2b5c8cf1e1bbb374fe887643b688a.